weimarnetz/weimarnetz #199: Updates: /etc/config/system enthält doppelte Sessions

andibraeu, 2015-12-07:
Ursache: Wechsel des Schemas für Benennung von Sections, betroffen sind:
```
config weblogin 'weblogin'                                    
config profile 'profile'                                      
config monitoring 'monitoring'                                
config fwupdate 'fwupdate'                                    
config vpn 'vpn'
```

Altes Schema:
```
config weblogin
config profile
config monitoring
config fwupdate
config vpn
```

Lösung: Nach Erzeugung von /etc/config/system sicherheitshalber die Sections nach dem alten Schema löschen. Muss getestet werden, damit wir die richtigen Sessions löschen

bittorf, 2015-12-08:
zeig mal die ausgabe von 'uci show system' mit doppelten eintraegen.
und: wo passiert das loeschen? in apply_profile?

andibraeu, 2015-12-08:
müssen es doch nicht unbedingt löschen, hab die zugriffe auf die variablen umgestellt

andibraeu, 2015-12-08:
läuft...

PR #203 (2015-12-08): use unique config sections from uci
fixes #199
closed 2015-12-08 by andibraeu