weimarnetz/weimarnetz #204: Käferbericht: Aliase wie z.B. n und lr funzen nicht

andibraeu, 2015-12-08:
Aliase aus /etc/profile.d/kalua.sh werden nicht gesetzt

andibraeu, 2015-12-08:
da liegt der Hase im Pfeffer: https://github.com/weimarnetz/weimarnetz/blob/GebrannteMandeln/openwrt-addons/etc/kalua_init.user#L90

glycoknob, 2015-12-08:
hmm, der neue profile.d support ist durch eine ältere /etc/profile überschrieben, frag mich nur woher die kommt: 

original: 

```
root@netbox:~ cat /rom/etc/profile
#!/bin/sh
[ -f /etc/banner ] && cat /etc/banner
[ -e /tmp/.failsafe ] && cat /etc/banner.failsafe
fgrep -sq '/ overlay ro,' /proc/mounts && {
        echo 'Your JFFS2-partition seems full and overlayfs is mounted read-only.'
        echo 'Please try to remove files from /overlay/upper/... and reboot!'
}

export PATH=/usr/sbin:/usr/bin:/sbin:/bin
export HOME=$(grep -e "^${USER:-root}:" /etc/passwd | cut -d ":" -f 6)
export HOME=${HOME:-/root}
export PS1='\u@\h:\w\$ '

[ -x /bin/more ] || alias more=less
[ -x /usr/bin/vim ] && alias vi=vim || alias vim=vi

[ -z "$KSH_VERSION" -o \! -s /etc/mkshrc ] || . /etc/mkshrc

[ -x /usr/bin/arp ] || arp() { cat /proc/net/arp; }
[ -x /usr/bin/ldd ] || ldd() { LD_TRACE_LOADED_OBJECTS=1 $*; }

[ -n "$FAILSAFE" ] || {
        for FILE in /etc/profile.d/*.sh; do
                [ -e "$FILE" ] && . "$FILE"
        done
        unset FILE
}
```

aktuell: 

```
root@netbox:~ ls -l /etc/profile
-rw-rw-r--    1 root     root           972 Sep 12 15:13 /etc/profile
```

```
#!/bin/sh
[ -f /etc/banner ] && cat /etc/banner
[ -e /tmp/.failsafe ] && cat /etc/banner.failsafe

export PATH=/usr/bin:/usr/sbin:/bin:/sbin
export HOME=$(grep -e "^${USER:-root}:" /etc/passwd | cut -d ":" -f 6)
export HOME=${HOME:-/root}
export PS1='\u@\h:\w\$ '

[ -x /bin/more ] || alias more=less
[ -x /usr/bin/vim ] && alias vi=vim || alias vim=vi

[ -z "$KSH_VERSION" -o \! -s /etc/mkshrc ] || . /etc/mkshrc

[ -x /usr/bin/arp ] || arp() { cat /proc/net/arp; }
[ -x /usr/bin/ldd ] || ldd() { LD_TRACE_LOADED_OBJECTS=1 $*; }
alias ll='ls -la'
alias flush='echo flushing_caches; echo 3 > /proc/sys/vm/drop_caches'
alias lr='logread'
alias regen='/etc/kalua_init ;. /tmp/loader'
[ -e /tmp/loader ] && . /tmp/loader
alias n='wget -qO - http://127.0.0.1:2006/neighbours'
export PS1="\[\033[36m\]\u\[\033[m\]@\[\033[32m\]\h:\[\033[33;1m\]\w\[\033[m\] "
alias myssh='echo "executing: ssh -i /etc/dropbear/dropbear_dss_host_key"; ssh -i /etc/dropbear/dropbear_dss_host_key'
```

glycoknob, 2015-12-09:
einfachste Lösung wäre die Datei aus dem /rom beim apply_profile.code rüberzukopieren.

bittorf, 2015-12-09:
die ältere /etc/profile kommt aus einem sysupgrade. es gibt auch einen patch dazu, da wurde aber bisher nicht akzeptiert: https://patchwork.ozlabs.org/patch/528164/

ein einfacher workaround waere wirklich bei apply_profile immer die datei aus dem /rom zu holen, also
```
mv /rom/etc/profile /etc
```

andibraeu, 2015-12-09:
ok, so läuft das auch schon mit der opkg.conf

andibraeu, 2015-12-09:
@glycoknob ist das auf trunk?

auf einem frisch installierten CC1505 sieht die /etc/profile so aus:
```
#!/bin/sh
[ -f /etc/banner ] && cat /etc/banner
[ -e /tmp/.failsafe ] && cat /etc/banner.failsafe

export PATH=/usr/bin:/usr/sbin:/bin:/sbin
export HOME=$(grep -e "^${USER:-root}:" /etc/passwd | cut -d ":" -f 6)
export HOME=${HOME:-/root}
export PS1='\u@\h:\w\$ '

[ -x /bin/more ] || alias more=less
[ -x /usr/bin/vim ] && alias vi=vim || alias vim=vi

[ -z "$KSH_VERSION" -o \! -s /etc/mkshrc ] || . /etc/mkshrc

[ -x /usr/bin/arp ] || arp() { cat /proc/net/arp; }
[ -x /usr/bin/ldd ] || ldd() { LD_TRACE_LOADED_OBJECTS=1 $*; }
```

andibraeu, 2015-12-09:
@bittorf in /etc/kalua_init.user erfolgt ja auch ein Aufruf von /etc/profile.d/kalua.sh, siehe https://github.com/weimarnetz/weimarnetz/issues/204#issuecomment-163037052
allerdings

bittorf, 2015-12-09:
@andibraeu die variable $OPENWRT_REV wird durch den 'loader' gesetzt.
"echo $OPENWRT_REV" sollte also was ausgeben.
die
/etc/profile die du da gesendet hast, hat noch keinen support fuer profile.d - das koennte man reinpatchen: https://patchwork.ozlabs.org/patch/514352/

glycoknob, 2015-12-09:
@bittorf das wäre wohl die beste Lösung, ich schieb den Patch mal in den Builder. Dann noch die Datei aus dem ROM kopieren und alles ist schön oder?

andibraeu, 2015-12-09:
@bittorf wenn ich den loader loade gibt das ```echo``` auch die korrekte $OPENWRT_REV aus. wenn ich dann allerdings /etc/kalua_init.user aufrufe, bleibt die Variable im Script trotzdem leer. Darin wird der Loader auch gar nicht aufgerufen...

glycoknob, 2015-12-09:
Also der Patch ist in trunk aber nicht in CC oder?

bittorf, 2015-12-09:
@andibraeu die datei /etc/kalua_init.user darf nur von /etc/kalua_init aufgerufen werden. die ist quasi teil der infrastruktur um den 'loader' zusammenzubauen:

ls -l /etc/kalua_init*

bittorf, 2015-12-09:
@glycoknob ja, ich glaube das fehlt in CC. entweder man strengt einen backport an oder man macht es einfach selbst rein. ich halte es auf jeden fall fuer sinnvoll das drin zu haben.

andibraeu, 2015-12-09:
@bittorf dann scheint da trotzdem noch ein fehler drin zu sein. Kannst auf diesem Router nachgucken: 10.63.76.97

Meine Vermutung, das funktioniert nirgends, und nur wegen der alten /etc/profile vom sysupgrade haben wir die Aliase auf unseren Routern

bittorf, 2015-12-09:
ich pruefe

bittorf, 2015-12-09:
das stimmt, erst ein
```
. /tmp/loader
. /etc/profile.d/kalua.sh
```
bringt die loesung.
ich bin dafuer den profile.d-support aktiv reinzunehmen und den hack mit
```
mv /rom/etc/profile /etc
```
in apply_profile zu machen.

storchi, 2015-12-09:
gestern reichte auch ein:
. /etc/profile.d/kalua.sh

andibraeu, 2015-12-09:
ja, ich denke auch, der Aufruf von /etc/kalua_init.user funktioniert nicht

bittorf, 2015-12-09:
@storchi ja, weil der loader selbst ueber die "alte" (durch sysupgrade gesicherte) /etc/profile geladen wird.
@andibraeu was willst du immer nur mit der "/etc/kalua_init.user"? das ist eine datei die du nicht anfassen sollst, sondern nur /etc/kalua_init

andibraeu, 2015-12-09:
@bittorf in dieser Datei steht drin, dass /etc/profile.d/kalua.sh aufgerufen wird. Und das passiert nicht!

bittorf, 2015-12-09:
@andi - das liest du nicht richtig. daraus wird code generiert und das auch nur auf dem VPN-Server ("OPENWRT_REV" = 0 ). All diese /etc/kalua_init.* Dateien geben Zeile fuer Zeile den 'loader' aus - es wird also TEXT generiert.

bittorf, 2015-12-09:
ich habe eine idee fuer einen workaround und sende einen pullrequest

andibraeu, 2015-12-10:
klappt noch nicht ganz: erst wenn ich . /tmp/loader aufrufe wird kalua.sh geladen

bittorf, 2015-12-10:
ok verstanden, das fixe ich noch

andibraeu, 2015-12-12:
geht immer noch nicht, aber hiermit sollte das gehen: https://github.com/weimarnetz/builder/pull/16

andibraeu, 2015-12-12:
mit dem Patch gehts nun

PR #205 (2015-12-10): kalua: loader: sync with upstream
always load '/etc/profile.d/kalua.sh' when interactive
and it seems there is no profile.d support. fixed #204
commit e9c20939 (2015-12-10): kalua: loader: sync with upstream

always load '/etc/profile.d/kalua.sh' when interactive
and it seems there is no profile.d support. fixed #204
PR #217 (2015-12-11): kalua: loader: sync with upstream
avoid double inclusion of /etc/profile.d/kalua.sh
fixes #204
closed 2015-12-12 by andibraeu