weimarnetz/weimarnetz #206: SmartGateway!?!?

glycoknob, 2015-12-09:
Hallo, 

weil ja eh grad alles updaten... macht es Sinn SmartGateway anzuknipsen? das kmod-ipip ist ziemlich klein, das wäre kein Problem - der Vorteil wäre wohl, dass schwankende Routen über Luft so stabiler sind bzw. 

Gibt bestimmte einen Grund warum das nicht aktiv ist @bittorf? 

Das hier sieht doch schön aus: http://www.olsr.org/files/multi-smart-gateway.pdf

Das würde auch erlauben nach Exit-Bandbreite zu sortieren? http://olsr.org/git/?p=olsrd.git;a=blob;f=README-Olsr-Extensions;hb=HEAD 

macht das Sinn? Oder eher nicht?

glycoknob, 2015-12-09:
Hier ist noch eine Erklärung: http://berlin.berlin.freifunk.narkive.com/19P6PrfH/berlin-wireless-olsr-und-smart-gateway

andibraeu, 2015-12-09:
es geht ja vor allem darum. dass die Routen nicht so schnell wechseln, und da gabs was, was "sticky routes" hieß und nicht so aufwändig war wie smart gateway... und ich würde jetzt auch ungern noch mehr komplexität reinpacken

bittorf, 2015-12-10:
also smartgateway macht in berlin viele sorgen. wir haben das problem mit den wechselnen DSLer garnicht und sowieso das "sticky" eingeschaltet. man kann kmod-ipip mit reinnehmen und das OLSR-plugin auch, aber nicht aktivieren - dann kann man zumindest leicht experimente machen.

glycoknob, 2015-12-10:
danke für die infos, dann schließ ich das mal.

closed 2015-12-10 by glycoknob