weimarnetz/weimarnetz #211: DHCP kaputt - Adresse kommt aus 10.0.0.0/24 Bereich!?

glycoknob, 2015-12-10:
Grund ist die adhoc subnetzmaske... 10.0.0.0/8 

Fix ist DHCP auf adhoc iface zu ignorieren: 

     config dhcp 'wlanadhoc'
        option interface 'wlanadhoc'
        option leasetime '1h'
        option force '1'
        option ignore '1'

Woher kommt das? 

Hier noch ein paar Daten: 

dnsmasq auf Router: 

```
Thu Dec 10 15:19:53 2015 daemon.info dnsmasq[16414]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-DNSSEC loop-detect inotify
Thu Dec 10 15:19:53 2015 daemon.info dnsmasq-dhcp[16414]: DHCP, IP range 10.63.188.2 -- 10.63.188.30, lease time 1h
Thu Dec 10 15:19:53 2015 daemon.info dnsmasq-dhcp[16414]: DHCP, IP range 10.63.188.34 -- 10.63.188.46, lease time 2h
Thu Dec 10 15:19:53 2015 daemon.info dnsmasq-dhcp[16414]: DHCP, IP range 10.0.0.100 -- 10.0.0.249, lease time 1h
``` 

via Kabel: 

```
Dez 10 15:17:05 zeno dhclient[9246]: DHCPDISCOVER on eth1 to 255.255.255.255 port 67 interval 6 (xid=0xd436c751)
Dez 10 15:17:05 zeno dhclient[9246]: DHCPREQUEST of 10.0.0.114 on eth1 to 255.255.255.255 port 67 (xid=0x51c736d4)
Dez 10 15:17:05 zeno dhclient[9246]: DHCPOFFER of 10.0.0.114 from 10.63.188.33
Dez 10 15:17:05 zeno dhclient[9246]: DHCPACK of 10.0.0.114 from 10.63.188.33
Dez 10 15:17:05 zeno NetworkManager[662]: <info>    address 10.0.0.114
Dez 10 15:17:05 zeno NetworkManager[662]: <info>    plen 8 (255.0.0.0)
Dez 10 15:17:05 zeno NetworkManager[662]: <info>    gateway 10.63.188.33
Dez 10 15:17:05 zeno NetworkManager[662]: <info>    server identifier 10.63.188.33
Dez 10 15:17:05 zeno NetworkManager[662]: <info>    lease time 3600
Dez 10 15:17:05 zeno NetworkManager[662]: <info>    hostname 'zeno'
Dez 10 15:17:05 zeno NetworkManager[662]: <info>    nameserver '10.63.188.33'
Dez 10 15:17:05 zeno NetworkManager[662]: <info>    domain name 'olsr'
Dez 10 15:17:05 zeno NetworkManager[662]: <info>  (eth1): DHCPv4 state changed unknown -> bound
Dez 10 15:17:05 zeno NetworkManager[662]: <info>  (eth1): device state change: ip-config -> ip-check (reason 'none') [70 80 0]
Dez 10 15:17:05 zeno avahi-daemon[673]: Joining mDNS multicast group on interface eth1.IPv4 with address 10.0.0.114.
Dez 10 15:17:05 zeno avahi-daemon[673]: New relevant interface eth1.IPv4 for mDNS.
Dez 10 15:17:05 zeno avahi-daemon[673]: Registering new address record for 10.0.0.114 on eth1.IPv4.
Dez 10 15:17:05 zeno NetworkManager[662]: <info>  (eth1): device state change: ip-check -> secondaries (reason 'none') [80 90 0]
Dez 10 15:17:05 zeno NetworkManager[662]: <info>  (eth1): device state change: secondaries -> activated (reason 'none') [90 100 0]
Dez 10 15:17:05 zeno NetworkManager[662]: <info>  NetworkManager state is now CONNECTED_LOCAL
Dez 10 15:17:05 zeno NetworkManager[662]: <info>  NetworkManager state is now CONNECTED_GLOBAL
Dez 10 15:17:05 zeno NetworkManager[662]: <info>  Policy set 'eth1' (eth1) as default for IPv4 routing and DNS.
Dez 10 15:17:05 zeno NetworkManager[662]: <info>  (eth1): Activation: successful, device activated.
Dez 10 15:17:05 zeno dnsmasq[1211]: reading /etc/resolv.conf
Dez 10 15:17:05 zeno dnsmasq[1211]: using nameserver 127.0.1.1#53
Dez 10 15:17:05 zeno dnsmasq[901]: setting upstream servers from DBus
Dez 10 15:17:05 zeno dnsmasq[901]: using nameserver 10.63.188.33#53
Dez 10 15:17:05 zeno dnsmasq[901]: using nameserver 8.8.8.8#53
``` 

macht natürlich alles kaputt. 

ip a auf router: 

``` 
root@netbox:~ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 10:fe:ed:2b:91:2f brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.103/24 brd 192.168.2.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fd26:b81a:60a:0:12fe:edff:fe2b:912f/64 scope global noprefixroute dynamic 
       valid_lft 7092sec preferred_lft 1692sec
    inet6 2003:78:ae5b:fa00:12fe:edff:fe2b:912f/64 scope global noprefixroute dynamic 
       valid_lft 7072sec preferred_lft 903sec
    inet6 fe80::12fe:edff:fe2b:912f/64 scope link 
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc prio state UP group default qlen 1000
    link/ether 10:fe:ed:2b:91:30 brd ff:ff:ff:ff:ff:ff
    inet 10.63.188.33/28 brd 10.63.188.47 scope global eth1
       valid_lft forever preferred_lft forever
    inet6 fe80::12fe:edff:fe2b:9130/64 scope link 
       valid_lft forever preferred_lft forever
4: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
6: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc prio state UP group default qlen 1000
    link/ether 10:fe:ed:2b:91:30 brd ff:ff:ff:ff:ff:ff
    inet 10.63.188.1/8 brd 10.255.255.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::12fe:edff:fe2b:9130/64 scope link 
       valid_lft forever preferred_lft forever
7: wlan0-1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 12:fe:ed:2b:91:30 brd ff:ff:ff:ff:ff:ff
    inet 10.63.188.2/27 brd 10.63.188.31 scope global wlan0-1
       valid_lft forever preferred_lft forever
    inet6 fe80::10fe:edff:fe2b:9130/64 scope link 
       valid_lft forever preferred_lft forever
69: tap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1434 qdisc fq_codel state UNKNOWN group default qlen 500
    link/ether ae:34:6e:9e:22:c7 brd ff:ff:ff:ff:ff:ff
    inet 10.63.188.58/30 scope global tap0
       valid_lft forever preferred_lft forever
    inet 10.63.188.58/32 scope global tap0
       valid_lft forever preferred_lft forever
    inet6 2001:bf7:1930:bcff::/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::ac34:6eff:fe9e:22c7/64 scope link 
       valid_lft forever preferred_lft forever
```

glycoknob, 2015-12-10:
Das Problem macht für mich alles unbenutzbar, Router nicht mehr erreichbar kein Netz/Zugang über Kabel oder Wifi

glycoknob, 2015-12-10:
Tritt bei Update und auch unter CC auf - grund ist wohl die Sektion für wlanadhoc bei dnsmasq die automatisch? angelegt wird?

glycoknob, 2015-12-10:
Problem tritt nur auf wenn es eine Sektion für wlanadhoc in /etc/config/dhcp gibt.... dürfte also nur meinen Router betreffen, aber können wir ja trotzdem fixen.

andibraeu, 2015-12-10:
hab jetzt auf die schnelle keinen anderen Router gefunden, wo es so eine Sektion gibt. applyprofile.code setzt die Sektion auch nicht, spannend ist, wie sie bei dir aufs Gerät kam.
Was genau willst du fixen?

glycoknob, 2015-12-10:
Hab das mal von Hand angelegt, hmmm fällt wohl unter selbst schuld. Ich schließe mal, Sorry.

closed 2015-12-10 by glycoknob