weimarnetz/weimarnetz #216: Anleitung Firmwareinstallation

andibraeu, 2015-12-11:
Im Wiki haben wir schon eine Anleitung zum Update: http://wireless.subsignal.org/index.php?title=Update#Einfaches_Update_.C3.BCber_das_.28LuCI-.29_Webinterface

Was soll gezeigt werden?

* Image aus Meshkit bauen
* richtiges Image auswählen: http://wireless.subsignal.org/index.php?title=Auswahl_der_passenden_Firmware
* Installation auf einem frischen Router
* Hinweis, das Adminpasswort zu setzen

siehe auch #187 #215 #193

storchi, 2015-12-11:
irgendwas müssen wir auch mit den Dateinamen der images machen, die aus dem meshkit purzeln. Die sind zu lang.
cs

bittorf, 2015-12-11:
ja, davon rede ich seit *jahren* 8-) wie waer mit
```
TP-LINK_TL-WR841N/ND_v1.5.sysupgrade.bin
TP-LINK_TL-WR841N/ND_v1.5.factory.bin
```
wir untertuetzen doch eh nur 5 modelle...

FreifunkUFO, 2015-12-11:
das ist doch kein weimarnetz problem, sondern ganz allgemein von openwrt :-o 
(seit wahrsch. genausoviel Jahren)

bittorf, 2015-12-11:
ja, aber OpenWrt wird es nicht aendern. Wir jedoch koennte einen simplen convert bauen:
openwrt-name -> guter_name

andibraeu, 2015-12-11:
Wie lang dürfen die Namen sein? Hab noch eine Option gefunden, die Versionsnummer wegzulassen

bittorf, 2015-12-11:
gerade die versionsnummer ist interessant. ich habe gerade obiges beispiel gewaehlt um zu zeigen das es nicht ganz einfach ist:

"TP-LINK TL-WR841N/ND v1.5"

hat also leerzeichen, schraegstriche und punkt drinne.
Zum einen koennte man zwei images bauen:

TP-LINK TL-WR841N v1.5
TP-LINK TL-WR841ND v1.5
(dann faellt der schraegstrich weg)

der punkt muss bleiben.
leerzeichen koennen "_" werden:

TP-LINK_TL-WR841ND_v1.5.sysupgrade.bin
(23 + 11 + 4 = 38 Zeichen!)

*.bin wuerde ich in jedem fall dranhaengen, das
macht sowohl programmen als auch mensch klar was es ist.

fehlt noch der hinweis auf weimarnetz und versionsnummer:

weimarnetz-GebrannteMandeln-TP-LINK_TL-WR841ND_v1.5.sysupgrade.bin

ich gebe zu das ist zu lang. 32 Zeichen waeren cool (macht dann auch auf dem AtariST
keine probleme mehr 8-)... Christian: kannst du pruefen wie lang der Dateiname bei TP-Link sein darf?

andibraeu, 2015-12-11:
Mit Versionsnummer meinte ich "OpenWRT-1505", was da gerade vorangestellt wird. Die Versionsinformation kann man immer noch ins Downloadverzeichnis packen

wir können an dieser Stelle nicht das grundlegende Namensschema von OpenWRT ändern, vor allem nicht, welche Images für welche Hardware gebaut werden. Das ist eher ein Fall für ein Issue bei OpenWRT, oder gibts da schon was?

bittorf, 2015-12-11:
wie ich schon sagte: openwrt da *nichts* aendern, weil das alle kaputtmachen wuerde (rückwärtskompatibilität). Wir jedoch koennen einfach aus DateiX eine DateiY auf den server ladieren...also vielleicht:
``
weCC1505-TP-LINK_TL-WR841ND_v1.5.bin
weCC1505-TP-LINK_TL-WR841ND_v1.5.factory.bin
```

andibraeu, 2015-12-11:
das meshkit macht auch nur ein openwrt-make und baut ein Image...

Bin eher dafür, dass man als Website löst, der Nutzer wählt sein Gerät aus und bekommt die richtige Datei zum Download angezeigt

storchi, 2015-12-12:
Wenn das alles so kompliziert wird, hilf evtl. auch nur ein Hinweis auf der Downloadseite. Das ist ja eh nur ein "Schönheitsfehler". Technisch ist es und wirds auch bleiben, da man um die Revisionsnummer der Geräte nicht herum kommt.
cs

storchi, 2015-12-12:
@bittorf hab keinen original tp link hier

bittorf, 2015-12-13:
wenn ich dies lese, sollte der dateiname max 40-46 zeichen lang sein: https://github.com/freifunk-berlin/firmware/issues/178

```
sysupgrade:  (36 zeichen)
weCC1505-TP-LINK_TL-WR841ND_v1.5.bin
fabrik: (44 zeichen)
weCC1505-TP-LINK_TL-WR841ND_v1.5.factory.bin
```