weimarnetz/weimarnetz #218: WDR3600 - VPN-Verbindung

andibraeu, 2015-12-14:
Beispielknoten: 319 und 117

Die Verbindung von diesen Knoten wird immer wieder zurückgesetzt und hält nur maximal 15 Sekunden durch. Vermutlich haut da ein Check dazwischen, der die ordnungsgemäße Verbindung falsch diagnostiziert und VPN neu zündet. Vielleicht liefert _switch show oder _watch switch_linkstatus falsche Ergebnisse

@kaese77 kannst du bitte mal deine Logs von 319 posten?

kaese77, 2015-12-14:
Moin,

mach grad noch das sysupgrade....

patience, pls.

René

storchi, 2015-12-14:
guck mal ob da olsr über da vpn if geht - port 698

storchi, 2015-12-14:
die arp Auflösung klappt nicht:
13:26:48.508754 ARP, Request who-has 10.63.122.57 tell 10.63.122.58, length 28
damit kein ip und kein olsr -> neuzündung

"Früher" haben wir mal im vpn up script die mac adresse des einen vpn servers fest reincodiert: ip n a ...
wo isn das hin und wie wird das jetzt gemacht?

cs

andibraeu, 2015-12-14:
ist die 122 auch ein wdr3600?

storchi, 2015-12-14:
ja, mit cc gm von heute

andibraeu, 2015-12-14:
frag mich, warum das problem nur auf 3600ern auftritt... switch-config?

kaese77, 2015-12-14:
Ich habe den Router ja in einer DMZ.

Für Port 698 habe ich kein eingehendes NAT definiert. Wäre das notwendig?

René

storchi, 2015-12-14:
nein, da es ja im tunnel läuft

storchi, 2015-12-14:
der arp request kommt auf dem server tap if nicht an!
cs

storchi, 2015-12-15:
@kaese77 kannst du auf mal testweise deinen knoten die vpn mtu auf 1400 stellen und mit vpn3 probieren?
ich seh im log "message to long" vom vtund - hat was mit fragmentierung zu tun
cs

storchi, 2015-12-16:
bei den kandidaten die nach ca 15s die verbindung abbrechen sehe ich mit tcpdump:
15:43:44.133578 IP (tos 0x0, ttl 64, id 52416, offset 0, flags [DF], proto UDP (17), length 792)
    77.87.48.35.5001 > 178.19.236.123.44293: [bad udp cksum 0x1f1f -> 0xcc7a!] UDP, length 764
	0x0000:  4500 0318 ccc0 4000 4011 4f0b 4d57 3023  E.....@.@.O.MW0#
	0x0010:  b213 ec7b 1389 ad05 0304 1f1f 02fa ffff  ...{............
	0x0020:  ffff ffff 0a0c cc8e f19f 0800 45c0 02ec  ............E...
	0x0030:  550d 4000 4011 62bc 0a3f 7539 ffff ffff  U.@.@.b..?u9....
immer mal wieder

glycoknob, 2015-12-16:
@storchi könntest den Dumpfile mal irgendwo hochladen? Oder mir schicken? Danke!

storchi, 2015-12-16:
jo
[1.txt](https://github.com/weimarnetz/weimarnetz/files/64125/1.txt)

storchi, 2015-12-16:
http://git.openwrt.org/?p=15.05/openwrt.git;a=commit;h=c84aea13d479f3ef784b363f907926e10e313a26
???

andibraeu, 2015-12-16:
du könntest mit dem Builder neu bauen und gucken ob es besser wird: 

```./build_release.sh meta.ffweimar-meshkit use_cc1505```

der zieht automatisch den letzten commit von CC1505

storchi, 2015-12-16:
hab gleich Feierabend...

storchi, 2015-12-16:
locking vom vtun ist auf 3.v.weimarnetz jetzt erstmal aus

glycoknob, 2015-12-16:
Im Dump fällt mir nur folgendes auf: Fragmentierung ist wohl noch da, interessant. Könnten wir beim Treffen nochmal MTU etc.pp durchgehen. Das müsste aber egal sein. 

Kaputte UDP Pakete.... könnte Checksum Offloading der Netzwerkkarte sein!? Bzw. irgendwas mit der NIC... so sieht man was aktiv ist: 

    sudo ethtool -k eth0
    sudo ethtool -K eth0 gso off

Bei Intel Karten zerhaut es mir regelmäßig SSH-Connections, wenn das aktiv ist mit "MAC malformed input" - je nach Nic könnte das hier auch was kaputt machen. 

Ansonsten hab ich keine Idee.

glycoknob, 2015-12-16:
Hier noch ein Link: https://sandilands.info/sgordon/segmentation-offloading-with-wireshark-and-ethtool

glycoknob, 2015-12-16:
Also eine Theorie wäre, dass wenn Fragmentiert wird irgendwas die UDP zerhaut.. würde auch erklären warum es ansonsten läuft. Zum testen könnte man die ganzen Offloading-Geschichten der NIC mal abschalten.

bittorf, 2015-12-16:
@storchi mit welchem tcpdump-befehl exakt siehst du die fehlermeldung "bad udp cksum"? auf dem server oder auf dem client?

glycoknob, 2015-12-16:
Hier ist das Ticket zum Commit: https://dev.openwrt.org/ticket/18922 

Darum wäre es z.B. schön wenn wir printk() und Crashlogging drin hätten.

bittorf, 2015-12-16:
@glycoknob der openwrt-fix hat IMHO nix mit  dem rpoblem zu tun.

storchi, 2015-12-17:
Am 16.12.2015 um 18:03 schrieb Bastian Bittorf:
> @glycoknob <https://github.com/glycoknob> der openwrt-fix hat IMHO nix
> mit dem rpoblem zu tun.

es gab nur sehr wenige änderungen seit dem das vpn gut ging. das war eine.
eine weitere waren die tippmanschen iptables filter pakete, die 
eingespart wurden
cs

glycoknob, 2015-12-17:
Hier ist der Commit: https://github.com/weimarnetz/builder/commit/bfe5a1dd7128e9d27e081da612bd110a0b5eb9d0

Gab aber keine Fehler. Eine andere Änderung wäre noch das notrack Target - aber das sollte die Pakete auch nicht anfassen.

storchi, 2015-12-17:
Am 17.12.2015 um 10:03 schrieb Martin Tippmann:
> Hier ist der Commit: weimarnetz/builder@bfe5a1d
> <https://github.com/weimarnetz/builder/commit/bfe5a1dd7128e9d27e081da612bd110a0b5eb9d0>
>
> Gab aber keine Fehler. Eine andere Änderung wäre noch das notrack Target
> - aber das sollte die Pakete auch nicht anfassen.

CONFIG_PACKAGE_kmod-lib-textsearch=y ?

storchi, 2015-12-17:
Am 16.12.2015 um 17:53 schrieb Bastian Bittorf:
> @storchi <https://github.com/storchi> mit welchem tcpdump-befehl exakt
> siehst du die fehlermeldung "bad udp cksum"? auf dem server oder auf dem
> client?
sudo tcpdump -Xvvvni eth0 host 178.19.236.123 -w /tmp/1.txt
auf 3.v...

storchi, 2015-12-17:
Am 16.12.2015 um 16:10 schrieb Andi Bräu:
> du könntest mit dem Builder neu bauen und gucken ob es besser wird:
>
> |./build_release.sh meta.ffweimar-meshkit use_cc1505|
>
> der zieht automatisch den letzten commit von CC1505

der builder läuft nicht durch:

cs: 2015-12-17 10:45: [openwrt]: ./build_release.sh: [START] 
'patch:0001-add-procd-respawn-flag-to-rpcd-and-cron-init-scripts.patch' 
from 'use_cc1505 imagebuilder options ffweimar_standard 
patch:0001-add-procd-respawn-flag-to-rpcd-and-cron-init-scripts.patch 
patch:0002-backport-profile.d-support.patch 
patch:luci-remove-freifunk-firewall.patch ffweimar_luci_standard hostapd 
vtunnoZlibnoSSL i18n_german nohttps owm shrink tc busybox busybox_swap'
Prüfe Patch package/system/rpcd/files/rpcd.init...
error: bei der Suche nach:
start_service() {
	procd_open_instance
	procd_set_param command "$PROG"
	procd_close_instance
}

error: Anwendung des Patches fehlgeschlagen: 
package/system/rpcd/files/rpcd.init:9
error: package/system/rpcd/files/rpcd.init: Patch konnte nicht 
angewendet werden
Prüfe Patch package/utils/busybox/files/cron...
error: bei der Suche nach:

	procd_open_instance
	procd_set_param command "$PROG" -f -c /etc/crontabs -l ${loglevel:-5}
	procd_close_instance
}

error: Anwendung des Patches fehlgeschlagen: 
package/utils/busybox/files/cron:29
error: package/utils/busybox/files/cron: Patch konnte nicht angewendet 
werden
cs: 2015-12-17 10:45: [openwrt]: ./build_release.sh: [ERROR] 
patch:0001-add-procd-respawn-flag-to-rpcd-and-cron-init-scripts.patch 
failed. aborting

glycoknob, 2015-12-17:
Läuft hier. Ich glaube du musst jedesmal das `openwrt` Verzeichnis im Builder löschen... da fällt mir ein, dass man das auch besser lösen kann, ich bau das später mal ein.

bittorf, 2015-12-17:
@andibraeu patches lasse sich nicht 2mal anwenden, daher vermutlich die fehlermeldung

storchi, 2015-12-17:
Am 17.12.2015 um 11:04 schrieb Martin Tippmann:
> Läuft hier. Ich glaube du musst jedesmal das |openwrt| Verzeichnis im
> Builder löschen... da fällt mir ein, dass man das auch besser lösen
> kann, ich bau das später mal ein.

kann nicht sein. mein system war nackig, zumindest beim 1. Mal

storchi, 2015-12-17:
Am 17.12.2015 um 11:04 schrieb Martin Tippmann:
> Läuft hier. Ich glaube du musst jedesmal das |openwrt| Verzeichnis im
> Builder löschen... da fällt mir ein, dass man das auch besser lösen
> kann, ich bau das später mal ein.

läuft nicht. hab alle ordner gelöscht und neu versucht. endet so:

logger: apply symbol: CONFIG_BUSYBOX_CONFIG_KLOGD is not set
logger: apply symbol: CONFIG_BUSYBOX_CONFIG_LOGGER=y
cs: 2015-12-17 11:22: [openwrt]: ./build_release.sh: [READY] 'busybox' 
from 'use_cc1505 imagebuilder options ffweimar_standard 
patch:0001-add-procd-respawn-flag-to-rpcd-and-cron-init-scripts.patch 
patch:0002-backport-profile.d-support.patch 
patch:luci-remove-freifunk-firewall.patch ffweimar_luci_standard hostapd 
vtunnoZlibnoSSL i18n_german nohttps owm shrink tc busybox busybox_swap'
cs: 2015-12-17 11:22: [openwrt]: ./build_release.sh: [START] 
'busybox_swap' from 'use_cc1505 imagebuilder options ffweimar_standard 
patch:0001-add-procd-respawn-flag-to-rpcd-and-cron-init-scripts.patch 
patch:0002-backport-profile.d-support.patch 
patch:luci-remove-freifunk-firewall.patch ffweimar_luci_standard hostapd 
vtunnoZlibnoSSL i18n_german nohttps owm shrink tc busybox busybox_swap'
cs: set_build() using '../openwrt-config/.config'
logger: apply symbol: CONFIG_BUSYBOX_CONFIG_SWAPONOFF=y
logger: apply symbol: CONFIG_BUSYBOX_CONFIG_FEATURE_SWAPON_DISCARD is 
not set
logger: apply symbol: CONFIG_BUSYBOX_CONFIG_FEATURE_SWAPON_PRI is not set
logger: apply symbol: CONFIG_PACKAGE_zram-swap is not set
logger: apply symbol: CONFIG_PACKAGE_swap-utils is not set
cs: 2015-12-17 11:22: [openwrt]: ./build_release.sh: [READY] 
'busybox_swap' from 'use_cc1505 imagebuilder options ffweimar_standard 
patch:0001-add-procd-respawn-flag-to-rpcd-and-cron-init-scripts.patch 
patch:0002-backport-profile.d-support.patch 
patch:luci-remove-freifunk-firewall.patch ffweimar_luci_standard hostapd 
vtunnoZlibnoSSL i18n_german nohttps owm shrink tc busybox busybox_swap'
cs: 2015-12-17 11:22: [openwrt]: ./build_release.sh: [START] executing 
'make defconfig  '
Collecting package info: done
#
# configuration written to .config
#
cs: 2015-12-17 11:22: [openwrt]: ./build_release.sh: [READY] executing 
'make defconfig  '
sed: kann package/base-files/files/etc/init.d/apply_profile.code nicht 
lesen: Datei oder Verzeichnis nicht gefunden
sed: kann package/base-files/files/etc/init.d/apply_profile.code nicht 
lesen: Datei oder Verzeichnis nicht gefunden
sed: kann package/base-files/files/etc/init.d/apply_profile.code nicht 
lesen: Datei oder Verzeichnis nicht gefunden
cs: not found '../openwrt-config/config_HARDWARE.unknown_model.txt' we 
are here: '/home/cs/src/builder/openwrt'
cs: unknown filenames, search in 'bin/ar71xx/...' and read 
'openwrt-config/HowTo_add_new_HARDWARE.md'
executing: 'make -j5'
  make[1] world
  make[2] tools/install
  make[2] package/cleanup
  make[3] -C tools/patch compile
  make[3] -C tools/sstrip compile
  make[3] -C tools/make-ext4fs compile
  make[3] -C tools/firmware-utils compile
  make[3] -C tools/patch-image compile
  make[3] -C tools/flock compile
  make[3] -C tools/sstrip install
  make[3] -C tools/patch-image install
  make[3] -C tools/flock install
make -r world: build failed. Please re-run make with -j1 V=s to see 
what's going on
make: *** [world] Fehler 1

storchi, 2015-12-17:
hier der gesamte build versuch: 
[1.txt](https://github.com/weimarnetz/weimarnetz/files/65240/1.txt)

andibraeu, 2015-12-17:
hm, da musst du mal ins openwrt Verzeichnis gehen und dort ```make V=s``` machen

entweder fehlt etwas in deinen build requirements oder er kann was nicht downloaden

andibraeu, 2015-12-17:
kmob_lib_textsearch ist fürs speicher sparen rausgeflogen. Was macht das?

glycoknob, 2015-12-17:
@andibraeu könnte auch noch durch andere Abhängigkeiten drin sein. Ist ein Modul was textsuche ermöglicht, das wird vermutlich für das iptables-filter Paket genutzt d.h. ich kann Regeln bauen wo ich nach Strings in Daten suchen kann... haben wir nicht. 

Siehe: https://wiki.openwrt.org/doc/howto/netfilter

     iptables-mod-filter		15347	iptables extensions for packet content inspection. Includes: - ibipt_string

andibraeu, 2015-12-17:
steht das vielleicht auch in Zusammenhang mit #131 ?

storchi, 2015-12-17:
Am 17.12.2015 um 14:12 schrieb Andi Bräu:
> steht das vielleicht auch in Zusammenhang mit #131
> <https://github.com/weimarnetz/weimarnetz/issues/131> ?

kann ich mir nicht vorstellen.

jetzt mit ausem vtun locking auf 3.v. verbindet sich auch die 122 mit cc 
gm von montag.
die 319 schafft es nicht

storchi, 2015-12-17:
voraussetzungen des builders sind installiert
der builder verhält sich komisch.
jetzt baut er - hab nix anders gemacht
ich kanns ja am di nochmal zeigen

glycoknob, 2015-12-17:
@storchi: failt bei mir auch öfters, meistens weil irgendwas nicht runter geladen werden kann. Ich bastel an einen Update zum Builder der git branches zum Patchen nutzt, dann kann man auch updaten und neubauen ohne immer die ganze Toolchain neubauen zu müssen.

andibraeu, 2015-12-20:
heute sind 319 und 778 erfolgreich mit vpn2 verbunden... @kaese77 @storchi habt ihr was gemacht?

Unterschied auf dem Server: Tinc ist abgestürzt

andibraeu, 2015-12-21:
hab noch ne Idee: Ist das vielleicht einfach nur ein doofes Timingproblem? Die Verbindungen brechen immer wieder nach maximal 15 Sekunden ab. Ich vermute, dass da ein minütlicher Check reinhaut und feststellt, das OLSR noch nicht läuft... dann wird VPN gestoppt und das Spiel beginnt von neuem

kaese77, 2015-12-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Ich habe nur 319, wie befohlen, neu gestartet.

René

> heute sind 319 und 778 erfolgreich mit vpn2 verbunden... @kaese77 <https://github.com/kaese77> @storchi <https://github.com/storchi> habt
> ihr was gemacht?
> 
> Unterschied auf dem Server: Tinc ist abgestürzt

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWd8JZAAoJEJuS/5F7Le+BgVoIAJqEwCaPFjH48jHU/90JUBZ8
43jOSiumjvypN+D0qJpqTgBSwey/FaiGmmNPbMsNKMtzi+BQ6y/+AKK+19mpm9GH
keDFCSyG1Yq4IZ3ywBAgcSsvdUSbe3kxbaZc1dbc+5A9K/WInCtVijlBCIfAetZC
Ddh//MsfWhqB7UMm0xRTdlFPXQFEkziOX4UL8gs7aP0uN8opj2HAq4rn0Zclyu9H
YtpTNGc+ymxLBEUHlfdKq2fTVJEVdC83c9imaJbdPOai6HuDAKtF3EhFN2q3uehu
ALStDuT2oUVK4B/FDKrzTedl9xoP0bz7ivM+yDyZsQGShKSYvMYoC7i1jFtEgek=
=TzsJ
-----END PGP SIGNATURE-----

bittorf, 2015-12-21:
@andibraeu  das wuerde aber im log stehen...

andibraeu, 2015-12-21:
die vtun-Verbindung würde sich aber nicht einfach so beenden...

Und heute verbindet sich der Router von @kaese77 wieder nicht. Gleiches Phänomen...

@kaese77 hast du dyndns und kannst mir ggf. ein Portforwarding auf den SSH-Port des Routers einrichten?

kaese77, 2015-12-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

mom

> die vtun-Verbindung würde sich aber nicht einfach so beenden...
>
> Und heute verbindet sich der Router von @kaese77 <https://github.com/kaese77> wieder nicht. Gleiches Phänomen...
>
> @kaese77 [...]

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWd8WrAAoJEJuS/5F7Le+B0DwIAOogIfv7sk1DVPqDQqq4Pqrd
4FA0ywffrCjt+BYAyo3qaduN4JLgvWwu5V3TX1Ckpzgzh5pcgqS6L7fYO2nuprH9
3wg4msZs/xWkpDqnTrkcBh9pJRmT62Ur79+X4GefkfNACu5BKcqOTskyreojkT2P
D4uZONKw8rhl4/lv+A0eeCuF6zFHSVovr0Hnvt/TmboY6uEAd3CJSwrE5bWgFXAR
UgglFe1xViPMXSfF7MI7NH30QvEQNdlTlmsNQx2Pot63t0w+9Vrql7Gz6CjCJyMX
u9YBXwF5Ypu77ak3JAWB2/gXboaeZZ27OFNabMzKfuVog93TSl78hk6ZNCo5IUM=
=sxjP
-----END PGP SIGNATURE-----

kaese77, 2015-12-21:
Das geht nicht so einfach. Ich habe bei Vodafone Kabel nur eine NAT-IP.

PF geht somit nicht.

Man müsste einen VPN-Tunnel haben. ;-)

René

storchi, 2015-12-21:
Am 21.12.2015 um 10:37 schrieb kaese77:
> Das geht nicht so einfach. Ich habe bei Vodafone Kabel nur eine NAT-IP.
>
> PF geht somit nicht.
>
> Man müsste einen VPN-Tunnel haben. ;-)

Doofkanne.

Dann veröffentliche doch hier deine aktuelle ip, die sich es bei kd 
wenig ändert, plus das ssh passwort

kaese77, 2015-12-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Hast Du NAT nicht verstanden?

Leg Dich wieder hin!

René

>> Das geht nicht so einfach. Ich habe bei Vodafone Kabel nur eine NAT-IP.
>> 
>> PF geht somit nicht.
>> 
>> Man müsste einen VPN-Tunnel haben. ;-)
> 
> Doofkanne.
> 
> Dann veröffentliche doch hier deine aktuelle ip, die sich es bei kd wenig ändert, plus das ssh passwort

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWd87DAAoJEJuS/5F7Le+BQJYH/jWAkmpR1zrMEe9iWmxlt/87
3zHnUCj0kTy1XyWYwFllf0xEAJisM0+BSbc5fNP+P9pMsZyDjmSqoe2t9xsWSn7q
xgBrXoJcGZQGSKOSlTy4NEaIIOfm4bWi2c3/31pFXubLz8uZVgUEZO5G2ZZcV3Cd
IuEiXkEJDY8dFfxLQQmqHRbqbldUD6z8QJ+1+WB1lv4Jd/AokhCtGLQItOEdfgGD
r2JtRaUGtIV7MjmUAY7nTfcnF0njJDfLwF4LiQdORIdBS2gv0XX3IkCGx8R2gFN+
FuMcWTevbp6UKRcNyCbn7fvywG2/+lSaiAiyXWunf7pwzUtMpgXm0EszDk3WRdA=
=aJTi
-----END PGP SIGNATURE-----

andibraeu, 2015-12-21:
@storchi  René hat so einen armseligen Internetzugang, dass er nicht mal eine öffentliche IP hat... Frechheit eigentlich dass die das als Internet verkaufen

storchi, 2015-12-21:
muaha. zum beschiss gehören immernoch zwei

kaese77, 2015-12-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

> @storchi <https://github.com/storchi> René hat so einen armseligen Internetzugang, dass er nicht mal eine öffentliche IP hat... Frechheit 
> eigentlich dass die das als Internet verkaufen

Das ist Business-Kabel-DSL für 45 Euronen / Monat!

René

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWd9EDAAoJEJuS/5F7Le+Bgo8H/2le5KDD9yQKczmFxk4dhLDj
R1Deni4z+mt74smZpfQAZ2aNAvFrhtLTxRua/PWcLPngY8mnpUm1rrHSSUBHufzs
sWEXvhCkj/iq1C3pt0gayM5xhGpiTSBPDJDDHeAbnXEqAGdkfjkNaEAi0CnUvG2I
a9H994xK4BD/85KiB6m/7shBm1oC4NPE0b39wd6WshY9WcdfMru0HNZTDH5V3rKS
mA9MjBJzwhCHwxTUkcz6sVv9eTGGwfrxjp72DeYEkhKB+g04pWcOn8E2lZVkVxax
fbEcUAduQXLQIxNtM3uAJpmmZtah94b2dxqbLcm2vvNoKzr7TVht0pmYoFVCkIs=
=9xTa
-----END PGP SIGNATURE-----

kaese77, 2015-12-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

150MBit/s peak Downstream, Junge! Dauwnschtriem!

> muaha. zum beschiss gehören immernoch zwei

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWd9E3AAoJEJuS/5F7Le+BOFIH/iAH89IaPLkm8ePKGcFfWBua
HAWQAgsi9mSEJt1MiDoKTCxkVNk+4PK58lSCyOFvMzYFc5I8TMC7RPFkGDIHxl/z
BxRKfCMNrzwZ+oQnuUNX27f+oD5usrfsWwE32tkTeeJQl3NpvcNJhDU5DAj0eu+5
dGuMrhj0n+mztS7jtYStjLfPUGXpCRREUv40zKwy6twYrMxWt9jGOaqMIAXrt7wH
iuiUX3pEu8NSHmW5Y6HoFGPkwP+YfCwOibeT29JHvH0rYtGEeVgRcMsxmVDbxGKp
/aIYI6T26u7ZGgVBbrh1XnvRZYi4Olo1JGWiOIh8gkwUxPHHJEsmbsohhxsCSig=
=F6am
-----END PGP SIGNATURE-----

storchi, 2015-12-21:
n alten arsch. koof dich mal ein internetanschluss.
ham die keine feste ip für dich? oder können die ein pf fuer dich machen?

kaese77, 2015-12-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Nein, das sind Wessis. Die geben auch Business-Kunden keine eigene IP-Adresse.

René

> n alten arsch. koof dich mal ein internetanschluss. ham die keine feste ip für dich? oder können die ein pf fuer dich machen?

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWd9NOAAoJEJuS/5F7Le+BJ2EIAOdiMd7eoxP+LVunaYyjyiOS
8EpfHVnDa73OqwPKd6aqtZluypPnfDwt3hdBQk+guap7d+MzfY4+PdrCW0Rc+zEs
iTdOdpXbK581vZWfh9unv+pzF0afFaSVpWYQxtJ3ypIZU+OlYmI0MCyXrfHIP92u
NdcSaM6Y162zHPasX5kuPnEFNWdoj0WFNU4lhISHChdLZ1FGS37vUWyJ9ni27WY2
YqSkRETvFMl6hlILPlTMmbsK8AAgmUKqtM1/dUTty/NrzGZDPRmtzmDFrTjQwIJO
NJ80iP0qKVBP1zYkYkZ9LKQg4v6zsvsDjIRIaB5iP4WWlQZ/xoF3ID2q+E9BC1k=
=X9c6
-----END PGP SIGNATURE-----

storchi, 2015-12-21:
und das stand da nirgends?
ipv6?

glycoknob, 2015-12-21:
Doppel-NAT _könnte_ ein Problem sein.... IPv6 klappt glaub ich nicht, da fehlen eventuell Patches im vTUN und auf dem Port hört auch kein vTUN auf v6 

Im Prinzip sagt aber der Server dem Client verbinde dich mal hier mit diesen Port bei mir und das klappt auch durch Doppel-NAT - sonst würde meine KVM Kiste auch kein VPN bekommen. 

Problem ist also irgendwo anders, kann aber auch sein das er probiert sich mit der v6 Adresse zu verbinden und dann Connection refused bekommt. ich glaub `getaddrinfo()` wechselt das... 

@kaese77: bekommt der eine v6 IP für WAN? also was sagt `ip` auf dem Router? 

Falls das so ist, dann klappt es in 50% der Fälle nicht: 

```
$ dig aaaa 3.v.weimarnetz.de                                                                                                                        ; <<>> DiG 9.9.5-11ubuntu1.1-Ubuntu <<>> aaaa 3.v.weimarnetz.de
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 20568
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;3.v.weimarnetz.de.             IN      AAAA

;; ANSWER SECTION:
3.v.weimarnetz.de.      17895   IN      AAAA    2001:bf7:b101:1::35

;; Query time: 36 msec
;; SERVER: 127.0.1.1#53(127.0.1.1)
;; WHEN: Mon Dec 21 14:12:53 CET 2015
;; MSG SIZE  rcvd: 74

$ nc -6 2001:bf7:b101:1::35 5001    
# nüx                                                                                                               
$ nc 3.v.weimarnetz.de 5001                                                                                                                         
VTUN server ver 3.X 06/16/2015
^C   
```

glycoknob, 2015-12-21:
Okay... kann nicht sein, vtun bekommt ne eine v6 Adresse... muss was anderes sein. Die Funktion, die vtun nutzt kann gar kein IPv6

andibraeu, 2015-12-21:
hab nochmal auf die Serverlogs gestarrt:

```vpn2 vtund[5854]: Connection closed by other side```

Der Router schließt also die Verbindung, deswegen müssen wir da mal draufgucken und beobachten können

glycoknob, 2015-12-21:
Okay... Also ein ungünstig getimtes `_watch vpn` könnte das schon verursachen. Das wird von _verschiedenen_ Scripten ausgeführt u.A. `/etc/init.d/S69watch_olsrd_fff+` sowie `/usr/sbin/cron.check_vtun_status` sowie hier: 

```
~ cat /etc/crontabs/root 
* * * * * . /tmp/CRON_IS_CAREFUL; . /tmp/loader; _watch switch_linkstatus
```

Fehlt bestimmt auch noch was, ist auch alles noch load abhängig und random sleeps sind auch überall drin... kein plan 

`/tmp/OLSR/ALL` muss da sein, die VPN-IP muss olsr Nachbar sein und vtun darf sich gerade nicht connecten - sonst `_vpn restart` - irgendwelche races und timing probleme sind da garantiert irgendwo - aber schwer zu sagen ob es wirklich daran liegt.

[43 more replies]

closed 2015-12-22 by glycoknob