weimarnetz/weimarnetz #220: Splash usabilty

glycoknob, 2015-12-15:
Live aus dem Falken mit nicht weimarnetz, nicht nerd Menschen getestet. Ist kaputt.

glycoknob, 2015-12-15:
Also könnte vieel  einfacher sein. Ist das häkchen wirklich mandatory?

andibraeu, 2015-12-15:
Der Nutzer muss halt den Haken aktiv setzen...

glycoknob, 2015-12-15:
Eine einfachere Seite ohne Menü oben mit einigen klaren Sätzen und Links und einem Großen Knopf fänd ich schön - nach kurzem wilden Googlen könnte das auch mit den AGB klargehen: http://www.behrmannhaertel.de/2012/12/13/checkbox-bei-agb-bestatigung-oder-kein-hakchen-kein-problem/ 

Oder wir werfen CSS drauf - z.B. bei der bösen kommerziellen Konkurrenz sieht's so aus:  https://airfy.com/assets/new/how/demo/en/screen-wifi-1-0fcd3bd559152dc014a9a3a8281d58bd.jpg 

Ich würd mal was bauen.

andibraeu, 2015-12-15:
Eine einfachere Seite ohne Menü oben mit einigen klaren Sätzen und Links und einem Großen Knopf fänd ich schön - nach kurzem wilden Googlen könnte das auch mit den AGB klargehen: http://www.behrmannhaertel.de/2012/12/13/checkbox-bei-agb-bestatigung-oder-kein-hakchen-kein-problem/@eins78 hatte schonmal ne Idee dazu, damit die Leute mit ihren dicken Fingern das alles treffen können. Das Menü sollte meiner Meinung nach drin bleiben. Die Seite soll schließlich auch ganz normal aufgerufen werden könnenOder wir werfen CSS drauf - z.B. bei der bösen kommerziellen Konkurrenz sieht's so aus: https://airfy.com/assets/new/how/demo/en/screen-wifi-1-0fcd3bd559152dc014a9a3a8281d58bd.jpgbin wenn dann für ein paar CSS-Häcks, z.b. auf kleinen Geräten die Infoboxen ausblenden, weil man im Captive Portal nur schwer drauf drücken will. Auf Desktops sollten die Infos schon lesbar sein. Aber ich nehm mal an, du hast das auch nur mobil betrachtet.Vorher sollten allerdings die dringlicheren Probleme bearbeitet werden. Viele Nutzer schaffen es ja, die haben wahrscheinlich nur nicht so viel Bier im Falken getrunken.

eins78, 2015-12-15:
hier sind die Skizzen von damals™:

extrem cool were so, aber rechtlich geht es eher nicht:
[image]

das war dann der Kompromiss:
[image]

technischer Hintergrund:
- die checkbox ist in einem `label` gewrappt – browser handeln das so das ein klick aufs label auf die checkbox aktiviert (einziger weg ein grösseres click-target zu bekommen)
- mit ein bissl gebastelt lässt sich der bootstrap-button-style auf das label anwenden.
- zusätzlich kann man dann noch mit dem CSS tricksen (wie airfy), und statt der checkbox ein dem status der checkbox entsprechendes icon anzeigen.

eins78, 2015-12-15:
@glycoknob Kannst du eine aktuelle Splashpage aufrufen und mir den Quelltext speichern und schicken? Dann könnte ich was statisches basteln.

andibraeu, 2015-12-15:
@eins78 http://home.andi95.de:3380/cgi-bin/luci/freifunk/captive

FreifunkUFO, 2016-01-13:
Also, mit Blick auf die vielen Splash-Seiten, die von gratis-WLANs angeboten werden bezweifle ich, dass der Haken wirklich sein muss :-o 
Wäre denn auch ein Button "nicht akzeptieren (Freifunk ohne Internet starten)" sinnvoll/machbar? Ein Texteingabefeld wäre auch toll (für Interaktionen der Freifunkenden untereinander bzw. mit dem Freifunk-Node)
@eins78 was ist eigentlich ein "Zugriff auf Meldungen und Diskussionen"? klingt ominös! Als Tourist würde ich hier zb an Facebook denken. ("Hat Weimarnetz Zugriff auf meine Timeline?")

andibraeu, 2016-01-13:
@FreifunkUFO das ist noch ein Satz aus alten Zeiten, als tatsächlich nichts ging bevor man sich freigeschaltet hat

mtippmann, 2018-01-16:
Splash ist Geschichte.

closed 2018-01-16 by mtippmann