weimarnetz/weimarnetz #221: VPN Server - locking ist zu allg.

storchi, 2015-12-15:
ein "defekter" vpn client lockt den Server und kein weiterer client kann sich verbinden. das locking wäre besser, wenn es nur den einen client beträfe.
cs

bittorf, 2015-12-16:
das kann ich leider nicht beheben, die beiden vpn-dateien unterscheiden sich zu stark...

andibraeu, 2015-12-17:
auf den Server läuft Kalua, es reicht, wenn du es dort anpasst

bittorf, 2015-12-18:
@andibraeu das ist bei mir schon geaendert, also im prinzip "nur" meine /etc/kalua/vpn-datei nehmen alle vorkommen von connection_name="n$NODENUMBER" durch "Node$NODENUMBER" ersetzen.

andibraeu, 2015-12-23:
storchi hat das mal komplett deaktiviert und es läuft damit seit 7 tagen gut

storchi, 2015-12-23:
Auf vpn3 ist es deaktiviert

storchi, 2016-01-20:
vpn2 hat auch kein locking mehr.
kalua ist angepasst

closed 2016-01-20 by storchi