weimarnetz/weimarnetz #224: vpn: workaround für Netze in den Ping NICHT GEHT!!!!

storchi, 2015-12-22:
muss man sich mal vorstellen, sowas.
vpn Verbindung kommt nicht zu stande.

bittorf, 2015-12-22:
vielleicht einfach nur stur testen, ob ein verbindungsaufbau geht?:
```
_net tcp_port_reachable "$server" "$port" 'VTUN server' && echo OK
```

andibraeu, 2015-12-22:
vor allem dem Netzwerkadmin auf die Finger hauen...

glycoknob, 2015-12-22:
Wer sperrt denn Ping ausgehend? Das macht alles kaputt. Vor allen wenn auch Path-MTU gesperrt ist.. 

Jedenfalls: Eigentlich ist der Ping in der Firmware seit dem zufälligen Auswählen nur ein Gimmick und kann auch raus. Wir testen ja auch direkt den Port.

andibraeu, 2015-12-22:
richtig.. wie funktioniert im olsr das mit dem dynamischen gateway? d.h. HNA für Internet wird rausgenommen, wenn kein Internet mehr da ist? ist das nicht auch mit ping?

glycoknob, 2015-12-22:
@andibraeu laut Doku schaut der nur auf die Routing Tabelle: http://www.olsr.org/docs/report_html/node140.html

PR #225 (2015-12-22): remove ping from vpn check
fixes #224

ping is still in the logs but it's not mandatory for a successful connection.

special thanks to the braindead network guy who did this.
commit ca1f274f (2015-12-22): remove ping from vpn check

fixes #224
PR #230 (2015-12-23): remove ping from vpn check
fixes #224
closed 2015-12-22 by bittorf