weimarnetz/weimarnetz #227: Wir hijacken DNS

glycoknob, 2015-12-22:
Wir zwingen alle Nutzer den DNS vom Router zu nutzen, auch wenn sie explizit einen anderen angeben. 

Wie schlimm das ist, darüber kann man sich streiten - aber: 
- DNS Requests gehen lokal (über 8.8.8.8) raus -> ein Nutzer leaked also seine Requests - find ich persönlich nicht so schlimm
- Wenn jemand explizit eigene DNS-Server angibt wird das ignoriert und der vom Router genommen. 

Der Redirect Ist eigentlich unnötig, DNS über Tunnel muss nicht sein, für Splash reicht es wenn Port 80 redirected wird. 

Verstößt gegen Pico-Peering Agreement IMHO: 
> Der Eigentümer bestätigt, die Daten, die seine freie Netzwerkinfrastruktur passieren, weder störend zu beeinträchtigen noch zu verändern.

Ich kann z.B. als Client am Freifunk keinen Tunnel zu Port 53 aufbauen. 

Korinthenkackerei? Vielleicht.. aber wir brauchen das eigentlich nicht, also warum nicht abschalten? Der Router-DNS kommt eh via DHCP und wenn jemand umbedingt was anderes machen will, so sollten wir ihn das machen lassen. 

Siehe Netalizer Report: 

http://n1.netalyzr.icsi.berkeley.edu/summary/id=43ca253f-16489-088eebfa-ca5d-4245-ad3b#udp_connectivity

Verantwortlich sind vermutlich folgende Regeln: 

```
root@alte-feuerwache--dachstuhl:~ iptables -L -n -v -t nat
Chain PREROUTING (policy ACCEPT 20229 packets, 2240K bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 DNAT       udp  --  wlan0  *       0.0.0.0/0            0.0.0.0/0            mark match 0x13/0xff to:10.63.186.65:53
    0     0 DNAT       tcp  --  wlan0  *       0.0.0.0/0            0.0.0.0/0            mark match 0x14/0xff to:10.63.186.65:53
    0     0 DNAT       tcp  --  wlan0  *       0.0.0.0/0            0.0.0.0/0            mark match 0x11/0xff to:10.63.186.65:80
    0     0 DNAT       udp  --  wlan0-1 *       0.0.0.0/0            0.0.0.0/0            mark match 0x13/0xff to:10.63.186.66:53
    0     0 DNAT       tcp  --  wlan0-1 *       0.0.0.0/0            0.0.0.0/0            mark match 0x14/0xff to:10.63.186.66:53
   91  5460 DNAT       tcp  --  wlan0-1 *       0.0.0.0/0            0.0.0.0/0            mark match 0x11/0xff to:10.63.186.66:80
 1371 94695 DNAT       udp  --  eth1   *       0.0.0.0/0            0.0.0.0/0            mark match 0x13/0xff to:10.63.186.97:53
    2   120 DNAT       tcp  --  eth1   *       0.0.0.0/0            0.0.0.0/0            mark match 0x14/0xff to:10.63.186.97:53
    0     0 DNAT       tcp  --  eth1   *       0.0.0.0/0            0.0.0.0/0            mark match 0x11/0xff to:10.63.186.97:80
    0     0 REDIRECT   udp  --  eth+   *       0.0.0.0/0            0.0.0.0/0            mark match 0x13/0xff redir ports 53
    0     0 REDIRECT   tcp  --  eth+   *       0.0.0.0/0            0.0.0.0/0            mark match 0x14/0xff redir ports 53
    0     0 REDIRECT   tcp  --  eth+   *       0.0.0.0/0            0.0.0.0/0            mark match 0x11/0xff redir ports 80
    0     0 REDIRECT   udp  --  wlan+  *       0.0.0.0/0            0.0.0.0/0            mark match 0x13/0xff redir ports 53
    0     0 REDIRECT   tcp  --  wlan+  *       0.0.0.0/0            0.0.0.0/0            mark match 0x14/0xff redir ports 53
    0     0 REDIRECT   tcp  --  wlan+  *       0.0.0.0/0            0.0.0.0/0            mark match 0x11/0xff redir ports 80
    0     0 REDIRECT   udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x13/0xff redir ports 53
    0     0 REDIRECT   tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x14/0xff redir ports 53
    0     0 REDIRECT   tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x11/0xff redir ports 80

Chain INPUT (policy ACCEPT 13159 packets, 1231K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain OUTPUT (policy ACCEPT 3473 packets, 269K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain POSTROUTING (policy ACCEPT 6140 packets, 399K bytes)
 pkts bytes target     prot opt in     out     source               destination         
 1905  154K MASQUERADE  all  --  *      eth0    0.0.0.0/0           !192.168.2.0/24 
```

andibraeu, 2015-12-22:
da geb ich dir recht!

storchi, 2015-12-23:
Bin für post gm

andibraeu, 2015-12-23:
@glycoknob würde das reichen?

bittorf, 2015-12-23:
der fix ist einfach, schauts euch an.

storchi, 2015-12-23:
und eure beiden varianten machen das gleiche? wäre ja unglaubbar
cs

andibraeu, 2015-12-23:
ich bin dafür, diesen DNS-Redirect ganz rauszuwerfen. Warum muss man das konfigurierbar machen?

Das hat in einer Freifunk-Firmware nichts zu suchen, siehe Martins Anmerkung zum PPA.

bittorf, 2015-12-23:
ich habe festgestellt das es ein haufen fehlerhafter geraete gibt bei deinen ein alter/kaputter DNS eingestellt ist und umleuten auf den router hilft da. PPA ist so eine sache - im Prinzip muesste auch die splash-page weg.

@storchi der patch macht was aehnliches, meiner ist halt ueberschaubarer/kleiner 8-)

glycoknob, 2015-12-23:
Wie gesagt - sehe das nicht als Weltuntergang - ist mir aufgefallen wegen Netalizer.

andibraeu, 2015-12-23:
dann ist das aber auch nur ein blöder Workaround, der den eigentlichen Fehler überdeckt und unter Umständen zu langen Fehlersuchen führt. Es geht dabei ja um Endgeräte (welche eigentlich?), dann sollte der Fehler auch dort behoben sein. Wenn das so problematisch wäre, würde das ja nicht nur bei uns gemacht werden.

Ich kenne das nur aus Unternehmen, wo Nutzer überwacht oder unliebige DNS-Einträge weggefiltert werden sollen. Anderswo haben solche Softwarestücke schon zu mächtigen Shitstorms geführt

bittorf, 2015-12-23:
das ist leider in der realität anders: leute wissen nicht, wie das geaendert wird oder haben garnicht die admin-rechte das zu aendern...wie gesagt: erfahrung aus der echten welt.

andibraeu, 2015-12-23:
Die Erfahrungen aus der echten Welt habe ich auch. Und da sind mir solche Probleme noch nicht begegnet. 99% werden sowieso den per DHCP zugewiesenen DNS nutzen.

Meine beliebte Gegenfrage: Warum geht das auf 25000 Freifunkroutern in ganz Deutschland ohne diesen dreckigen Hack und wir brauchen das für unsere 100 Knoten? Wenn es ein Problem wäre müssten die Mailinglisten und Foren voll von Meldungen sein.

glycoknob, 2015-12-24:
Moin, der pull request löst das doch und der  Knopf (default: aus) stört
nicht oder?

andibraeu, 2015-12-24:
mein Pull Request löst das auch :-) Warum nicht #228 nehmen?

#231 hat übrigens noch viele andere Sachen mit im Gepäck und ist an master gerichtet

bittorf, 2015-12-24:
@andibraeu ein nutzer bei dem es nicht geht wird sich wohl kaum beschweren. es ist kein so seltener fall, das ein DHCP gemacht wird, aber DNS fest eingetragen ist. Das ist auch ok, solange der DNS funktioniert...aber das ist nicht immer so und ist nur "schulterzucken".

andibraeu, 2015-12-25:
ich finde, das hilft den Nutzern in keinster Weise, wenn wir ihre Fehler "heilen". Im Gegenteil, das macht die Fehlersuche am Ende nur komplizierter "in meinem Netz gehts, in deinem nicht - also ist dein Netz kaputt".

Das SSID-Umbenennen-Feature wird nicht ausnahmslos gut gefunden, sondern sehr kontrovers diskutiert. Ich finde das _nicht_ gut, weil es zementiert ja nur "Freifunk ist nichts außer Internet". Nach unserem Selbstverständnis wollen wir ein Netz zum experimentieren sein und ausdrücklich kein Internet Service Provider. Außerdem ist Internet Glückssache :-)

glycoknob, 2015-12-26:
Hab nochmal versucht drüber nachzudenken: Wenn z.B. die Telekom das machen würde, wäre der Aufschrei riesig. Dann würden wir beim Treffen drüber lachen. Zwangs DNS passt nicht zu Freifunk. Einen Knopf dafür zu haben, weil Konfiguration bei xyz komisch ist... warum nicht. Mir ging es nur darum, dass es per Default off ist. 

SSID: Lasst mal beim Treffen drüber sprechen. Bin da auch eher bei bei @andibraeu aber in 99% der Fälle ist halt ohne VPN/Internet auch nix Freifunk internes da zum experimentieren...

storchi, 2015-12-26:
Bin für natives dns und ne angepasste splashpage wenn kein internet da ist

bittorf, 2015-12-29:
gerade wenn das internet NICHT da ist, braucht man einen fakeDNS - sonst kommt kein splash.
(weil schon DNS fehlschlaegt) ich bin dafuer erstmal den simplen 3zeiler-patch zu nehmen und das dann fuer das naechste release schoenzumachen.

andibraeu, 2015-12-29:
naja, es wird für die funktionieren, die DNS per DHCP bekommen, nämlich den vom Router. Der Rest sieht bei "keinem Internet" eben nichts, aber das sehe ich nicht als unser Problem an :beers:

glycoknob, 2015-12-30:
@bittorf: Ich glaube so ein richtiges fakeDNS wie in kalua haben wir nich in der Firmware... der Router gibt dann auch SRVFAIL oder so zurück, wenn er kein Netz hat. Ist also momentan auch schon kaputt. Der fakeDNS für kaputte Kisten wäre ne gute Idee... mir geht's nur darum, dass _nach_ dem Splash-Klick alles läuft. Vorher können wir da IMHO auch rum manipulieren wie wir wollen.

bittorf, 2015-12-30:
ich wuerde genau deswegen den code auch erstmal drinlassen. der patch deaktiviert es und es koennte pro client wieder angeschaltet werden...(z.b. wenn internet kaputt)

commit 512a8b97 (2016-01-09): kalua: netfilter: make dns-spoofing configurable, default = off - fixes #227
closed 2016-01-09 by bittorf