weimarnetz/weimarnetz #232: richtiges IPv6 überall

glycoknob, 2015-12-26:
Meine Mutti hat richtiges IPv6 auf'n Dorf in Brandenburg. 

Wir haben das nicht. Da hängen echt ein paar Probleme dran, aber so eine mittelfristige Lösung, dass man auch als Client am AP natives v6 bekommt wäre schon fein. 

Der Berg an Problemen: 
- machen wir ULA mit Prefix-Deligation (wie sieht das dann mit VPN aus?) 
- Was ist mit Knoten, die nur via OLSR angebunden sind? Kann OLSR/OLSRv2 Präfixe auch an anderen Knoten deligieren? 
- Wie bekommt der Client seine IP? SLAAC (muss dafür der Router ein /56 haben? DHCPv6?) 
- Meshing via v6 - OLSRv2? 
- Wie mit v4 umgehen? DS-Lite? 
- VPN - vtun kann ohne Patch nix mit v6 für eine VPN-Verbindung anfangen, tunneldigger auch noch nicht. 
- wie verteilen wir die Adressen? 
- wie klappt alles noch ohne Internet (ULA?) 
- etc.pp 

Wir müssten da mal kieken was geht, wenn die Lösung OLSRv2 und inkompatibles hartes Update heißt, dann wäre das mir zur Not auch genehm... Router können wir auch updaten, so viele sind es ja nicht. 

Das Ticket hier ist eher als Brainstorming gedacht... ich sehe momentan nicht wie wir das in die vorhandene Infrastruktur irgendwie einbauen könnten.

andibraeu, 2015-12-26:
übrigens.. wir haben schon ne potentielle ipv6-config in _ipsystem, die devices werden bloß noch nicht konfiguriert: 
```
root@BraeuSys:~ _ipsystem ffweimar
NODE_NUMBER=52
NODE_NUMBER_RANDOM=false
OLSR_HNA=10.63.52.0/26

LAN_ADR=10.63.52.33
LAN_IP=10.63.52.33/28
LAN_IP_1ST=10.63.52.34/28
LAN_PRE=28
LAN_DHCP_PRE=28
LAN_DHCP_NUM=13
LAN_DHCP_LEASETIME=12h
LANADR=10.63.52.33
LAN1ST=10.63.52.34
LANMSK=255.255.255.240
LAN6ADR=2001:bf7:1930:3400::
LAN6MSK=64
LANBRC=10.63.52.47
LANNET=10.63.52.32
LANPRE=28
LANDHCPSTART=34
LANDHCPNUM=13

WIFI_ADR=10.63.52.1
WIFI_IP=10.63.52.1/8
WIFI_IP_1ST=10.63.52.2
WIFI_PRE=8
WIFI_DHCP_PRE=27
WIFI_DHCP_NUM=29
WIFI_DHCP_LEASETIME=30m

DHCP_LAN="10.63.52.34/28,13,12h"
DHCP_WIFI="10.63.52.2/27,29,30m"

FFF_CITY=63
FFF_NET=10
FFF_NODE=52
FFF_HNA=10.63.52.0/26

IPV6NET=2001:bf7:1930:34::
IPV6MSK=56

WIFIADR=10.63.52.1
WIFI1ST=10.63.52.2
WIFIMSK=255.255.255.224
WIFI6ADR1=2001:bf7:1930:3401::
WIFI6MSK1=64
WIFI6ADR2=2001:bf7:1930:3402::
WIFI6MSK2=64
WIFIBRC=10.63.52.31
WIFINET=10.63.52.0
WIFIPRE=27

WANADR=10.63.52.61
WAN1ST=10.63.52.62
WANMSK=255.255.255.252
WANBRC=10.63.52.63
WANNET=10.63.52.60
WANPRE=30
WAN_ADR=10.63.52.61
WAN_PRE=30
WAN_IP=10.63.52.61/30

WIFIDHCPSTART=10.63.52.2
WIFIDHCPNUM=29

WIFIVPNNET=10.63.52.56
WIFIVPNSERVERADR=10.63.52.57
WIFIVPNCLIENTADR=10.63.52.58
WIFIVPNBRC=10.63.52.59
WIFIVPNPRE=30

VPN6ADR=2001:bf7:1930:34ff::
VPN6MSK=64

ROAMING_IP4=172.31.63.10
ROAMING_PRE=24
```

Das alles ergibt sich direkt aus der Knotennummer, find ich ganz praktisch. Hier hatte ich das mal aufgeschrieben: http://wireless.subsignal.org/index.php?title=IP-System#Gedanken_zu_IPv6

Ich hatte sogar mal ne Beispielstrecke mit olsr und ipv6 gebaut und konnte so einen Router hinterm NAT direkt aus dem großen weiten (ipv6)-Internet erreichen.

storchi, 2015-12-29:
wenn dann wünsche ich mir eine Lösung ohne Knotennummer und damit auch ohne registrator
cs

bittorf, 2015-12-29:
@storchi der wunsch allein bringt nix voran. Konkrete Vorschläge bitte, sonst wünsch ich mir Weltfrieden...

storchi, 2015-12-30:
dann hast du ein zusammen arbeiten nicht verstanden

glycoknob, 2015-12-30:
@storchi @bittorf knotennummer von ip-konfiguration entkoppeln fänd ich auch cool. müssen wir mal schauen, ich sehe da noch nicht so durch. @andibraeu: das sieht gut aus, aber würde uns fest auf den /44 von freifunk festzerren oder?

glycoknob, 2015-12-30:
Konfigurationslos könnte es so klappen: 

Für das Mesh: OLSRv2 mit ULA Adressen (bräuchten einen gemeinsamen Prefix) - Knoten sehen sich intern automatisch und meshen über IPv6 link-local  - das wäre der Mesh-Part 

Wie kommen die Router und Clients an v6 Adressen? 

Mal shncpd testen: 
http://permalink.gmane.org/gmane.network.routing.babel.user/2134 

Die Idee ist das z.B. die VPN-Server ein v6 Prefix über shncpd announcen z.B. ein Teil vom /44 auf dem Weimarnetz-Server oder für andere VPNs ein /64 oder so... shncpd kümmert sich darum, dass dieser Prefix im Mesh verteilt wird - wenn ein Knoten kein VPN hat aber gerne seinen v6 Prefix announcen will ginge das auch. 

D.h. jeder Router auch, wenn er nicht direkt am VPN hängt bekommt einen Prefix und kann diesen via DHCPv6 wieder weiter verteilen... für v4 klappt das auch - könnte man mit dem 10.64.0.0/16 Netz testen. 

Hier nochmal meine Liste und was damit gelöst wäre (erstmal nur v6 only): 

Der Berg an Problemen:

> machen wir ULA mit Prefix-Deligation (wie sieht das dann mit VPN aus?)

shncpd to the rescue! 

> Was ist mit Knoten, die nur via OLSR angebunden sind? Kann OLSR/OLSRv2 Präfixe auch an anderen Knoten deligieren?

shncpd to the rescue! 

> Wie bekommt der Client seine IP? SLAAC (muss dafür der Router ein /56 haben? DHCPv6?)
Meshing via v6 - OLSRv2?

Wenn der Router ein Prefix hat via DHCPv6 

> Wie mit v4 umgehen? DS-Lite?

Noch unklar. 

> VPN - vtun kann ohne Patch nix mit v6 für eine VPN-Verbindung anfangen, tunneldigger auch noch nicht.

solange wie v6 im Tunnel läuft ist alles fein. 

> wie verteilen wir die Adressen?

ULA+globaler Prefix via shncpd! 

> wie klappt alles noch ohne Internet (ULA?)

Dann nur link-local meshing mit OLSRv2 - klappt auch. Wenn ein Router an der Wolke ist, lernt er die Prefixe und kann die verteilen. 

> etc.pp

Mal testen. Schön wäre es ja, wenn wir das einfach nur auf einen Router knallen und alles läuft automatisch.

bittorf, 2015-12-30:
@storchi ich meine das ernst. es bringt nix einfach so was hinzurotzen und wieder zu gehen. versuche wenigstens grob zu umreissen wie du deine wunsch umsetzen willst.

glycoknob, 2015-12-30:
Also da geht wohl in Rostock was von der Opennet-Initiative - sehe z.B. die Diskussion hier: http://thread.gmane.org/gmane.network.olsr.user/6358

bittorf, 2015-12-30:
@glycoknob es sagt aber ganz klar, das 'shncpd' auf keinen fall geht. warum kann man nicht ganz "normal" einen fixen meshweiten ULA-prefix haben?

glycoknob, 2015-12-30:
@bittorf: So wie ich es verstanden habe ist ULA sowieso da... olsrv2 mesht via link-local und setzt auch entsprechend die routen um andere ULA Adressen zu erreichen... das läuft also alles. 

Das einzige was shncpd macht wäre einen _globalen_ Prefix also ein Teil des öffentliche v6 Netzes im Mesh zu verteilen... d.h der Router hat sein normales ULA, link-local etc.pp aber auch nen Stück vom ipv6 prefix also z.B. ein /64 vom Weimarnetz VPN - damit bekommen alle User eine oder mehre native öffentlich routbare v6 Adressen - könnten auch durchaus mehrere sein. 
 
Vermutlich würde es auch klappen, wenn irgendwo ein Gateway announced wird und das Netz dort rausgeht... !?

glycoknob, 2015-12-31:
Vielleicht können wir ja das mal gemeinsam beim Treffen durchkauen... ist ziemlich komplex, aber könnte mir vorstellen, dass wir mit wenig Aufwand ein vollautomatisches Mesh hätten.. kann ja erstmal nur Spielerei sein, aber wenn man mal schnell was vernetzen muss wäre das ziemlich cool, wenn man eine Firmware auf 20 Router spielt und irgendwo hinwerfen kann und es läuft.

glycoknob, 2015-12-31:
Nur mit ULA würds wohl auch klappen, dann müssten die VPN-Server oder der Router NAT66 oder NPTv6 machen... hier ist das erklärt: http://www.howfunky.com/2013/12/ipv6-nat66-and-nptv6-it-seems-there-is.html 

Vorteil von shncpd wäre, dass du direktes v6 Internet global routbar direkt an jeder Kiste hast. 

VPN (/44 davon sagen wir mal /56 für Freifunk) <---> Router (/64) <-- OLSRv2 + shncpd --> Router 2 (anderes /64) <--> AP-Client (/80 oder so) 

Will eine File mit dir Teilen? Kannst _direkt_ über v6 auf den IP-Client zugreifen. Kein NAT, keine Knotennummern, alles automatisch :)

glycoknob, 2015-12-31:
Da geht einiges - wenn alle Clients die link-local v6 Adresse als Gateway haben klappt sogar Roaming (so wie jetzt in kalua, wenn man noch macvtap für die wifi mac einbaut - siehe:  https://tools.ietf.org/html/draft-barth-homenet-wifi-roaming-00

Ich versuch mal ein minimales Bastel-Image zu bauen... :)

glycoknob, 2015-12-31:
Das ist alles doch ziemlich komplex - vielleicht wäre es erstmal eine gute Idee die IPv6 Ideen von @andibraeu  zum laufen zu bekommen. Das müsste überschaubar sein vom Aufwand.

storchi, 2015-12-31:
@bittorf Ich meine das ebenso ernst.Seit wann dürfen hier nur code Ablieferer was schreiben? Sei froh, wenn dir hier jemand auf die Finger schaut, sonst wäre das WN längst tot, wie es vor wenigen Jahren schon fast war, als es eine EinMannShow wurde.

bittorf, 2015-12-31:
@storchi es geht hier NICHT um code, sondern darum einfach nur irgendwas ohne substanz in den raum zu rufen. das ist typisch politiker-style. wenigstens solltest du grob umreissen, wie du es machen moechtest, aber nicht einfach nur "ich will weltfrieden".

bittorf, 2015-12-31:
@glycoknob andi's IPv6 adressen waere glaube ich der erste und einfache schritt. also ULA zum meshen und globale IPv6 zum verteilen auf dem router. das hat als abhaengigkeit "nur" die knotennummer, funktioniert aber einfach so ohne extra daemon.

storchi, 2015-12-31:
Am 31.12.2015 um 16:20 schrieb Bastian Bittorf:
> @glycoknob <https://github.com/glycoknob> andi's IPv6 adressen waere
> glaube ich der erste und einfache schritt. also ULA zum meshen und
> globale IPv6 zum verteilen auf dem router. das hat als abhaengigkeit
> "nur" die knotennummer, funktioniert aber einfach so ohne extra daemon.

knotennummern sind doof, die machen einen registrator nötig...
cs

storchi, 2015-12-31:
Am 31.12.2015 um 16:19 schrieb Bastian Bittorf:
> @storchi <https://github.com/storchi> es geht hier NICHT um code,
> sondern darum einfach nur irgendwas ohne substanz in den raum zu rufen.
> das ist typisch politiker-style. wenigstens solltest du grob umreissen,
> wie du es machen moechtest, aber nicht einfach nur "ich will weltfrieden".

es ist zwecklos. du willst es nicht kapieren.
cs

glycoknob, 2015-12-31:
können wir die Zofferei aufs Treffen verlagern? Ist eh nur ein "wünsch dir was" Ticket und ich finde auch, dass wir Knotennummern und Registrator überflüssig machen sollten, wenn wir das können. Die Daten dafür kann es ja alles noch geben für owm und Karte aber die Abhängigkeit IP<->Knotennummer müsste nicht sein.... ist eh alles nur Wunschdenken momentan.

kaese77, 2015-12-31:
# funkdisziplin off
Frohes Neues und Happy IPv6-Jahr 2016!
# funkdisziplin on
VG, René

glycoknob, 2016-01-01:
/off

Frohes Neues! Hab euch alle lieb und habt euch doch mal alle lieb! Gern
auch über IPv6!

glycoknob, 2016-01-01:
Zurück zum Thema: IPv4 würde mit einer Technik namens 464XLAT laufen und alles wäre fein. Router die Internet anbieten wären dafür Gateways und ebenso z.B. VPN-Server. OLSRv2 kann automatisch link-local (169.xxx.) Adressen vergeben und sorgt dafür, dass diese im Netz unique sind... daraus könnte man den lokalen IPv6 DHCP-Bereich berechnen und hier auch mit macvlan und Basti's cooler Technik Roaming machen.

glycoknob, 2016-01-02:
Das wäre die wunschliste: https://gist.github.com/glycoknob/c43002e4519e7085cd5a :)

bittorf, 2016-01-03:
das 464XLAT -> CLAT sieht spannend aus: https://forum.openwrt.org/viewtopic.php?id=57455

glycoknob, 2016-01-04:
Das ist ziemlich cool. Braucht man aber nur, wenn man IPv6 only Mesh hat - man benötigt eine IPv4 Adresse für das Interface bzw. ein Subnetz und einen IPv6 Präfix - v4 Pakete werden als IPv6 Pakete mit der IPv4 Adresse in der IPv4 Adresse an den Gateway geschickt ... die Route zum Präfix ist eine "normale" IPv6 Route Das Gateway (der PLAT-Teil) macht daraus wieder IPv4 - das coole ist, dass OLSR auch identische Präfixe announcen kann also Anycast.. d.h. jeder Knoten wo die Daten wieder rauskommen gibt das selbe 464XLAT-Präfix an und die kürzeste Route wird genommen. 

http://tools.ietf.org/html/rfc6877#appendix-A

andibraeu, 2016-01-14:
http://wiki.freifunk.net/Ideas#Automatic_Network_Configuration

glycoknob, 2016-01-30:
Noch ein Update hier - Juliusz Chroboczek (der Autor von babeld und Guru auf dem Gebiet) hat mir noch auf der OLSR Liste geantwortet: 

> I'm toying around with some IPv6 mesh concepts and came across
> 464XLAT[1,2] for dealing with IPv4 in IPv6 Networks.

Please don't.  v4-to-v6-to-v4 translation is a horrible idea (have a look
at RFC 6145 and 6146 for the gory details).  DS-Lite, which uses tunnelling
insted, is a much cleaner solution.

glycoknob, 2016-08-12:
Die kleine Brötchen backen Variante wäre ein einheitlicher ULA-Prefix für die Firmware und olsr6 v1  - Software ist schon im Image, ULA-Prefix sind 2-3 Zeilen in `apply_profile.code` und die VPN Server bräuchte noch entweder IPv6 oder nen Tunnel nach in-berlin mit Subnetz.

glycoknob, 2017-01-19:
Aktueller Plan ist IPv6 zusammen mit OLSRv2 und 802.11s zu machen und das vorher erstmal sein zu lassen.

bittorf, 2017-12-13:
parallel zu adhoc+olsr1 oder als harten schnitt?

mtippmann, 2017-12-14:
Wir haben noch keine Idee. Vielleicht erstmal auf 802.11s umstellen...

mtippmann, 2018-08-14:
mit olsr1 erstmal kein ipv6.... dann ist sowieso alles neu und anders...

closed 2018-08-14 by mtippmann