weimarnetz/weimarnetz #233: Splash Netzwerkweit Synchronisieren / Persistenz über Reboot hinaus

glycoknob, 2015-12-26:
Noch ein Brainstorming Catch-All Ticket. Freifunk Jena hat schon entsprechende Software dafür - siehe https://github.com/FreifunkAdvanced - die ganzen härteren Probleme sind da schon sauber gelöst. Könnten wir einbauen um Nutzern Weimarnetzweit für $duration den nervigen Splash zu ersparen. Mit iptables ipset müsste das auch bis zu 100k MACs auf 32MB Geräten ohne Probleme klappen... 

Vorteil wäre, dass Leute die einmal geklickt haben erstmal Weimarweit ihre Ruhe haben diesbezüglich... wenn wir eine Einheitliche SSID für den AP _hätten_ würden sich die Geräte wohl auch automatisch einwählen. 

Ansonsten wäre das für alle Nutzer, die das täglich Nutzen ein paar Minuten/Sekunden Lebenszeit mehr um eine nervige Meldung die sie eh kapiert haben wegzuklicken :) 

Wir sollten uns die Technik mal von den Jenaern erklären lassen (wenn die Lust drauf haben), tendenziell find ich das wir das unbedingt haben sollten. Reduziert den Nerv-Faktor für Nutzer und ist technisch auch ganz cool.

bittorf, 2015-12-29:
alles viel simpler und schon lange fertig...ich mach bald nen pullrequest fuer eine 'roaming' kalua-klasse, da ist dann alles dafuer drinne und es sind echt nur 3 shell-functions

andibraeu, 2015-12-29:
ich würde mir auch gern mal die Technik der Jenaer anschauen. Tellerrand und so. Auch andere Communities haben schöne Shell-Scripte

glycoknob, 2015-12-30:
Hier ist wohl der relevante Code - das ganze ist einfach ein p2p key,value store. Hier ist das Shellwrapper: https://github.com/FreifunkAdvanced/p2ptbl und hier die Implementierung des Broadcasters: https://github.com/FreifunkAdvanced/hbbp -  habs mir noch nicht im Detail angeschaut, aber das nutzt IPV6 Multicast - ff02::1	All nodes on the local network segment. Also völlig dezentral.

glycoknob, 2016-08-12:
Kein Splash ist die Lösung, lohnt nicht da Energie/Zeit zu investieren.

closed 2016-08-12 by glycoknob