weimarnetz/weimarnetz #235: seltsamer Fehler im vpn

storchi, 2016-01-07:
3600er hat internet und hat eine Verbindung mit vpn3
alle routen sind da, olsr läuft auf dem tap0
ping auf die server vpn adresse geht nicht
_vpn restart brachte nix
alle vpn checks sind erfolgreich
ip rules sind da
reboot des routers hilft

auf den clients geht kein internet
ideen?

bittorf, 2016-01-07:
ich denke wir sollten wirklich  testen (jede minute) ob ping auf die andere seite geht. wenn nicht: ifdown ausloesen

bittorf, 2016-01-07:
habe versehentlich gepusht, bitte anschauen:
https://github.com/weimarnetz/weimarnetz/commit/ab195fffe5bec7e4a543a2feddeab5d400bc3f26

bittorf, 2016-01-07:
trotzdem muss mal mal schauen, warum ein '_vpn restart' da nix bringt

storchi, 2016-01-07:
achso, vergessen. olsr vom vpn server sagte mir:
```
Jan  7 10:07:14 vpn3 olsrd[8876]: Node 10.63.1.5 is publishing your ip 10.63.122.57 as MID!
```

glycoknob, 2016-01-07:
> 10.63.1.5

Da stinkt irgendwas gewaltig: 

https://github.com/weimarnetz/vpnconfig/blob/5926156fc01b693e424b9980c3f0388f65ca64af/README.md#ip-schema

glycoknob, 2016-01-07:
10.63.1.4/30 ist VPN2 

Damit gehört 10.63.1.5 zu VPN2 - dann gibt es quasi eine doppelte IP, erklärt auch warum Ping kaputt ist.

storchi, 2016-01-07:
andy wars, der hat einen anderen router zur 122 gemacht...
an den Pranger mit ihm!

closed 2016-01-07 by storchi

andibraeu, 2016-01-07:
andy wars, der hat einen anderen router zur 122 gemacht... an den Pranger mit ihm!ich bedauere nichts!

bittorf, 2016-01-07:
teeren und federn! 8-)

bittorf, 2016-01-07:
ernstgemeinte frage: soll der  pingtest nun drinbleiben?

storchi, 2016-01-08:
ich finde den test, ob im olsr die server ip da ist ausreichend

bittorf, 2016-01-08:
das problem am OLSR-test ist, das die IP auch drin bleibt bei "INFINITE" und das
der test mit neueren OLSR-daemons nicht mehr funktioniert...

glycoknob, 2016-01-08:
Ich find den pingtest gut, ist eine Art Heartbeat für IP - haben wir nicht in vtun - weil layer2 - 200 Pings pro Minute wird der Server auch überleben, aber ist schon ein wenig mehr Traffic.

storchi, 2016-01-08:
ich gebe zu bedenken, dass in diesem unseren Fall der pingtest auch nix bringt
cs

bittorf, 2016-01-08:
ich denke das es gut is bei schlechtem ping den server-auswahl-algo neu anzustossen. nur weil der test vor 12h serverX ausgewaehlt hat, muss das ja nicht immer die  beste wahl bleiben. in diesem speziellen fall haette der pingtest nur bedingt was geholfen...