weimarnetz/weimarnetz #236: mon-ami west (Knoten 106) - cron blockiert

andibraeu, 2016-01-09:
Auf dem Router habe ich festgestellt, dass er häufig keine Meldungen ans Monitoring schickt. Er ist trotzdem per VPN verbunden.

Eine Ursache habe ich festgestellt: Irgendwann verabschiedet sich der logger mit segmentation fault. logread gibt dann nichts mehr aus und mit logger können keine Logmeldungen mehr geschrieben werden. Bei einem Neustart des Cron Watchdogs bleibt dieser dann hängen und verursacht eine Load von > 1. In der Folge laufen viele Prüfscripts nicht mehr, dnsmasq bleibt irgendwann auch hängen. Auch der nächtliche Reboot unterbleibt.

Was ich außerdem festgestellt habe: Es sind sehr viele Clients verbunden, >20. Wird da evtl. doch der Speicher knapp?

Ich würde vorschlagen wir ersetzen das Gerät (841nd v8) durch einen Gerät mit mehr RAM (3600/4300), da hab ich noch was zu Hause rumliegen.

bittorf, 2016-01-09:
@andibraeu die geraete haben alle 32mb ram. betrifft das die neue oder die alte firmware?

glycoknob, 2016-01-09:
@andibraeu: Könnte speicher sein, könntest die Ausgabe vom /proc/meminfo posten? Ansonsten klingt WDR3600 / 4300 gut - die haben 128MB RAM oder? 

Ähnliches hat auch der s140 Router gemacht. Müssen wir mal schauen ob es bei vielen Clients Probleme gibt.

glycoknob, 2016-01-09:
segfaults hab ich bei wenig RAM mit zram gesehen. Könnte auch ein Problem sein. Problem ist wohl, dass tmpfs, regulärer RAM und zram um Speicher streiten und dabei ist irgendwas kaputt.

andibraeu, 2016-01-09:
das ist die neue Firmware. Das besondere an dem Router ist, der wird halt richtig gut benutzt... >20 WLAN-Clients is schon was

glycoknob, 2016-01-09:
Anderes mögliches Problem wäre cron watchdog <-> procd raspawn für cron

bittorf, 2016-01-09:
@andibraeu @glycoknob ihr habt recht, die 128mb speicher.
trotzdem ist das komisch. ich habe geraete mit >100 wlan-clients in der landschaft und die haben nur 32mb RAM...

bittorf, 2016-01-09:
@andibraeu dieser patch von martin für auto-restart von crond via procd ist drinne, oder?

glycoknob, 2016-01-09:
@bittorf ja, der müsste da drinnen sein.

andibraeu, 2016-01-10:
cron war nicht das Problem, log hing. 

Das führte allerdings dazu, dass /tmp/cron_watchdog.sh nicht durchlief, sondern an irgendeiner Zeile hing, die Logs schreibt

andibraeu, 2016-01-10:
is drin: https://github.com/weimarnetz/builder/blob/master/openwrt-patches/0001-add-procd-respawn-flag-to-rpcd-and-cron-init-scripts.patch

bittorf, 2016-01-10:
diese komische "cron_watchdog.sh" ist bei mir schon lange rausoperiert. stattdessen laeuft dieses netifd-refresht-alle-30min-ein-dummy-interface geschichte. ich lass das mal als erinnerung offen, das ist dummerweise eine zu starke aenderung als es noch mit in die "stable" zu tun.

andibraeu, 2016-01-10:
die stable ist ja nun auch schon draußen :-) mach doch ein issue im feature paradise auf...

vielleicht sollten wir uns am dienstag einen neuen Releasenamen überlegen und dann dort neue features eintüten

glycoknob, 2016-01-10:
würd ich noch mit warten, erstmal den master Füllen - die init.d Änderungen haben auch noch Probleme.

andibraeu, 2016-01-10:
wir müssen ja noch keinen Branch anlegen, so lange wir noch Features sammeln, ich meinte eher, dass wir Milestones haben und nicht nur das Feature Paradise

glycoknob, 2016-01-10:
Ja, das wäre cool!

glycoknob, 2016-01-23:
https://github.com/weimarnetz/weimarnetz/issues/251

closed 2016-01-23 by glycoknob