weimarnetz/weimarnetz #237: aufräumen der /etc/init.d/-Skripte

bittorf, 2016-01-09:
da gibt es noch fehler:

```
bastian@X301:~/software/weimarnetz$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean

bastian@X301:~/software/weimarnetz$ for F in openwrt-addons/etc/init.d/*;do grep -q 'sh /etc/rc.common'$ "$F" && continue; echo $F; doneopenwrt-addons/etc/init.d/S12switch_etx
openwrt-addons/etc/init.d/S43ethers_fff+
openwrt-addons/etc/init.d/S51crond_fff+
openwrt-addons/etc/init.d/set_etc_hardware
```
ich schicke einen PR

glycoknob, 2016-01-09:
Nehmen wir das Ticket hier mal als catch-all für alle Probleme - in den Logfiles waren auch noch ein paar ungereimtheiten.

glycoknob, 2016-01-11:
noch eine Frage dazu: Würde das erzeugen der /tmp/NETPARAMS und alles in S43ethers nach /etc/kalua auslagern - netparams nach _net und ethers vielleicht in eigene Datei. Gibt da momentan noch ein Problem mit der Erstellung von /tmp/NETPARAMS und so ist's vielleicht leichter zu debuggen.

bittorf, 2016-01-11:
@glycoknob ok, das hört sich gut an und ist sowieso auf meiner liste. ich mach das.

bittorf, 2016-01-11:
das gleich mit "S43ethers" kommt noch.

glycoknob, 2016-08-12:
Ich teste nochmal!

glycoknob, 2017-01-23:
Ist nur noch `apply_profile.code` und das könnte man noch umziehen z.b. nach `/usr/sbin`

andibraeu, 2017-07-23:
ist mit der Umstellung der apply_profile.code gelöst

closed 2017-07-23 by andibraeu