weimarnetz/weimarnetz #240: Roaming mit macvlan

glycoknob, 2016-01-09:
Wie stehen die Chancen, dass wir Roaming wie in kalua auch bei uns hinbekommen? 

Klappt das eventuell auch mit 10.63.0.0/16? Ansonsten könnten wir 2.4+5Ghz für Roaming bridgen und z.B. 100.64.0.0/10 nutzen. Für LAN und Adhoc wäre alles beim alten. Roaming IP-Range würde sich aus Knotennummer errechnen damit wir keine Konflikte haben. 

Vorteil wäre Roaming ;) aber auch APs auf 5Ghz und ein /24 oder /25 für Clients - bei der Gelegenheit könnten wir auch das kaputte IP-System fixen und saubere Subnetze bauen. 

Zudem könnten wir stadtweit eine einheitliche SSID haben - insbesondere wo nicht nur ein Router ist, wäre das von Vorteil. 

Nachteil wäre, dass AP-Clients nicht mehr direkt über 10.63.0.0/16 erreichbar sind. Aber Dienste laufen bei uns bislang nur über LAN und ohne OLSR gibt es auch keine Möglichkeit für Clients diese anzubieten.

bittorf, 2016-01-09:
ich ueberlege, ob wir nicht gleich den schritt nach IPv6 machen und das roaming dafuer bauen.

andibraeu, 2016-01-10:
wie wär's, wenn wir das als Google Summer of Code-Projekt ausschreiben?

Inhalte:

* automatische Konfiguration der Netzwerkinterfaces
* Evaluation verschiedener Lösungen
* Entwicklung der notwendigen Werkzeuge als OpenWRT-Paket
* Integration von DNS
* Netzwerkweite Information über Freischaltung auf Infoseite
* Vermeidung zentraler Dienste wie Registrator
* usw. usf.

sollte sich doch in 3 Monaten konzentrierter Arbeit machen lassen. @storchi macht Mentor und haut demjenigen, der die Arbeit macht ( @glycoknob ?) regelmäßig auf die Finger

bittorf, 2016-01-10:
zusaetzlich muss evtl. noch ein klein wenig an OLSR2 rumgehakt werden (um dynamisch HNA's hinzu- bzw. wegzunehmen). GSoC klingt gut. Ich würde auch Mentor machen...

andibraeu, 2016-01-14:
http://wiki.freifunk.net/Ideas#Automatic_Network_Configuration

bittorf, 2016-11-02:
wenn man momentan 2 gleiche SSID's hat, haengt das Netz sofort und fuer immer (bis zum naechsten DHCP-lease). Dies passiert z.b. auch, wenn zwischen 2 und 5 GHz auf dem gleichen Router hin und her geschaltet wird. Das passiert häufig und ist keine Ausnahme.

Selbst wenn nicht geroamt (im engeren Sinne) wird, haengt das Netz beim verbinden:
1) Nutzer hat Netz und steckt Handy in die Tasche/Bildschirm aus
2) Nutzer ist woanders und schaltet Telefon ein, Verbindung wird sofort hergestellt (gleiche SSID) und DHCP-lease wird NICHT erneuert. Netz haengt.

Das ist Mist. Einfacher workaround waere macvlan mit NAT fuer alle WLAN-Clients und alle AP-Interfaces in einer bridge. Geht vollstaendig per uci und funktioniert.

Christian: du sagtest es ist NICHT besser als unsere bisherige Lösung: warum?

kaese77, 2016-11-02:
Kann man dem Client nicht irgendwie signalisieren, dass er sich neue Daten per DHCP holen muss?

Es gibt dafür RFC 3203. Was weder ISC DHCP noch dnsmasq unterstützen:

https://www.ietf.org/rfc/rfc3203.txt

Ich kann CS verstehen, weil es ein kleines Problem mit enormer Komplexität erschlägt und nicht FF-konform ist. Von allem technischen Nutzen mal abgesehen.

bittorf, 2016-11-03:
Es gibt keinen mechanismus um vom Server aus den Client zu einem neuen Lease zu zwingen. Selbst wenn man das lease jede Minute erneuern laesst, hat man im zweifel 60 sekunden lang ein kaputtes .netz

die komplexitaet ist gering (macvlan+bridge) und was soll daran nicht ff-konform sein?

kaese77, 2016-11-03:
Das hat Christian schon gesagt. Das Netz 100.64.0.0/10 ist ein privates und kein FF-Netz und es kann zu Problemen kommen, wenn andere Communities das selbe tun. Dann hast Du plötzlich mehrere Netze, die sich überschneiden und InterCity-VPN macht dann Probleme.

Das das auf dem Router mit macvlan+bridge funktioniert hatten wir, glaube ich, verstanden.

Ich hoffe, ich habe das so richtig zusammengefasst.

Für mich sieht es so als sei die nachhaltigste Lösung eine Erweiterung von dnsmasq um die Funktionalität aus RFC3203. Also ran! Geben wir der Community was zurück. ;-)

glycoknob, 2016-11-03:
> Das hat Christian schon gesagt. Das Netz 100.64.0.0/10 ist ein privates und kein FF-Netz und es kann zu Problemen kommen, wenn andere Communities das selbe tun. Dann hast Du plötzlich mehrere Netze, die sich überschneiden und InterCity-VPN macht dann Probleme.

Das ist nur für die AP SSID, wir könnten das auch so bauen, dass das nichtmal zum VPN geht - dann haben wir haben doppeltes NAT - das Netz darf nicht in den Intercity-VPN - Knoten, LAN-Clients, WLAN-Clients auf einer zweiten SSID ohne Roaming sind weiterhin im 10.63.0.0/16 Netz. 

Das ist kein großes Problem IMHO und das Roaming-Netz ist nur zusätzlich zum vorhandenen Freifunk-Netz aktiv und geht nicht über Weimarnetz hinaus. Zudem ist der Usecase, dass der Client innerhalb des Weimarnetzes erreichbar sein muss eher exotisch - wir könnten nochmal überlegen ob das 100.64.0.0/10 Netz wirklich per HNA raus muss, da können wir mit @bittorf drüber schauen. 

> Für mich sieht es so als sei die nachhaltigste Lösung eine Erweiterung von dnsmasq um die Funktionalität aus RFC3203. Also ran! Geben wir der Community was zurück. ;-)

Das ist alles ziemlich haarig - beim RFC3203 steht, dass es durch RFC6704 abgelöst wird (https://tools.ietf.org/html/rfc6704) und dort in der Einleitung steht, dass das nur geht wenn man ein eine Möglichkeit hat den DHCP Server zu authentizifieren - dann haben wir immer noch das Problem, dass wir Clients vom Android, iPhone, Windows 7, OS X etc.pp haben wo das alles vermutlich nicht implementiert ist. 

```
   The DHCP reconfigure extension defined in [RFC3203] is a useful
   mechanism allowing dynamic reconfiguration of a single host triggered
   by the DHCP server.  Its application is currently limited by a
   requirement that a Forcerenew message is always authenticated using
   procedures as described in [RFC3118].  Authentication for DHCP
   [RFC3118] is mandatory for FORCERENEW; however, as it is currently
   defined, [RFC3118] requires distribution of constant token or shared-
   secret out-of-band to DHCP clients.

   The motivation for making authentication mandatory in DHCP FORCERENEW
   was to prevent an off-network attacker from taking advantage of DHCP
   FORCERENEW to accurately predict the timing of a DHCP renewal.
   Without DHCP FORCERENEW, DHCP renewal timing is under the control of
   the client, and an off-network attacker has no way of predicting when
   it will happen, since it doesn't have access to the exchange between
   the DHCP client and DHCP server.

   However, the requirement to use the DHCP authentication described in
   [RFC3118] is more stringent than is required for this use case and
   has limited adoption of DHCP FORCERENEW.  [RFC3315] defines an
   authentication protocol using a nonce to prevent off-network
   attackers from successfully causing clients to renew.  Since the off-
   network attacker doesn't have access to the nonce, it can't trick the
   client into renewing at a time of its choosing.
```

Das sieht alles nicht so einfach aus...

bittorf, 2016-11-04:
das 100.64.0.0/10 netz kann so geNATed werden, das es nicht als HNA drin sein muss. es tauchen dann also nur 10.63er IP's auf. NAT meint in dem fall umschreiben z.b. auf die LAN-Adresse. (die Main-IP vom OLSR am besten). Macht berlin auch so...

glycoknob, 2016-11-04:
guter plan!

storchi, 2016-11-04:
genatete wlan clients. man fasst es nicht!
cs

andibraeu, 2016-11-04:
Ich möchte, dass die Endgeräte direkt erreichbar sind. Roaming finde ich nicht wichtig, dann setzen wir besser die Leasetime runter.

@bittorf kannst du uns zeigen, wo berlin "das so macht"?

bittorf, 2016-11-04:
In berlin sind alle WLAN-clients hinter einem NAT.

mach mal auf einem berliner router: iptables -t nat -nxvL

kaese77, 2016-11-15:
> Ich habe auf meinem Berliner Router nachgeschaut und dort ist kein NAT für WLAN-Clients konfiguriert. Zur Sicherheit habe ich auch nochmal bei
> den Firmwareentwicklern nachgefragt.
> 
> Was man macht: man bridged alle APs und das LAN-Interface

Dafür!

storchi, 2016-11-15:
+1
auch wenn ich mich wiederhole

glycoknob, 2017-01-23:
Roaming ist jetzt im master, mit NAT :D :hammer:  

Mit ein wenig basteln bekommen wir aber auch beides parallel hin bzw. können das an und abschalten.

storchi, 2017-01-24:
was machst du dir die Arbeit?
nimm doch die frimware von Basti

glycoknob, 2017-01-24:
Schau mal in die commit logs - alles mit "cleanup" - @bittorf hat ziemlich viele spezifische Sachen drinnen, die wir nicht brauchen - und ich finds cool ein wenig roaming zu haben, ist 2017 :D

storchi, 2017-01-24:
ich finds wichtiger wlan clients im freifunk zu haben

glycoknob, 2017-01-24:
Ja, das kommt noch - Roaming SSID + SSID für 10.63.x.x - dann sollten alle glücklich sein.

mtippmann, 2018-08-14:
fake roaming läuft seit ner weile. good enough erstmal.

closed 2018-08-14 by mtippmann