weimarnetz/weimarnetz #243: flash von 842er endet in einer reboot schleife

storchi, 2016-01-15:
firmware aus dem meshkit
router war fabrikneu
telnet geht kurz, das log sagt:

```
root@cs-node111:/ lr
Fri Jan 15 14:38:09 2016 daemon.info dnsmasq-dhcp[1459]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Fri Jan 15 14:38:09 2016 daemon.info dnsmasq[1459]: using local addresses only for domain lan
Fri Jan 15 14:38:09 2016 daemon.warn dnsmasq[1459]: no servers found in /tmp/resolv.conf.auto, will retry
Fri Jan 15 14:38:09 2016 daemon.info dnsmasq[1459]: read /etc/hosts - 1 addresses
Fri Jan 15 14:38:09 2016 daemon.info dnsmasq[1459]: read /tmp/hosts/dhcp - 1 addresses
Fri Jan 15 14:38:09 2016 daemon.info dnsmasq-dhcp[1459]: read /etc/ethers - 0 addresses
Fri Jan 15 14:38:13 2016 daemon.info dnsmasq-dhcp[1459]: DHCPREQUEST(br-lan) 192.168.1.116 f0:de:f1:ce:10:0e 
Fri Jan 15 14:38:13 2016 daemon.info dnsmasq-dhcp[1459]: DHCPACK(br-lan) 192.168.1.116 f0:de:f1:ce:10:0e TP-T420s
Fri Jan 15 14:38:16 2016 daemon.warn odhcpd[1375]: DHCPV6 CONFIRM IA_NA from 00046bdf9308af38bf5e331f69879f7d18c3 on br-lan: not on-link 
Fri Jan 15 14:38:16 2016 daemon.info dnsmasq[1459]: read /etc/hosts - 1 addresses
Fri Jan 15 14:38:16 2016 daemon.info dnsmasq[1459]: read /tmp/hosts/odhcpd - 0 addresses
Fri Jan 15 14:38:16 2016 daemon.info dnsmasq[1459]: read /tmp/hosts/dhcp - 1 addresses
Fri Jan 15 14:38:16 2016 daemon.info dnsmasq-dhcp[1459]: read /etc/ethers - 0 addresses
Fri Jan 15 14:38:17 2016 daemon.warn odhcpd[1375]: DHCPV6 SOLICIT IA_NA from 00046bdf9308af38bf5e331f69879f7d18c3 on br-lan: ok fd3b:fb92:9c58::691/128 
Fri Jan 15 14:38:18 2016 daemon.warn odhcpd[1375]: DHCPV6 REQUEST IA_NA from 00046bdf9308af38bf5e331f69879f7d18c3 on br-lan: ok fd3b:fb92:9c58::691/128 
Fri Jan 15 14:38:18 2016 daemon.info dnsmasq[1459]: read /etc/hosts - 1 addresses
Fri Jan 15 14:38:18 2016 daemon.info dnsmasq[1459]: read /tmp/hosts/odhcpd - 1 addresses
Fri Jan 15 14:38:18 2016 daemon.info dnsmasq[1459]: read /tmp/hosts/dhcp - 1 addresses
Fri Jan 15 14:38:18 2016 daemon.info dnsmasq-dhcp[1459]: read /etc/ethers - 0 addresses
Fri Jan 15 14:38:34 2016 daemon.info kalua: /etc/rc.common: netfilter_ipforwarding_start() writing 1 to /proc/sys/net/ipv4/ip_forward
Fri Jan 15 14:38:34 2016 user.emerg syslog: /etc/rc.common: line 5: /etc/init.d/S52vpnc_fff+: not found
Fri Jan 15 14:38:34 2016 daemon.info kalua: /etc/rc.common: net_local_inet_offer() early_boot_estimation: false
Fri Jan 15 14:38:34 2016 user.emerg syslog: Device "wlan1" does not exist.
Fri Jan 15 14:38:34 2016 user.emerg syslog: Device "wlan2" does not exist.
Fri Jan 15 14:38:34 2016 user.emerg syslog: Device "wlan3" does not exist.
Fri Jan 15 14:38:36 2016 daemon.debug kalua: /etc/init.d/S51crond_fff+: cron_config_write() profile = newstyle
Fri Jan 15 14:38:36 2016 daemon.debug kalua: /etc/init.d/S51crond_fff+: cron_config_write() + '* * * * * . /tmp/CRON_IS_CAREFUL'
Fri Jan 15 14:38:36 2016 daemon.info kalua: /etc/init.d/S51crond_fff+: cron_daemon_start() removing unneeded initfile
Fri Jan 15 14:38:36 2016 daemon.info kalua: /etc/init.d/S51crond_fff+: cron_daemon_start() [OK] started
Fri Jan 15 14:38:36 2016 user.emerg syslog: /etc/init.d/S51crond_fff+: line 5: can't open /tmp/cron_watchdog.pid: no such file
Fri Jan 15 14:38:36 2016 user.emerg syslog: /etc/init.d/S51crond_fff+: line 5: can't open /proc/nonexistent/cmdline: no such file
Fri Jan 15 14:38:36 2016 daemon.info kalua: /etc/init.d/S51crond_fff+: cron_watchdog_is_running() cron_watchdog is missing
Fri Jan 15 14:38:36 2016 daemon.info cron_watchdog: checking each 60 sec for crondpid 1653, see /tmp/cron_watchdog.sh
Fri Jan 15 14:38:37 2016 user.emerg syslog: olsrd: /etc/rc.d/S65olsrd: olsrd_write_loadplugin() Warning: Plugin library 'olsrd_httpinfo.so.0.1' not found, skipped
Fri Jan 15 14:38:37 2016 daemon.info olsrd: /etc/rc.d/S65olsrd: olsrd_write_loadplugin() Warning: Plugin library 'olsrd_httpinfo.so.0.1' not found, skipped
Fri Jan 15 14:38:37 2016 user.emerg syslog: olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface() Warning: Interface 'wlan' not found, skipped
Fri Jan 15 14:38:37 2016 daemon.info olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface() Warning: Interface 'wlan' not found, skipped
Fri Jan 15 14:38:38 2016 user.emerg syslog: olsrd: /etc/rc.d/S65olsrd6: olsrd_write_interface() Warning: Interface 'wlan' not found, skipped
Fri Jan 15 14:38:38 2016 daemon.info olsrd: /etc/rc.d/S65olsrd6: olsrd_write_interface() Warning: Interface 'wlan' not found, skipped
Fri Jan 15 14:38:40 2016 user.emerg syslog: cat: can't open '/tmp/GATEWAY_CHECK_RECENT': No such file or directory
Fri Jan 15 14:38:41 2016 user.emerg syslog: /usr/bin/lua: /usr/lib/lua/luci/util.lua:160: attempt to get length of local 'str' (a nil value)
Fri Jan 15 14:38:41 2016 user.emerg syslog: stack traceback:
Fri Jan 15 14:38:41 2016 user.emerg syslog:     /usr/lib/lua/luci/util.lua:160: in function 'split'
Fri Jan 15 14:38:41 2016 user.emerg syslog:     ?: in function 'get'
Fri Jan 15 14:38:41 2016 user.emerg syslog:     /usr/sbin/owm.lua:66: in main chunk
Fri Jan 15 14:38:41 2016 user.emerg syslog:     [C]: ?
Fri Jan 15 14:38:41 2016 user.emerg syslog: root: Starting olsr prio...
Fri Jan 15 14:38:41 2016 user.notice root: Starting olsr prio...
Fri Jan 15 14:38:41 2016 daemon.err mount_root: no jffs2 marker found
Fri Jan 15 14:38:41 2016 cron.info crond[1747]: crond (busybox 1.23.2) started, log level 5
Fri Jan 15 14:38:43 2016 daemon.err mount_root: failed - mount -t jffs2 /dev/mtdblock3 /rom/overlay: Input/output error
Fri Jan 15 14:38:43 2016 user.emerg syslog: setting up led WAN
Fri Jan 15 14:38:43 2016 user.emerg syslog: setting up led LAN1
Fri Jan 15 14:38:43 2016 user.emerg syslog: setting up led LAN2
Fri Jan 15 14:38:43 2016 user.emerg syslog: setting up led LAN3
Fri Jan 15 14:38:43 2016 user.emerg syslog: setting up led LAN4
Fri Jan 15 14:38:43 2016 user.emerg syslog: setting up led WLAN
Fri Jan 15 14:38:43 2016 user.emerg syslog: setting up led USB
Fri Jan 15 14:38:43 2016 user.emerg syslog: root: /etc/init.d/apply_profile [BOOT] start - /etc/init.d/apply_profile
Fri Jan 15 14:38:43 2016 user.notice root: /etc/init.d/apply_profile [BOOT] start - /etc/init.d/apply_profile
Fri Jan 15 14:38:43 2016 user.emerg syslog: root: /etc/init.d/apply_profile.code: boot is set - non interactive mode
Fri Jan 15 14:38:43 2016 user.notice root: /etc/init.d/apply_profile.code: boot is set - non interactive mode
Fri Jan 15 14:38:43 2016 user.emerg syslog: root: /etc/init.d/apply_profile.code: uptime now: 65 seconds, waiting 5sec
Fri Jan 15 14:38:43 2016 user.notice root: /etc/init.d/apply_profile.code: uptime now: 65 seconds, waiting 5sec
[... 86 lines cut ...]
Fri Jan 15 14:39:01 2016 user.emerg syslog: root: /etc/init.d/apply_profile.code: ....working on section 'dhcp'
Fri Jan 15 14:39:01 2016 user.notice root: /etc/init.d/apply_profile.code: ....working on section 'dhcp'
Fri Jan 15 14:39:01 2016 user.emerg syslog: cfg07411c
Fri Jan 15 14:39:02 2016 user.emerg syslog: root: /etc/init.d/apply_profile.code: [ok] ready with section 'dhcp'
Fri Jan 15 14:39:02 2016 user.notice root: /etc/init.d/apply_profile.code: [ok] ready with section 'dhcp'
Fri Jan 15 14:39:03 2016 user.emerg syslog: /etc/init.d/apply_profile.code: line 1525: /etc/init.d/cron: not found
Fri Jan 15 14:39:03 2016 user.emerg syslog: root: /etc/init.d/apply_profile.code: disabling olsrd6
Fri Jan 15 14:39:03 2016 user.notice root: /etc/init.d/apply_profile.code: disabling olsrd6
Fri Jan 15 14:39:04 2016 user.emerg syslog: ln: /etc/rc.d/S90olsr-prio: File exists
Fri Jan 15 14:39:04 2016 user.emerg syslog: 
Fri Jan 15 14:39:04 2016 user.emerg syslog: [ok] R E A D Y
Fri Jan 15 14:39:04 2016 user.emerg syslog: 
Fri Jan 15 14:39:04 2016 user.emerg syslog: removing /etc/init.d/apply_profile with:
Fri Jan 15 14:39:04 2016 user.emerg syslog: rm /etc/init.d/apply_profile
Fri Jan 15 14:39:04 2016 user.emerg syslog: 
Fri Jan 15 14:39:04 2016 user.emerg syslog: maybe kill S98apply_profile!
Fri Jan 15 14:39:04 2016 user.emerg syslog: now type: reboot && exit
Fri Jan 15 14:39:04 2016 user.emerg syslog: root: /etc/init.d/apply_profile [BOOT] ready - removing '/etc/init.d/apply_profile', autorebooting in 30 sec, avoid this by 'touch /tmp/STOP'
Fri Jan 15 14:39:04 2016 user.notice root: /etc/init.d/apply_profile [BOOT] ready - removing '/etc/init.d/apply_profile', autorebooting in 30 sec, avoid this by 'touch /tmp/STOP'
Fri Jan 15 14:39:04 2016 user.emerg syslog: root: /etc/init.d/apply_profile [BOOT] led_firework, loop 0/10
Fri Jan 15 14:39:04 2016 user.notice root: /etc/init.d/apply_profile [BOOT] led_firework, loop 0/10
root@cs-node111:/ lr -f
Fri Jan 15 14:39:07 2016 user.emerg syslog: root: /etc/init.d/apply_profile [BOOT] led_firework, loop 1/10
Fri Jan 15 14:39:07 2016 user.notice root: /etc/init.d/apply_profile [BOOT] led_firework, loop 1/10
```

storchi, 2016-01-15:
nach gefühlt 10 neustarts kam er dann hoch

bittorf, 2016-01-15:
bug ist hier beschrieben: https://dev.openwrt.org/ticket/21468

storchi, 2016-01-20:
evtl lags auch daran, dass ich ein sysupgrade image genommen habe
cs

closed 2016-01-20 by storchi