weimarnetz/weimarnetz #246: Features festlegen

andibraeu, 2016-01-17:


bittorf, 2016-01-18:
OLSR2 fuer IPv6 sollte nicht weh tun.

glycoknob, 2016-08-12:
Alles nicht sooo viel Aufwand: 
https://github.com/weimarnetz/weimarnetz/issues/46
https://github.com/weimarnetz/weimarnetz/issues/252
https://github.com/weimarnetz/weimarnetz/issues/256
https://github.com/weimarnetz/weimarnetz/issues/252
https://github.com/weimarnetz/weimarnetz/issues/240
https://github.com/weimarnetz/weimarnetz/issues/237
https://github.com/weimarnetz/weimarnetz/issues/122
https://github.com/weimarnetz/weimarnetz/issues/36

Eventuell noch:
https://github.com/weimarnetz/weimarnetz/issues/109
https://github.com/weimarnetz/weimarnetz/issues/42

andibraeu, 2017-07-23:
Feature Freeze!!1!

closed 2017-07-23 by andibraeu