weimarnetz/weimarnetz #248: Releaseparty planen

andibraeu, 2016-01-17:
Vorlage siehe #125

storchi, 2016-01-20:
was solln das fuer ne party werden bei: DD

INGREDIENTS

1 part cranberry juice cocktail
1 part grapefruit juice
1 part orange juice
1 part pineapple juice
Collins glass

da fällt mein Part schon mal weg

andibraeu, 2016-01-20:
ich dachte, wir patchen das

kaese77, 2016-01-20:
Am 20.01.2016 um 10:50 schrieb Andi Bräu:
> ich dachte, wir patchen das

So?

--- dd.txt      2016-01-20 11:05:06.401162056 +0100
+++ dd-new.txt  2016-01-20 11:05:05.057162132 +0100
@@ -2,4 +2,5 @@
 1 part grapefruit juice
 1 part orange juice
 1 part pineapple juice
+4 part Vodka
 Collins glass

VG, René

storchi, 2016-01-20:
nur wegen des namens will ich den:

Dirty Dick recipe
rating 8.0
6 votes
serve in Highball Glass
alcohol 24%
1 oz Jack Daniel's® Tennessee whiskey
1 oz Southern Comfort® peach liqueur
1 oz Bacardi® 151 rum
1 oz Absolut® vodka
2 oz Sprite® soda
1 tbsp grenadine syrup
1 pinch salt
1 oz orange juice
1 splash cranberry juice

bittorf, 2016-01-20:
ich bin dafuer!

glycoknob, 2016-08-12:
Jetzt ist August :), neue Ticket für neues Release, bis dahin würd ich das erstmal schließen.

closed 2016-08-12 by glycoknob