weimarnetz/weimarnetz #249: vhs knoten 63 - webserver aus / vpn verbunden / uptime 7d

storchi, 2016-01-20:
```
root@weimarnetz-63-vhs:~ lr
Wed Jan 20 12:23:37 2016 daemon.info hostapd: wlan0-1: STA f4:09:d8:17:8c:da IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jan 20 12:32:45 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.19 90:b6:86:b4:01:8e 
Wed Jan 20 12:32:45 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.19 90:b6:86:b4:01:8e wardriver-wifi18-n63
Wed Jan 20 12:37:01 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.29 a8:44:81:23:fd:78 
Wed Jan 20 12:37:01 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.29 a8:44:81:23:fd:78 wardriver-wifi28-n63
Wed Jan 20 12:41:44 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.9 44:d4:e0:3b:9a:46 
Wed Jan 20 12:41:44 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.9 44:d4:e0:3b:9a:46 wardriver-wifi8-n63
Wed Jan 20 12:42:33 2016 daemon.info hostapd: wlan0-1: STA f0:24:75:6c:06:fb IEEE 802.11: authenticated
Wed Jan 20 12:42:33 2016 daemon.info hostapd: wlan0-1: STA f0:24:75:6c:06:fb IEEE 802.11: associated (aid 1)
Wed Jan 20 12:42:36 2016 daemon.info dnsmasq-dhcp[11785]: DHCPDISCOVER(wlan0-1) f0:24:75:6c:06:fb 
Wed Jan 20 12:42:36 2016 daemon.info dnsmasq-dhcp[11785]: DHCPOFFER(wlan0-1) 10.63.63.20 f0:24:75:6c:06:fb 
Wed Jan 20 12:42:36 2016 daemon.info dnsmasq-dhcp[11785]: DHCPDISCOVER(wlan0-1) f0:24:75:6c:06:fb 
Wed Jan 20 12:42:36 2016 daemon.info dnsmasq-dhcp[11785]: DHCPOFFER(wlan0-1) 10.63.63.20 f0:24:75:6c:06:fb 
Wed Jan 20 12:42:37 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.20 f0:24:75:6c:06:fb 
Wed Jan 20 12:42:37 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.20 f0:24:75:6c:06:fb wardriver-wifi19-n63
Wed Jan 20 12:42:39 2016 daemon.info hostapd: wlan0-1: STA f0:24:75:6c:06:fb IEEE 802.11: disassociated
Wed Jan 20 12:47:18 2016 daemon.info hostapd: wlan0-1: STA 44:d4:e0:3b:9a:46 IEEE 802.11: disassociated
Wed Jan 20 12:47:19 2016 daemon.info hostapd: wlan0-1: STA 44:d4:e0:3b:9a:46 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jan 20 12:48:26 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: authenticated
Wed Jan 20 12:48:26 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: associated (aid 1)
Wed Jan 20 12:48:30 2016 daemon.info dnsmasq-dhcp[11785]: DHCPDISCOVER(wlan0-1) 34:23:ba:0e:fa:57 
Wed Jan 20 12:48:30 2016 daemon.info dnsmasq-dhcp[11785]: DHCPOFFER(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 
Wed Jan 20 12:48:30 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 
Wed Jan 20 12:48:30 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 wardriver-wifi27-n63
Wed Jan 20 12:50:00 2016 daemon.info hostapd: wlan0-1: STA f0:24:75:6c:06:fb IEEE 802.11: authenticated
Wed Jan 20 12:50:00 2016 daemon.info hostapd: wlan0-1: STA f0:24:75:6c:06:fb IEEE 802.11: associated (aid 4)
Wed Jan 20 12:50:00 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.20 f0:24:75:6c:06:fb 
Wed Jan 20 12:50:00 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.20 f0:24:75:6c:06:fb wardriver-wifi19-n63
Wed Jan 20 12:51:57 2016 daemon.info hostapd: wlan0-1: STA a8:44:81:23:fd:78 IEEE 802.11: disassociated due to inactivity
Wed Jan 20 12:51:58 2016 daemon.info hostapd: wlan0-1: STA a8:44:81:23:fd:78 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jan 20 12:54:24 2016 daemon.info hostapd: wlan0-1: STA 90:b6:86:b4:01:8e IEEE 802.11: disassociated due to inactivity
Wed Jan 20 12:54:25 2016 daemon.info hostapd: wlan0-1: STA 90:b6:86:b4:01:8e IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jan 20 12:55:07 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: authenticated
Wed Jan 20 12:55:07 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: associated (aid 1)
Wed Jan 20 12:55:23 2016 daemon.info dnsmasq-dhcp[11785]: DHCPDISCOVER(wlan0-1) 34:23:ba:0e:fa:57 
Wed Jan 20 12:55:23 2016 daemon.info dnsmasq-dhcp[11785]: DHCPOFFER(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 
Wed Jan 20 12:55:23 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 
Wed Jan 20 12:55:23 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 wardriver-wifi27-n63
Wed Jan 20 12:56:46 2016 daemon.info hostapd: wlan0-1: STA f0:24:75:6c:06:fb IEEE 802.11: disassociated due to inactivity
Wed Jan 20 12:56:47 2016 daemon.info hostapd: wlan0-1: STA f0:24:75:6c:06:fb IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jan 20 12:56:57 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: authenticated
Wed Jan 20 12:56:57 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: associated (aid 1)
Wed Jan 20 12:56:57 2016 daemon.notice hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: did not acknowledge authentication response
Wed Jan 20 12:56:58 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 
Wed Jan 20 12:56:58 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 wardriver-wifi27-n63
Wed Jan 20 12:57:00 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 
Wed Jan 20 12:57:00 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 wardriver-wifi27-n63
Wed Jan 20 12:57:02 2016 daemon.info dnsmasq-dhcp[11785]: DHCPDISCOVER(wlan0-1) 34:23:ba:0e:fa:57 
Wed Jan 20 12:57:02 2016 daemon.info dnsmasq-dhcp[11785]: DHCPOFFER(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 
Wed Jan 20 12:57:13 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 
Wed Jan 20 12:57:13 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.28 34:23:ba:0e:fa:57 wardriver-wifi27-n63
Wed Jan 20 13:04:01 2016 daemon.notice hostapd: wlan0-1: STA 08:ee:8b:a3:e9:3f IEEE 802.11: did not acknowledge authentication response
Wed Jan 20 13:04:03 2016 daemon.info hostapd: wlan0-1: STA 08:ee:8b:a3:e9:3f IEEE 802.11: authenticated
Wed Jan 20 13:04:03 2016 daemon.info hostapd: wlan0-1: STA 08:ee:8b:a3:e9:3f IEEE 802.11: associated (aid 2)
Wed Jan 20 13:04:14 2016 daemon.info dnsmasq-dhcp[11785]: DHCPDISCOVER(wlan0-1) 08:ee:8b:a3:e9:3f 
Wed Jan 20 13:04:14 2016 daemon.info dnsmasq-dhcp[11785]: DHCPOFFER(wlan0-1) 10.63.63.23 08:ee:8b:a3:e9:3f 
Wed Jan 20 13:04:29 2016 daemon.info dnsmasq-dhcp[11785]: DHCPREQUEST(wlan0-1) 10.63.63.23 08:ee:8b:a3:e9:3f 
Wed Jan 20 13:04:29 2016 daemon.info dnsmasq-dhcp[11785]: DHCPACK(wlan0-1) 10.63.63.23 08:ee:8b:a3:e9:3f wardriver-wifi22-n63
Wed Jan 20 13:06:57 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: disassociated due to inactivity
Wed Jan 20 13:06:58 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jan 20 13:09:10 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: authenticated
Wed Jan 20 13:09:10 2016 daemon.notice hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: did not acknowledge authentication response
Wed Jan 20 13:09:10 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: associated (aid 1)
Wed Jan 20 13:09:41 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: disassociated
Wed Jan 20 13:09:41 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: disassociated
Wed Jan 20 13:09:41 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: disassociated
Wed Jan 20 13:09:42 2016 daemon.info hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jan 20 13:09:42 2016 daemon.notice hostapd: wlan0-1: STA 34:23:ba:0e:fa:57 IEEE 802.11: did not acknowledge authentication response
[... 142 lines cut ...]
SwapFree:          11916 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          1584 kB
Mapped:             1484 kB
Shmem:                48 kB
Slab:              10512 kB
SReclaimable:       1760 kB
SUnreclaim:         8752 kB
KernelStack:         304 kB
PageTables:          296 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:       28940 kB
Committed_AS:       9400 kB
VmallocTotal:    1048372 kB
VmallocUsed:        2140 kB
VmallocChunk:    1026868 kB
root@weimarnetz-63-vhs:~ cat /tmp/dhcp.leases 
1453301343 7c:7d:3d:49:14:8e 10.63.63.11 wardriver-wifi10-n63 *
1453304628 f4:09:d8:17:8c:da 10.63.63.30 wardriver-wifi29-n63 01:f4:09:d8:17:8c:da
```

hab uhttpd neu gestartet

storchi, 2016-01-20:
wenn ich im webinterface unter administration rumklicke kommt jedes mal die autorisierung

andibraeu, 2016-01-20:
cron loift da nicht.. deswegen kein reboot

storchi, 2016-01-20:
ham wir da nicht 2 mechanismen?

storchi, 2016-01-20:
zram ist zu testzwecken aus, router wurde neu gestartet

bittorf, 2016-01-20:
diese crongeschichte wird im naechsten release gefixt.

bittorf, 2016-01-20:
cron ist der dreh und angelpunkt, kann aber bei speichermangel abgeschossen werden. daher muss es einen restart durch procd geben oder das cron.minutely direkt durch procd machen lassen.

storchi, 2016-01-21:
waere es da nicht sinnvoller den grund fuer den speichermangel zu finden?
sonst haben wir den nächsten ungefixten workarround drin.
beim naechsten testen der firmware sollten wir den täglichen workarround-restart weg lassen, der verschleiert zu viele fehler
cs

glycoknob, 2016-01-21:
Grund könnte sein, dass der procd respawn per Default nur 5x läuft. 

```
# respawn automatically if something died, be careful if you have an alternative process supervisor
# if process dies sooner than respawn_threshold, it is considered crashed and after 5 retries the service is stopped
  procd_set_param respawn ${respawn_threshold:-3600} ${respawn_timeout:-5} ${respawn_retry:-5}
```

Wenn der wegen Speichermangel 5x innerhalb einer Stunde gekillt wird wäre das schon möglich. 

Das dürfte es aber _nicht_ sein, in der Firmware ist eine Option drin, dass der Router sich rebootet, wenn ein Prozess vom OOM-Killer geholt wird. 

> wenn ich im webinterface unter administration rumklicke kommt jedes mal die autorisierung

Das hatte ich auch schonmal, da ist rpcd oder ubus tot. 

Bleibt nur: 
- Crashes - könnte wegen zram ein Problem sein. 
- Defekte Hardware (RAM kaputt z.B.) 
- irgendwas anderes :/

```
root@weimarnetz-63-vhs:~ cat /proc/meminfo 
MemTotal:          29220 kB
MemFree:            3708 kB
MemAvailable:       6408 kB
``` 

Da sind 6MB RAM frei... auf meiner Kiste sind es etwa 7MB. 

Ich wäre dafür USB rauszukicken und printk() wieder einzubauen mit Crash-Logging, dann sind die SIGSEV OOM im logfile/dmesg. Zumindest zum Debuggen für 841N wäre so ein Image gut. Ich bau mal was.

glycoknob, 2016-01-21:
Ähnliche Probleme gibt's übrigens in der s140

storchi, 2016-01-21:
bitte mal zu testzwecken auf s140 zram auschalten - hab kein pw
auf monami west ist zram ebenso wie auf der 63 aus!
cs

andibraeu, 2016-01-21:
> 
> bitte mal zu testzwecken auf s140 zram auschalten - hab kein pw
>  auf monami west ist zram ebenso wie auf der 63 aus!
>  cs

ist aus auf s140...

Viele Grüße

Andi

bittorf, 2016-01-21:
reboot-on-oom klappt nur bei oom, nicht jedoch bei crashes. ein problem koennte sein, das bei euch die crashdumps in der ramdisk landen und schnell mehrere mb gross sind. checkt mal /tmp/*.core

bei mir zuende ich extra ein tmpfs-disk /coredumps die eine limitierte groesse hat. z.b. nur 128k ohne debug.

glycoknob, 2016-01-22:
Coredumps müssten deaktiviert sein. 

zram scheint einige Probleme in 3.18 zu haben: https://lwn.net/Articles/635446/ - das liest sich so, dass es mehr Schaden als Nutzen könnte. Zusätzlich gibt es einen Segfault Bug wenn wenig RAM frei ist, http://permalink.gmane.org/gmane.linux.kernel.commits.mm/91055 - ich weiß aber nicht ob das auch 3.18 betrifft. 

Würd sagen wir sollten das erstmal wegwerfen, ich teste nochmal mit zram und ohne - ohne zram wird jedenfalls ein OOM getriggert und rebooted - mit zram hängt der Router, ist langsam und vermutlich ist dann irgendwas kaputt.

glycoknob, 2016-01-22:
So wie ich das verstanden hab: Könnte sein, dass zram dafür sorgt, dass letzlich weniger RAM verfügbar ist und diesen auch nicht mehr freigibt. Zudem gibt's wohl Probleme wenn das System RAM von zram rück haben will und das kann zu hängern führen... das mit dem Segfaults muss ich nochmal testen, aber genau so ein Verhalten hatte ich mit dem OLSR-Routen Problem gesehen.

glycoknob, 2016-01-22:
Ist aber nur Spekulation.. könnte auch völlig anders sein.

glycoknob, 2016-01-22:
Mit zram: 

OLSR-Routes Seite aufrufen... 
```
root@ffweimar-hybrid-dhcp-231:~ logread -f
 Segmentation fault
```

bittorf, 2016-01-22:
das ist alles quark. die aktuelle weimarnetz-stable ist kernel 3.18.23 - also (fast) latest stable vom 3.18 und da gibts keine zram sorgen:

https://cdn.kernel.org/pub/linux/kernel/v3.x/ChangeLog-3.18.24
https://cdn.kernel.org/pub/linux/kernel/v3.x/ChangeLog-3.18.25

glycoknob, 2016-01-22:
segfaults kommen bei mir nur mit zram... die Probleme in den Links von mir sind alle noch in 3.18 - bist sicher, dass es das nicht sein kann?

glycoknob, 2016-01-22:
Irgendwas ist kaputt - mit zram: 

``` 
root@ffweimar-hybrid-dhcp-231:~# logread -f 
[...]
Fri Jan 22 16:34:01 2016 daemon.debug kalua: /bin/sh: watch_vpn() tunnel-olsr neigh available
Segmentation fault

^-- passiert nur mit zram 

root@ffweimar-hybrid-dhcp-231:~# 
root@ffweimar-hybrid-dhcp-231:~# logread -f -l 100
Thu Jan  1 01:00:00 1970 kern.emerg kernel:

^--- WTF! 
 
root@ffweimar-hybrid-dhcp-231:~# logread 
Fri Jan 22 16:34:31 2016 kern.info kernel: [  299.550000] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[...]
Corrupted MAC on input.
ssh_dispatch_run_fatal: Connection to 192.168.2.125: message authentication code incorrect

^--- Corrupted MAC 

root@ffweimar-hybrid-dhcp-231:~ logread 
Failed to find log object: Not found
Failed to find log object: Not found
Failed to find log object: Not found
Failed to find log object: Not found
Failed to find log object: Not found
Failed to find log object: Not found

^--- WTF? 

``` 

Ohne zram sieht man nur OOMs im logfile...

andibraeu, 2016-01-22:
das sind doch alles 841 v8 oder v9, oder?

da gibts ein Problem mit dem Ethernet, und man arbeitet an eimem fix für 15.05.1

glycoknob, 2016-01-22:
So, hab es mal geschafft die segfaults loggen zu lassen - passiert für mich _nur_ mit aktiven zram! 

```
vm.panic_on_oom=0
kernel.panic_on_oops=0
kernel.panic=0
print_fatal_signals=1
``` 

```
[  220.220000] potentially unexpected fatal signal 11.
[  220.230000] CPU: 0 PID: 3837 Comm: luci Not tainted 3.18.23 #3
[  220.230000] task: 819c5d10 ti: 811b4000 task.ti: 811b4000
[  220.240000] $ 0   : 00000000 00000001 2c0a2020 770d66b8
[  220.240000] $ 4   : fffffffc 00000004 770d66ec 00009540
[  220.250000] $ 8   : 00000002 6e746572 94fffffb 7fffffff
[  220.260000] $12   : 00000025 00000002 00a923a8 009d48b8
[  220.260000] $16   : 770d66b8 00025568 00cc2b40 2e323330
[  220.270000] $20   : 770d66ec 00000000 00000044 770d66b8
[  220.270000] $24   : 00000001 770b0ee0                  
[  220.280000] $28   : 770da390 7fa3a068 770d66ec 770b10b8
[  220.280000] Hi    : 00000005
[  220.290000] Lo    : 19999999
[  220.290000] epc   : 770b1170 0x770b1170
[  220.290000]     Not tainted
[  220.300000] ra    : 770b10b8 0x770b10b8
[  220.300000] Status: 0100f413	USER EXL IE 
[  220.300000] Cause : 8080000c
[  220.310000] BadVA : 2c0a2028
[  220.310000] PrId  : 0001974c (MIPS 74Kc)
[  220.310000] potentially unexpected fatal signal 11.
[  220.320000] CPU: 0 PID: 1014 Comm: logd Not tainted 3.18.23 #3
[  220.330000] task: 818d6b98 ti: 80b52000 task.ti: 80b52000
[  220.330000] $ 0   : 00000000 00000001 78cbaab8 00000000
[  220.340000] $ 4   : 77cb1410 20202273 77ca2000 00000000
[  220.340000] $ 8   : 00000000 0000f400 30303139 37346320
[  220.350000] $12   : 00000000 00000002 00000000 33613036
[  220.350000] $16   : 77cb1410 77ca9730 7fefc690 77c31b40
[  220.360000] $20   : 7ffc0328 00000000 00000000 00000000
[  220.360000] $24   : 00000000 77c432e4                  
[  220.370000] $28   : 77cb1410 7fefc570 7ffc02d0 77c95b73
[  220.370000] Hi    : 00000000
[  220.380000] Lo    : 5368a000
[  220.380000] epc   : 77c95c3f 0x77c95c3f
[  220.380000]     Not tainted
[  220.390000] ra    : 77c95b73 0x77c95b73
[  220.390000] Status: 0100f413	USER EXL IE 
[  220.400000] Cause : 00800008
[  220.400000] BadVA : 78cbaab8
[  220.400000] PrId  : 0001974c (MIPS 74Kc)
[  220.980000] potentially unexpected fatal signal 11.
[  220.980000] CPU: 0 PID: 1423 Comm: hostapd Not tainted 3.18.23 #3
[  220.990000] task: 819a7a20 ti: 817b8000 task.ti: 817b8000
[  221.000000] $ 0   : 00000000 00000000 0a686238 0a1fe019
[  221.000000] $ 4   : 0a1fe018 0a202020 7790f6ec 7790f9dc
[  221.010000] $ 8   : 00000001 7790f6ec 7790f6d8 fffffffe
[  221.010000] $12   : fffffffc 00000000 00000000 26100004
[  221.020000] $16   : 7790f6b8 00004009 00484220 00000040
[  221.020000] $20   : 7790f6ec 00000000 80000000 7790f6b8
[  221.030000] $24   : 0044440c 00000001                  
[  221.030000] $28   : 77913390 7fd071b8 7790f6ec 778ea1bc
[  221.040000] Hi    : 00000224
[  221.040000] Lo    : 00000495
[  221.050000] epc   : 778ea43c 0x778ea43c
[  221.050000]     Not tainted
[  221.050000] ra    : 778ea1bc 0x778ea1bc
[  221.060000] Status: 0100f413	USER EXL IE 
[  221.060000] Cause : 8080000c
[  221.060000] BadVA : 0a686238
[  221.070000] PrId  : 0001974c (MIPS 74Kc)
[... 120 lines cut ...]
[  251.290000] Status: 0100f413	USER EXL IE 
[  251.290000] Cause : 00800010
[  251.300000] BadVA : 000076b9
[  251.300000] PrId  : 0001974c (MIPS 74Kc)
[  268.520000] potentially unexpected fatal signal 11.
[  268.530000] CPU: 0 PID: 2114 Comm: sh Not tainted 3.18.23 #3
[  268.530000] task: 819a5838 ti: 8058c000 task.ti: 8058c000
[  268.540000] $ 0   : 00000000 00000001 004573b8 7734a4b4
[  268.550000] $ 4   : 011c7741 0086c7ec 0086c7ed 0086c99b
[  268.550000] $ 8   : 0086c7f4 81010100 ffe6c95b 00000013
[  268.560000] $12   : 0000000e 00000007 00000001 802bc560
[  268.560000] $16   : 011c7737 0086c7ec 7f930640 77349b40
[  268.570000] $20   : 7ffc0558 00000000 00000000 00000000
[  268.570000] $24   : 004573b8 7734a4b4                  
[  268.580000] $28   : 7737d390 7f92fd80 7ffc0508 0041673b
[  268.580000] Hi    : 0000000a
[  268.590000] Lo    : 0000004a
[  268.590000] epc   : 7734a4b4 0x7734a4b4
[  268.590000]     Not tainted
[  268.600000] ra    : 0041673b 0x41673b
[  268.600000] Status: 0100f413	USER EXL IE 
[  268.600000] Cause : 00800008
[  268.610000] BadVA : 011c7741
[  268.610000] PrId  : 0001974c (MIPS 74Kc)
```

glycoknob, 2016-01-22:
wie man sieht frisiert der einmal alles weg... ohne zram sehe ich nur OOM killer einträge aber alles andere läuft stabil, meistens wird auch nur luci gekillt als verursacher für den OOM und _keine_ segfaults! 

Ich sag zram ist schuld :P

bittorf, 2016-01-22:
ich würde eher sagen luci ist die ursache 8-)
ursache ist in jedem fall uclibc. bau mal das gleiche image mit 'musl' (also gleiche revision).

glycoknob, 2016-01-23:
DESIGNATED DRIVER (Bleeding Edge, r48456)
     Linux ffweimar-hybrid-dhcp-231 4.1.15 #2 Sat Jan 23 02:13:53 CET 2016 mips GNU/Linux

mit zram: 

```
root@ffweimar-hybrid-dhcp-231:~ dmesg
[  368.876321] potentially unexpected fatal signal 11.
[  368.881438] CPU: 0 PID: 366 Comm: ubusd Not tainted 4.1.15 #2
[  368.881457] task: 81970a10 ti: 80ed8000 task.ti: 80ed8000
[  368.881469] $ 0   : 00000000 00000000 3721b092 00000000
[  368.881492] $ 4   : 00000001 77b9c3e0 77ba4020 00000000
[  368.881513] $ 8   : 00000000 0000f400 00000011 0008003c
[  368.881533] $12   : 7ffb1b70 00000000 00000000 771e4320
[  368.881553] $16   : 77b9c3e0 77b82430 77c430b0 77c41538
[  368.881573] $20   : 77c3e000 77c3e000 7fcfc354 77c41490
[  368.881594] $24   : 00000001 77bba560                  
[  368.881614] $28   : 77ba4020 7fcfc290 00000000 77b87bf9
[  368.881635] Hi    : 00714c73
[  368.881646] Lo    : d928b5fb
[  368.881663] epc   : 77b87c1b 0x77b87c1b
[  368.881677] ra    : 77b87bf9 0x77b87bf9
[  368.881687] Status: 0100f413 USER EXL IE 
[  368.881704] Cause : 00800014
[  368.881715] BadVA : 3721b092
[  368.881727] PrId  : 0001974c (MIPS 74Kc)
[  369.525327] 
[  369.525327] do_page_fault(): sending SIGSEGV to 10dhcpscript for invalid read access from 772c6320
[  369.535098] epc = 772c6320 inra  = 772c6320 in
[  369.539721] 
[  369.553752] potentially unexpected fatal signal 11.
[  369.558822] CPU: 0 PID: 4723 Comm: 10dhcpscript Not tainted 4.1.15 #2
[  369.558841] task: 8182b758 ti: 81d74000 task.ti: 81d74000
[  369.558852] $ 0   : 00000000 00000001 00000000 00000000
[  369.558876] $ 4   : 0090e8c0 00000000 00000000 00000000
[  369.558896] $ 8   : 00000000 00001075 00000000 8030b9c0
[  369.558917] $12   : 7f896e80 0090c640 00000000 00440182
[  369.558937] $16   : fffffffc ffffffff 772c10b0 772bf538
[  369.558958] $20   : 772bc000 772bc000 7f89d6a4 772bf490
[  369.558978] $24   : 00000000 77280e80                  
[  369.558998] $28   : 772c6320 7f897500 00000000 772c6320
[  369.559019] Hi    : 0000001c
[  369.559031] Lo    : 00000031
[  369.559048] epc   : 772c6320 0x772c6320
[  369.559061] ra    : 772c6320 0x772c6320
[  369.559072] Status: 0100f413 USER EXL IE 
[  369.559089] Cause : 00800008
[  369.559100] BadVA : 772c6320
[  369.559112] PrId  : 0001974c (MIPS 74Kc)
[  370.361210] 
[  370.361210] do_page_fault(): sending SIGSEGV to ash for invalid read access from 00000000
[  370.369783] epc = 00000000 in busybox[400000+48000]
[  370.372885] ra  = 00000000 in busybox[400000+48000]
[  370.375813] 
[  370.377956] potentially unexpected fatal signal 11.
[  370.383014] CPU: 0 PID: 2215 Comm: ash Not tainted 4.1.15 #2
[  370.383032] task: 818ede30 ti: 80a5a000 task.ti: 80a5a000
[  370.383043] $ 0   : 00000000 00000001 0000130b 00000000
[  370.383066] $ 4   : ffffffff 7fe606f4 00000002 00000000
[  370.383087] $ 8   : 00000000 80064f20 811a2460 00040014
[  370.383108] $12   : 0000001a 00000013 0000000e 00000007
[  370.383127] $16   : 00000000 00001028 00000008 00000008
[  370.383147] $20   : 00000000 77e54000 7fe60e44 77e57490
[  370.383168] $24   : 00000001 77e2d6c0                  
[  370.383188] $28   : 00000367 7fe606a0 00000000 00000000
[  370.383208] Hi    : 00000000
[  370.383220] Lo    : 7ae147b0
[  370.383233] epc   : 00000000   (null)
[  370.383244] ra    : 00000000   (null)
[  370.383255] Status: 0100f413 USER EXL IE 
[  370.383272] Cause : 00800008
[  370.383282] BadVA : 00000000
[  370.383295] PrId  : 0001974c (MIPS 74Kc)
[  370.391661] potentially unexpected fatal signal 11.
[  370.396797] CPU: 0 PID: 1770 Comm: dropbear Not tainted 4.1.15 #2
[  370.396816] task: 8188b758 ti: 81612000 task.ti: 81612000
[  370.396828] $ 0   : 00000000 00000000 fffffffc 00000000
[  370.396852] $ 4   : 00000009 7f9fb4d8 00000000 00000001
[  370.396873] $ 8   : 00000001 4659739d d399c5bb 60cf0dde
[  370.396893] $12   : 1965cff1 00000000 00000000 00440000
[  370.396913] $16   : 0043a8b4 0000102e 7742e614 7f9fb4d8
[  370.396934] $20   : 00000000 7f9fb378 00000006 7f9fb4d8
[  370.396954] $24   : 00439ff8 773fd6c0                  
[  370.396974] $28   : 7742e320 bbf4089f f5fa0a4c 00000804
[  370.396995] Hi    : 000fed58
[  370.397007] Lo    : 97335404
[  370.397023] epc   : 00000804 0x804
[  370.397036] ra    : 00000804 0x804
[  370.397047] Status: 0100f413 USER EXL IE 
[  370.397063] Cause : 00800008
[  370.397074] BadVA : 00000804
[  370.397087] PrId  : 0001974c (MIPS 74Kc)
[  374.795542] potentially unexpected fatal signal 10.
[  374.800629] CPU: 0 PID: 1800 Comm: uhttpd Not tainted 4.1.15 #2
[  374.800648] task: 81972d48 ti: 80500000 task.ti: 80500000
[  374.800660] $ 0   : 00000000 00000000 c8332d5d 00418250
[  374.800683] $ 4   : c7f1ab0d 0041bcc0 00000000 00000020
[  374.800704] $ 8   : 00000001 80808080 00933207 002d4445
[  374.800724] $12   : 73797361 00000000 00000000 7574683d
[  374.800744] $16   : 0041bcc0 7fef8f60 779310b0 7792f538
[  374.800765] $20   : 7792c000 7792c000 7fef91b4 7792f490
[  374.800785] $24   : 00000005 778758c5                  
[  374.800805] $28   : 77936320 7fef8f48 00000000 004040c7
[  374.800826] Hi    : 00000069
[  374.800837] Lo    : 00000019
[  374.800854] epc   : 004040c1 0x4040c1
[  374.800867] ra    : 004040c7 0x4040c7
[  374.800878] Status: 0100f413 USER EXL IE 
[  374.800895] Cause : 80800010
[  374.800906] BadVA : c8332d65
[  374.800919] PrId  : 0001974c (MIPS 74Kc)

``` 

ohne zram rebootet sich der Router immer, obwohl er das nicht sollte - mal sehen. Jedenfalls keine Crashes ohne zram.

glycoknob, 2016-01-23:
Nach reboot mit richtigen sysctl settings _ohne_ zram: 

"normale" OOMs: 

```
[  138.138973] odhcpd invoked oom-killer: gfp_mask=0x201da, order=0, oom_score_adj=0
[  138.146816] CPU: 0 PID: 1108 Comm: odhcpd Not tainted 4.1.15 #2
[  138.152926] Stack : 80301974 00000000 00000001 80330000 00000000 00000000 00000000 00000000
[  138.152926]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  138.152926]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  138.152926]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  138.152926]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  138.152926]    ...
[  138.190072] Call Trace:[<80071400>] 0x80071400
[  138.194765] [<80071400>] 0x80071400
[  138.198372] [<800c8c90>] 0x800c8c90
[  138.201990] [<800c90c0>] 0x800c90c0
[  138.205621] [<800c9738>] 0x800c9738
[  138.209223] [<800cc150>] 0x800cc150
[  138.212828] [<800c990c>] 0x800c990c
[  138.216465] [<800ccc10>] 0x800ccc10
[  138.220074] [<800cfa08>] 0x800cfa08
[  138.223788] [<800c844c>] 0x800c844c
[  138.227394] [<8009d0a4>] 0x8009d0a4
[  138.231025] [<800e0dc8>] 0x800e0dc8
[  138.234710] [<800e3718>] 0x800e3718
[  138.238325] [<800691b8>] 0x800691b8
[  138.241945] [<801278fc>] 0x801278fc
[  138.245701] [<80128110>] 0x80128110
[  138.249352] [<80075f78>] 0x80075f78
[  138.252966] [<8012924c>] 0x8012924c
[  138.256624] [<800991c4>] 0x800991c4
[  138.260234] [<8012946c>] 0x8012946c
[  138.263866] [<800adb2c>] 0x800adb2c
[  138.267473] [<80060820>] 0x80060820
[  138.271124] 
[  138.272666] Mem-Info:
[  138.275064] active_anon:2239 inactive_anon:325 isolated_anon:0
[  138.275064]  active_file:124 inactive_file:139 isolated_file:0
[  138.275064]  unevictable:0 dirty:0 writeback:0 unstable:0
[  138.275064]  slab_reclaimable:195 slab_unreclaimable:1956
[  138.275064]  mapped:2 shmem:191 pagetables:102 bounce:0
[  138.275064]  free:159 free_pcp:0 free_cma:0
[  138.307971] Normal free:636kB min:676kB low:844kB high:1012kB active_anon:8956kB inactive_anon:1300kB active_file:496kB inactive_file:556kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:32768kB managed:28928kB mlocked:0kB dirty:0kB writeback:0kB mapped:8kB shmem:764kB slab_reclaimable:780kB slab_unreclaimable:7824kB kernel_stack:400kB pagetables:408kB unstable:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:6988 all_unreclaimable? yes
[  138.352643] lowmem_reserve[]: 0 0
[  138.356247] Normal: 24*4kB (R) 9*8kB (R) 5*16kB (R) 0*32kB 0*64kB 1*128kB (R) 1*256kB (R) 0*512kB 0*1024kB 0*2048kB 0*4096kB = 632kB
[  138.368881] 463 total pagecache pages
[  138.372665] 0 pages in swap cache
[  138.376132] Swap cache stats: add 24, delete 24, find 0/0
[  138.381709] Free swap  = 0kB
[  138.384700] Total swap = 0kB
[  138.387677] 8192 pages RAM
[  138.390471] 0 pages HighMem/MovableOnly
[  138.394448] 960 pages reserved
[  138.397605] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
[  138.406493] [  367]     0   367      287       17       5       0        0             0 ubusd
[  138.415419] [  368]     0   368      214       10       3       0        0             0 askfirst
[  138.424615] [ 1027]     0  1027      285       22       3       0        0             0 logd
[  138.433540] [ 1028]     0  1028      325       22       5       0        0             0 logread
[  138.442701] [ 1037]     0  1037      351       23       3       0        0             0 rpcd
[  138.451573] [ 1083]     0  1083      414       41       4       0        0             0 netifd
[  138.460617] [ 1108]     0  1108      312       24       4       0        0             0 odhcpd
[  138.469638] [ 1447]     0  1447      366       32       4       0        0             0 hostapd
[  138.478753] [ 1494]     0  1494      284        9       4       0        0             0 udhcpc
[  138.487827] [ 1495]     0  1495      220       12       4       0        0             0 odhcp6c
[  138.496948] [ 1753] 65534  1753      252       18       4       0        0             0 dnsmasq
[  138.506065] [ 1754]     0  1754      249       16       4       0        0             0 dnsmasq
[  138.515271] [ 1765]     0  1765      255       11       3       0        0             0 dropbear
[  138.524586] [ 1804]     0  1804      320       30       3       0        0             0 uhttpd
[  138.533642] [ 1823]     0  1823      271       26       4       0        0             0 dropbear
[  138.542901] [ 1862]     0  1862      284        9       4       0        0             0 crond
[  138.551857] [ 1898]     0  1898      284        9       3       0        0             0 sh
[  138.560631] [ 2296]     0  2296      287       14       5       0        0             0 ash
[  138.569407] [ 2433]     0  2433      284        9       3       0        0             0 lock
[  138.578245] [ 2532]     0  2532      285       11       3       0        0             0 ntpd
[  138.587223] [ 2605]     0  2605      284        9       3       0        0             0 lock
[  138.596078] [ 2931]     0  2931      284        8       3       0        0             0 sleep
[  138.605019] [ 2940]     0  2940      324       21       4       0        0             0 logread
[  138.614154] [ 3475]     0  3475      227       13       3       0        0             0 vtund
[  138.623056] [ 3590]     0  3590      559      164       4       0        0             0 olsrd
[  138.631983] [ 3901]     0  3901     2257     1772       5       0        0             0 luci
[  138.640896] Out of memory: Kill process 3901 (luci) score 238 or sacrifice child
[  138.648662] Killed process 3901 (luci) total-vm:9028kB, anon-rss:7088kB, file-rss:0kB
``` 

Kernel 4.4 hab ich nicht zum laufen bekommen, dort müsste das eventuell gefixt sein - ich probiers mal - also ich bleib dabei :)

glycoknob, 2016-01-23:
https://github.com/weimarnetz/weimarnetz/issues/251

closed 2016-01-23 by glycoknob

bittorf, 2016-01-23:
nunja, da muss man sich aber fragen was man will. meiner meinung nach sollte es einen reboot geben. sonst ist die kiste in einem undefinierbarem zustand. auch sollte es garnicht zu einem OOM kommen.

glycoknob, 2016-01-23:
ja, passiert halt manchmal - OOM ist ja kein Problem, dann rebootet der Router - auch nicht ideal aber gut genug, leider killt zram zufällig prozesse und das sorgt für die Probleme. Bis das gefixt ist haben dir wohl keine Wahl als das auszuschalten.

storchi, 2016-01-24:
ist das zram überhaupt nötig? 16mb geräte sind raus...
ich sach nur: keep it simple
cs

bittorf, 2016-01-25:
prinzipiell ist swap eine gute idee und besser als geschnitten brot. man kann aber selektiv (zum testen)  mal auf einigen routern (insbesondere belastete) das zram ausschalten und beobachten. (mal ne woche)