weimarnetz/weimarnetz #250: respawn crond klappt nicht

storchi, 2016-01-21:
kann das mal jemand testen der ahnung hat?
cs

#249

storchi, 2016-01-21:
man könnte auch crond aud dem oom raus nehmen:
http://backdrift.org/oom-killer-how-to-create-oom-exclusions-in-linux

glycoknob, 2016-01-21:
Blöderweise ist es wohl nicht OOM, weil Basti hat schon für den Fall vorgesorgt und der Router reboot nach 10s wenn ein OOM aufrtitt - siehe hier: https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-addons/etc/init.d/generate_kalua_loader

Da muss irgendwas anderes Faul sein, oder das wird nicht gesetzt/funktioniert nicht. Idee ist trotzdem gut und sollten wir einbauen.

storchi, 2016-01-21:
wie tue ich ein oom triggern?

glycoknob, 2016-01-21:
> wie tue ich ein oom triggern?
>

     echo f > /proc/sysreq-trigger

glycoknob, 2016-01-21:
oder auf olsr routen gehen in luci bei 32mb geräten :)  - das triggert aber mit zram seltsamerweise eben keinen OOM.... mal weiter debugger

glycoknob, 2016-01-23:
https://github.com/weimarnetz/weimarnetz/issues/251 geht's weiter.

closed 2016-01-23 by glycoknob