weimarnetz/weimarnetz #251: zram segfaults

glycoknob, 2016-01-23:
zram sorgt für segfaults und killed diverse Prozesse.. führt zu Zombie-Routern ohne cron, ssh, uhttpd 

Ursache für https://github.com/weimarnetz/weimarnetz/issues/250 https://github.com/weimarnetz/weimarnetz/issues/249 https://github.com/weimarnetz/weimarnetz/issues/236

glycoknob, 2016-01-23:
Für ChaosCalmer bleibt uns nur deaktivieren oder einen Workaround finden - trunk hat das Problem momentan auch noch, eventuell hilft Kernel 4.4

glycoknob, 2016-01-23:
/etc/init.d/zram disable && /etc/init.d/zram stop

glycoknob, 2016-01-23:
Die Patches die das Problem lösen sollten sind in 4.1 - seltsam.

bittorf, 2016-01-23:
wer sagt das, das die in 4.1 sind?

glycoknob, 2016-01-23:
Hier, das ist der Commit  https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/drivers/block/zram?h=v4.1&id=8c7f01025f7becd577bed9af93f974fc6798c5a3 

Löst laut http://permalink.gmane.org/gmane.linux.kernel.commits.mm/91055 ein Segfault Problem

glycoknob, 2016-01-23:
Das andere ist das hier: https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/drivers/block/zram?h=v4.1&id=4e3ba87845420e0bfa21e6c4f7f81897aed38f8c - aber da hab ich grad entdeckt, dass man das manuell triggern muss damit er aufräumt. Automatisch klappt das erst in 4.4

glycoknob, 2016-01-23:
Hab noch versucht was rauszubekommen - die segfaults kommen nicht immer oder automatisch, wenn man z.B. tmpfs langsam füllt dann füllt sich 100% vom zram swap ohne Probleme und das System auch nutzbar (wird nur unendlich langsam)  - wenn man es schnell füllt dann gibts's SIGSEVs... ob LZO oder LZ4 spielt keine Rolle (für lz4 muss man noch einen Patch einspielen). 

Keine Ahnung wie man das debuggen soll :/ 
Ein Limit setzen oder weniger RAM nehmen scheint keine Wirkung zu haben - da scheint was kaputt zu sein. 

Übrigens ist im `scheduler` ne Endlosschleife drinnen: 

```
while [ $( _system disc_free /tmp percent ) -lt 40 ]; do {
          df -h | grep /tmp$ | _log it high_discusage_before dae
          rm "$( ls -S1 /tmp/user-* | head -n1 )"
          df -h | grep /tmp$ | _log it high_discusage_after daem
} done   
```

bittorf, 2016-01-23:
der scheduler ist bei euch leider uralt und muss weg 8-) leider ist das ein riesenpatch.
ich mach mal ein ticket dafuer auf.

glycoknob, 2016-01-23:
Prozesse werden auch schon bei wenig Last/wenig RAM-Auslastung gekillt, würde die kaputten Router erklären. Hab versucht einen Debug-Kernel zu bauen, aber keine weiteren Infos bekommen...

glycoknob, 2016-01-23:
weiteres Problem ist ein teilweise unbenutzbar langsames System mit zram. System rödelt mit 99%

glycoknob, 2016-01-24:
Hab mal OpenWRT-Devel genervt

storchi, 2016-01-24:
guter mann

storchi, 2016-01-24:
bei vhs 65 war crond aus.  hab zram deaktiviert
cs

glycoknob, 2016-01-24:
Das zram Device selbst scheint okay zu sein - kann da drauf schreiben und lesen ohne Probleme - wird auch nichts korrumpiert auf den ersten Blick. Problem scheint irgendwo bei Vorgängen ums Swappen selbst aufzutreten. Wäre interessant das mal mit swap auf einem USB-Stick oder übers Netzwerk zu testen, ob es da auch noch auftritt. Ansonsten hab ich keine Ahnung wie man das debuggen soll - das sind so Linux Memory Subsystem Interna von dene ich so gar keine Ahnung hab - vermutlich tritt das Problem auch nur auf MIPS auf. 

Also ich sehe jedenfalls sowohl unter ChaosCalmer wie auch trunk viele Crashes, auch wenn nicht viel RAM benutzt wird - dabei wird zufällig irgendwas gekillt, vermutlich irgendwelche Speicherseiten die gerade vom oder zum Swap wandern. 

Ohne Swap ist alles in Ordnung und es gibt keine Crashes. Wenn RAM voll ist gibt's einen OOM und einen Reboot. Bislang hab ich es nur geschafft, das mit der OLSR-Routen Seite zu schaffen, alles andere läuft. 

Wir haben irgendwas zwischen 7 und 10Mbyte freien RAM auf den 32MB Geräten - für die Splashseite reicht das locker und alles andere wie conntrack und iptables dürfte nicht ins Gewicht fallen. Zumindest kann ich ordentlich rumsurfen und nichts ändert sich. 

Wäre also dafür das überall abzuschalten bis das geklärt ist. Momentan sorgt es dafür das alles crashed!

glycoknob, 2016-01-25:
Hab keine Zeit mehr dafür, irgendwas ist da kaputt - hier ist das OpenWRT-Ticket: https://dev.openwrt.org/ticket/21705 

Wenn jemand zu viel Langeweile hat - Gibt ein Buch wo Begriffe und Konzepte erklärt werden: https://www.kernel.org/doc/gorman/pdf/understand.pdf 

Es crashed beim `do_page_fault()` laut den Logs - also vermutlich wenn der Speicher wieder aus dem Swap gelesen wird. 

Könnte auch irgendwas tief in den Kernel MIPS Interna sein - u.A. liest sich das hier eher gruselig: http://patchwork.linux-mips.org/patch/7615/ 

> Bug can pops-up in heavy paging environment in form of random application/kernel
crashes. It is unusual in embedded and that may be a reason why it is not
catched yet.

Das hier versteh ich nicht wirklich, aber könnte auch ein Problem sein: https://patchwork.linux-mips.org/patch/8635/ - zram müsste nicht DMA sein, da ja komplett in Software, wenn ich das richtig verstehe. 

Die Patches sind nicht im Kernel, weiß auch nicht ob das überhaupt damit zusammenhängt. Lustiger Bug jedenfalls

glycoknob, 2016-08-12:
zram ist erstmal deaktiviert, müssten mit LEDE und Kernel 4.4. nochmal testen. Ich schließe das erstmal.

closed 2016-08-12 by glycoknob

bittorf, 2016-10-21:
wie ist der stand: laeuft LEDE ohne zram? gibt es schon tests mit aktuellem 4.4?

glycoknob, 2016-10-25:
Nope weiterhin crashes - siehe den (LEDE Bug)[
https://bugs.lede-project.org/index.php?do=details&task_id=206&string=zram&advancedsearch=on&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=
]