weimarnetz/weimarnetz #252: neuen scheduler verwenden, spart speicher/CPU

bittorf, 2016-01-23:
der neue Scheduler (bei mir schon 2 jahre) ist besser:
das fuellen geht schneller und das abarbeiten ist besser.

Beispiel:
```
echo >>$SCHEDULER '_watch vpn'
echo >>$SCHEDULER_IMPORTANT '_watch olsr'
_scheduler run
```
insbesondere die ersten beiden sachen *kann* der neue loader schon (die variablen sind also bekannt und werden angelegt in openwrt-addons/etc/kalua/scheduler (mit erklaerung))

glycoknob, 2016-01-24:
Genügt es da die scheduler datei zu kopieren und die Aufrufe anzupassen?

bittorf, 2016-01-25:
zusaetzlich muss noch cron.minutely UND cron.taskplanner kopiert werden.

glycoknob, 2016-01-25:
hmm, lässt sich nicht 90% davon mit 2-3 zusätzlichen cron Zeilen erschlagen die eben aller 15m oder täglich laufen? Oder gibt's gute Gründe dafür, dass so auszulagern?

bittorf, 2016-01-25:
auf lange sicht moechte ich von cron weg und das durch procd erledigen lassen. cron hat 2 entscheidene nachteile:

1)
zwei verschiedene jobs (z.b. alle minute und alle 5 minuten) werden zur vollen minute *gleichzeitig* gestartet. auf einem desktop ok, aber nicht auf den schwachen routern.

2)
wenn ein job haengt, wird der selbe nicht nochmal gestartet. das ist doof.

storchi, 2016-01-25:
> zwei verschiedene jobs (z.b. alle minute und alle 5 minuten) werden zur vollen minute gleichzeitig gestartet. auf einem desktop ok, aber nicht auf den schwachen routern.
zählt nicht als argument, dazu nimmt man primzahlen(zeiten)

bittorf, 2016-01-25:
egal wie man es macht, cron.minutely wird immer mit irgendwas gleichzeitig zünden.
und wie ich auch drueber nachdenke wird es kollidieren...oder? 2/3/5/7...

storchi, 2016-01-26:
dann mach doch cron.secondly mit primzahlen, oder weniger checks, oder effektive skripte oder oder oder

bittorf, 2016-01-27:
auch wenn cron.secondly mit primzahlen laeuft, kollidiert das mit cron.minutely.
das ganze hier ist aber jammern auf hohem niveau, aber ich haette es halt gerne.

glycoknob, 2016-01-28:
procd respawn könnte sich dafür eignen... 

     procd_set_param respawn ${respawn_threshold:-3600} ${respawn_timeout:-5} ${respawn_retry:-5}

die Parameter sind alle 32-bit unsigned integer, wäre sekundengenau - hätte auch den Vorteil, das keine doppelten Checks laufen, wenn einer länger braucht.

glycoknob, 2016-01-28:
z.B. 

     procd_set_param respawn 3600 900 0

Müsste für alle ewigkeit aller 15 Minuten laufen.

glycoknob, 2016-01-28:
Sauberste Lösung wäre runit (ist in Busybox) und runwhen https://wiki.uberspace.de/system:runwhen

storchi, 2016-01-28:
das dusselige respawn funktioniert doch schon bei crond nicht, oder hab ich was verpasst?

glycoknob, 2016-01-28:
@storchi da hab ich das default genommen, der hört dann nach 5x crashen auf. Wenn ich Zeit hab aktualisier ich den Patch dafür nochmal.

storchi, 2016-01-28:
ACHSO
kannst du da zuguckn ober das probiert?

glycoknob, 2016-01-28:
Ja, müsste man sehen können. Bin noch nicht zum testen gekommen.

bittorf, 2016-01-28:
ich werde mit respawn bei gelegenheit mal spielen. danke!

glycoknob, 2016-08-12:
@bittorf was brauchen wir für den neuen scheduler aus kalua?

bittorf, 2016-10-21:
fuer den neuen scheduler reicht es:
/etc/kalua/scheduler
zu uebernehmen und alle aufrufe von 'scheduler -a' zu ersetzen mit
echo >>$SCHEDULER "meinjob"
dann noch irgendwo (cron) ein '_scheduler run' einfuegen

PR #267 (2017-01-18): sync _scheduler with kalua
fixes #252 

cron.taskplanner.default fehlt hier, weil zu viele abhängigkeiten von anderen kalua scripten. 

ist auch erstmal crash&burn - ungetestet.
closed 2017-01-18 by glycoknob