weimarnetz/weimarnetz #263: _vpn Skripte haben ein Problem

glycoknob, 2016-06-21:
ich sehe endlos-schleifen und konstante olsr Neustarts - Ursache ist unklar, könnte ein _watch Script script sein. Reboot hilft manchmal, manchmal nicht. Da ist irgendwas kaputt.

andibraeu, 2016-06-24:
hast du vielleicht ein Log, eine Knotennummer?

glycoknob, 2016-06-24:
hab ich lokal bei mir beabachtet (188) und man sieht es auch auf dem mr router (43), wenn man `_vpn restart` eintippt. man sieht in den logfiles dass olsr sich ständig neustartet, vielleicht sich das auch wieder aber scheint eine schleife zu triggern

andibraeu, 2016-06-24:
die 43 könnte mit der doppelten Knotennummer kollidieren. Und wenn er sich mit dem gleichen VPN-Server verbindet, dann knallt es eben...

storchi, 2016-11-16:
funzt

closed 2016-11-16 by storchi