weimarnetz/weimarnetz #264: Doppelte IP Knoten 43

glycoknob, 2016-06-21:
Taucht im Log auf. Ist wohl Abhängig vom VPN Server? 

Ist definitiv was faul, SSH-Key warnt z.B. je nach Verbindungsversuch.

glycoknob, 2016-06-21:
```
root@h332:~ ssh root@10.63.43.1

ssh: Connection to root@10.63.43.1:22 exited: 

ssh-rsa host key mismatch for 10.63.43.1 !
Fingerprint is md5 a5:0a:c2:4d:1c:96:46:11:33:bf:96:87:01:09:fa:42
Expected md5 97:50:da:56:21:b4:d6:ed:32:47:a5:ad:69:b3:74:b9
If you know that the host key is correct you can
remove the bad entry from ~/.ssh/known_hosts
root@h332:~ ssh root@10.63.43.1
root@10.63.43.1's password: 
```

andibraeu, 2016-06-21:
On 21.06.2016 21:41, Martin Tippmann wrote:
> Taucht im Log auf. Ist wohl Abhängig vom VPN Server?
> 
> Ist definitiv was faul, SSH-Key warnt z.B. je nach Verbindungsversuch.

Lösung a) Dem Maschinenraumrouter ne neue Knotennummer verpassen

Lösung b) Iras Router ne neue Knotennummer verpassen.

Die 43 wurde halt doppelt genutzt

glycoknob, 2016-08-12:
maschinenraum hat ne neue Nummer

closed 2016-08-12 by glycoknob