weimarnetz/weimarnetz #277: Meshkit Firmware auf WDR4300v1 ist kaputt

glycoknob, 2017-02-07:
jffs2 lässt sich wohl nicht mounten, reboot schleife nach Factory boot. 

mount_root - no jffs2 marker found.

glycoknob, 2017-02-07:
Workaround: 

- Failsafe modus 
- Aktuelles LEDE Image (haben factory genommen) via Netcat vom Rechner auf Router ziehen 
- Flaschen 
- Firmware via sysupgrade draufbügeln

andibraeu, 2017-07-23:
Ggf. Zusammenhang mit #283 ?

glycoknob, 2017-07-28:
Das müsste in LEDE gefixt sein - wir hatten das Problem letztens auch beim flaschen der WDR4300er von Toni - OpenWrt hat irgendein Problem mit dem erstellen vom jffs2 aber nur nach einem Flash mit Factory Firmware. LEDE drüberbrügeln hat geholfen - beim 1x LEDE flaschen ist das noch nicht passiert.

mtippmann, 2018-01-16:
tritt nur für OpenWRT 15.05 auf -> legacy -> Mülltonne.

closed 2018-01-16 by mtippmann