weimarnetz/weimarnetz #282: check für vpn aus Gebrannte Mandeln übernehmen

andibraeu, 2017-07-23:


glycoknob, 2017-07-31:
bin jetzt doch für netifd + vtun - wir können alles aus dem alten vpn script abbilden brauchen weniger code und viel kann nach uci wandern - erster anfang ist: https://github.com/weimarnetz/weimarnetz/blob/ffwizard/files/lib/netifd/proto/vtun.sh 

dann müsste das watch script nur `ifstatus vpn` prüfen - wenn überhaupt.

bittorf, 2017-12-13:
sieht toll aus! - wie sieht die uci-config dafuer aus?

mtippmann, 2017-12-14:
so z.B: 

```
config vtun_vpn
        list endpoint '2.v.weimarnetz.de'
        list endpoint '3.v.weimarnetz.de'

config interface 'vpn'
        option proto 'vtun'
        option auto '0'
        option ipaddr '10.63.190.58'
        option netmask '255.255.255.252'
        option gateway '10.63.190.57'
```

das proto vpn sagt, dass er netifd nutzen soll, das auto=0 sorgt dafür, dass es nicht beim booten startet und die die section mit den servern wird vom netifd script ausgewertet - also ist quasi die serverliste. das ist momentan 1.v.weimarnetz bis 5.v.weimarnetz.de

mtippmann, 2018-01-16:
läuft.

closed 2018-01-16 by mtippmann