weimarnetz/weimarnetz #293: Multi VPN WAN

glycoknob, 2017-08-01:
Research

glycoknob, 2017-09-18:
Mit Policy-Routing und mwan3 kommt man schon recht weit: 

https://wiki.openwrt.org/doc/howto/mwan3 

Geht von Load-Balancing der Verbindung über Failover etc.pp.  

Unrealistich, dass als Feature einzubauen. Aber zum Testen/Basteln sicherlich interessant u.A. weil man so über USB mehrere LTE-SIM-Karten nutzen könnte oder 2 DSL-Anschlüsse via Round-Robin bedienen könnte. 

Man könnte probieren kompatibel mit mwan3 zu bleiben. Solange es aber keinen konkreten Grund gibt, dass zu nutzen eher nicht. 

Generell sieht das hier interessant aus: https://youtu.be/LTHLbdkdCjE?t=1009

mtippmann, 2018-03-20:
nope.

mtippmann, 2018-03-20:
okay, bleibt erstmal als idee hier.

mtippmann, 2018-08-14:
unrealistisch im moment.

closed 2018-08-14 by mtippmann