weimarnetz/weimarnetz #295: registrator ubus interface + owm.lua

glycoknob, 2017-08-12:
Wenn der registrator ubus kann, dann können wir bequem den Status über owm.lua abfragen und auch in Luci verlässliche Infos zur Knontenummer rausgeben.

mtippmann, 2018-01-16:
nice to have, geht evtl. auch ohne. wäre gut für lua api. siehe https://wiki.openwrt.org/doc/techref/ubus#lua_module_for_ubus

mtippmann, 2018-08-14:
uclient-fetch kann keinen request body von 40x und 50x responses ausgeben, plan wäre luci-lib-httpclient zu nutzen und das ganze als ubus rpcd service zu basteln...

mtippmann, 2018-09-08:
sieht z.B. so aus: https://github.com/weimarnetz/packages/blob/master/utils/luci-app-owm/files/owm.lua

andibraeu, 2018-09-08:
Schritt 1: Registratorclient auf lua umbauen
Schritt 2: in rpcd einbinden

andibraeu, 2019-07-21:
noch zu tun:

- [ ] rpcd-scripte erstellen
- [ ] owm.lua fragt rpcd nach und gibt die aktuelle lokale Knotennummer und die Antwort von Registrator zurück. Damit könnten wir sehen, wenn sich die registrierte Knotennummer von der aktuell gewählten Knotennummer unterscheidet

closed 2019-10-05 by andibraeu