weimarnetz/weimarnetz #300: Migration alte Firmware testen

glycoknob, 2017-09-12:
Beispiel für Config, die nicht stimmt: 

```
root@fh327:/etc/config :) cat uhttpd 

config uhttpd 'main'
        list listen_http '0.0.0.0:80'
        list listen_http '[::]:80'
        option redirect_https '1'
        option home '/www'
        option max_connections '100'
        option cert '/etc/uhttpd.crt'
        option key '/etc/uhttpd.key'
        option cgi_prefix '/cgi-bin'
        option script_timeout '60'
        option network_timeout '30'
        option http_keepalive '20'
        option tcp_keepalive '1'
        option rfc1918_filter '0'
        option index_page 'weimar-splash-index.sh'
        option error_page '/weimar-splash-index.sh'
        option config '/etc/httpd.conf'
        option interpreter '.sh=/bin/ash'
        option max_requests '5'
        option ubus_prefix '/ubus'

config cert 'px5g'
        option days '730'
        option bits '1024'
        option country 'ZZ'
        option state 'Somewhere'
        option location 'Uknown'
        option commonname 'OpenWrt'
```

glycoknob, 2017-09-19:
```

config interface 'wlan'
        option proto 'static'
        option ipaddr '10.63.65.65'
        option netmask '255.0.0.0'

config interface 'wlanRADIO1'
        option proto 'static'
        option ipaddr '10.63.65.67'
        option netmask '255.0.0.0'
```

bittorf, 2017-12-13:
was genau stimmt da nicht? evtl. ist die 'uci show uhttpd' besser...

mtippmann, 2017-12-14:
Das größte Problem müsste das hier sein: 

```
        option index_page 'weimar-splash-index.sh'
        option error_page '/weimar-splash-index.sh'
```

Die Seiten gibt es nicht mehr. Sollte sich leicht fixen lassen. 

Ein paar Sections in uci heissen jetzt anders, da könnte man die alten Sections noch löschen.

storchi, 2017-12-14:
upgrade einer nanostation m5: geht
hostnamen wird vergessen, ist danach openwrt
hostnamen gesetzt, ffwizzard
hostname ist wieder openwrt
siehe 10.63.45.33 passwort ist bekannt

storchi, 2017-12-14:
weiter:
ich bekomme der dhcp keine ip