weimarnetz/weimarnetz #302: offene Registrator Probleme

glycoknob, 2017-09-19:
- falsches Passwort überall richtig behandeln 
- Strategie für Umgang mit RSA/DSS Fingerprints 
- Passwort Ändern Funktion.

mtippmann, 2018-01-16:
serverseitig ist alles gelöst, clientcode muss noch angepast werden.

mtippmann, 2018-02-20:
```
root@ffweimar-190:~ :) registrator heartbeat
Failed to parse json data: unexpected end of data
registrator: heartbeat: 
```

mtippmann, 2018-02-20:
problem liegt wohl beim router

mtippmann, 2018-02-27:
uclient-fetch ignoriert den body bei einer 4xx antwort o_O

```
# uclient-fetch -O- "http://reg.weimarnetz.de/PUT/ffweimar/knoten/190?mac=34:ce:00:63:2b:cc&pass=13:0d:2f:1e:36:cb:ef:e4:01:0f:e3:fe:1f:c5:f5:4c:99:39:68:c
8"
Downloading 'http://reg.weimarnetz.de/PUT/ffweimar/knoten/190?mac=34:ce:00:63:2b:cc&pass=13:0d:2f:1e:36:cb:ef:e4:01:0f:e3:fe:1f:c5:f5:4c:99:39:68:c8'
Connecting to 77.87.48.24:80
HTTP error 401

$ curl  "http://reg.weimarnetz.de/PUT/ffweimar/knoten/190?mac=34:ce:00:63:2b:cc&pass=13:0d:2f:1e:36:cb:ef:e4:01:0f:e3:fe:1f:c5:f5:4c:99:39:68:c8"
{
  "message" : "Unauthorized, wrong pass",
  "status" : 401,
  "result" : {
    "key" : 450,
    "number" : 190,
    "mac" : "34ce00632bcc",
    "created_at" : 1510743234022,
    "last_seen" : 1513808161335,
    "location" : "/ffweimar/knoten/190",
    "network" : "ffweimar"
  }
```

andibraeu, 2018-03-04:
wahrscheinlich ist bei 4xx-Fehlern ein Body auch nicht unbedingt vorgesehen...

mtippmann, 2018-08-14:
siehe #295

closed 2018-08-14 by mtippmann