weimarnetz/weimarnetz #306: gzip für Webserver auf mapapi.weimarnetz.de

glycoknob, 2017-11-03:
@andibraeu müsste sich schnell mit `a2enmod deflate` erledigen lassen. Würde den Download der Map-Daten von 1,3MB auf 140kb reduzieren.

andibraeu, 2017-11-08:
das ist schon aktiviert...

andibraeu, 2017-11-08:
funktioniert auch: mapapi.weimarnetz.de/db/_all_docs?include_docs=true

bittorf, 2017-12-13:
Scheint nicht zu tun:

<code>
# sudo iptables -I INPUT -s mapapi.weimarnetz.de

# wget -SO /dev/null "http://mapapi.weimarnetz.de/db/_all_docs?include_docs=true"
--2017-12-13 16:18:23--  http://mapapi.weimarnetz.de/db/_all_docs?include_docs=true
Auflösen des Hostnamen »mapapi.weimarnetz.de (mapapi.weimarnetz.de)«... 77.87.48.24, 2001:bf7:b101:1::24
Verbindungsaufbau zu mapapi.weimarnetz.de (mapapi.weimarnetz.de)|77.87.48.24|:80... verbunden.
HTTP-Anforderung gesendet, warte auf Antwort... 
  HTTP/1.1 200 OK
  Date: Wed, 13 Dec 2017 15:18:24 GMT
  Server: CouchDB/1.6.1 (Erlang OTP/17)
  ETag: "E2U86BG5BX8FWTSXLBM1S3MP2"
  Content-Type: text/plain; charset=utf-8
  Cache-Control: must-revalidate
  Vary: Accept-Encoding
  Access-Control-Allow-Origin: *
  Keep-Alive: timeout=5, max=100
  Connection: Keep-Alive
  Transfer-Encoding: chunked
Länge: nicht spezifiziert [text/plain]
In »»/dev/null«« speichern.

/dev/null                               [               <=>                                               ]   1,20M   338KB/s    in 3,6s    

# sudo iptables -nxvL INPUT
Chain INPUT (policy ACCEPT 1283 packets, 1642811 bytes)
    pkts      bytes target     prot opt in     out     source               destination         
     881  1304987            all  --  *      *       77.87.48.24          0.0.0.0/0          
</code>

Das laden des JS ist garnicht so das Problem.

andibraeu, 2017-12-13:
nein, ist nicht kaputt. Ich denke, das ist eher ein client-Problem. Am Server ist es aktiviert, mehr kann ich da nicht tun

bittorf, 2018-01-02:
ok, das geht jetzt. Keine Ahnung was da war: gleiche Kommandozeile tut jetzt und überträgt daten.

mtippmann, 2018-01-16:
ich schließe das mal.

closed 2018-01-16 by mtippmann