weimarnetz/weimarnetz #316: Status ICVPN überprüfen

FreifunkUFO, 2017-12-15:
Wie funktioniert Euer ICVPN noch? Bitte überprüft das mal und gebt uns IPs, die über icvpn dauerhaft erreichbar sein müssten.

Hintergrund: Freifunk Jena will auch ICVPN einrichten und wir suchen noch Testpartner - und Anwendungsszenarien

siehe auch https://plan.leipzig.freifunk.net/issues/506

andibraeu, 2017-12-15:
> Am 15.12.2017 um 17:17 schrieb FreifunkUFO <notifications@github.com>:
> 
> Wie funktioniert Euer ICVPN noch? Bitte überprüft das mal und gebt uns IPs, die über icvpn dauerhaft erreichbar sein müssten.
> 
> 
ja, läuft, kann 10.61.211.54 von 10.63.52.33 aus erreichen. Kannste auch als Test nehmen.

Grüße

Andi

FreifunkUFO, 2017-12-15:
also ping geht, aber http nicht

http://lg.ff3l.de/traceroute/gw9/ipv4?q=10.63.52.33

und, andere IPs lassen sich nicht anpingen, scheitern hier an der 10.63.1.49

FreifunkUFO, 2017-12-15:
welche 10.207-IP hat jena1, was ausser dem public-key für tinc habt ihr anpassen muessen? (git irgendwo?)

FreifunkUFO, 2017-12-15:
Bei Jena sieht die Datei hier https://github.com/freifunk/icvpn-meta/blob/master/jena noch anders aus als leipzig und weimar. da fehlt die bgp-sektion
ping @egon0

egon0, 2017-12-28:
Mein Merge-Request wurde ja akzeptiert, Routen kommen an, diverse Tinc-Verbindungen stehen. Ich kann  10.63.1.49 pingen, die Leipziger Test-IPs gehen atm nicht.

FreifunkUFO, 2017-12-28:
hm ja, Leipzig scheint gerade nicht im ICVPN zu sein http://lg.ff3l.de/traceroute/gw9/ipv4?q=10.61.211.54

jena routen sind scheinbar auch schon bei ff3l angekommen: http://lg.ff3l.de/prefix_bgpmap/gw9/ipv4?q=10.17.253.1

trotzdem sieht dort der traceroute noch merkwürdig aus: 
vgl jena http://lg.ff3l.de/traceroute/gw9/ipv4?q=10.17.253.1
mit weimar (funktioniert richtig) http://lg.ff3l.de/traceroute/gw9/ipv4?q=10.63.141.1

der erste hop bei traceroute sollte (m.e.) die ic-vpn-adresse (10.207..) zeigen..

FreifunkUFO, 2018-01-03:
ich schliesse dieses ticket, weimarnetz ic_vpn scheint ja zu funktionieren http://lg.ff3l.de/traceroute/gw9/ipv4?q=10.63.141.1

@egon0 bitte dokumentiere im .o.g. leipzig ticket deine openwrt/lede variante! 
weitere disussion biite dort.

closed 2018-01-03 by FreifunkUFO