weimarnetz/weimarnetz #318: vtun zickt rum....

mtippmann, 2018-03-20:
```
Tue Mar 20 23:23:52 2018 daemon.notice netifd: vpn (9318): vtund[9318]: VTun client ver 3.X 03/20/2018 started
Tue Mar 20 23:23:52 2018 daemon.info vtund[9318]: Connecting to 77.87.48.35
Tue Mar 20 23:23:52 2018 daemon.notice netifd: vpn (9318): vtund[9318]: Connecting to 77.87.48.35
Tue Mar 20 23:23:53 2018 daemon.debug vtund[9318]: Remote Server sends <UeK> .
Tue Mar 20 23:23:53 2018 daemon.info vtund[9318]: Session Node331[77.87.48.35] opened
Tue Mar 20 23:23:53 2018 daemon.notice netifd: vpn (9318): vtund[9318]: Remote Server sends <UeK>
Tue Mar 20 23:23:53 2018 daemon.notice netifd: vpn (9318): .
Tue Mar 20 23:23:53 2018 daemon.notice netifd: vpn (9318): vtund[9318]: Session Node331[77.87.48.35] opened
Tue Mar 20 23:23:53 2018 daemon.info vtund[9318]: UDP connection initialized
Tue Mar 20 23:23:53 2018 daemon.notice netifd: vpn (9318): vtund[9318]: UDP connection initialized
Tue Mar 20 23:23:53 2018 daemon.info procd: Not starting instance ohybridproxy::data, command not set
Tue Mar 20 23:23:53 2018 daemon.info vtund[9318]: Connection closed by other side
Tue Mar 20 23:23:53 2018 daemon.notice netifd: vpn (9318): vtund[9318]: Connection closed by other side
Tue Mar 20 23:23:53 2018 daemon.info vtund[9318]: Session Node331[77.87.48.35] closed
Tue Mar 20 23:23:53 2018 daemon.info vtund[9318]: Exit
Tue Mar 20 23:23:53 2018 daemon.notice netifd: vpn (9318): vtund[9318]: Session Node331[77.87.48.35] closed
Tue Mar 20 23:23:53 2018 daemon.notice netifd: vpn (9318): vtund[9318]: Exit
Tue Mar 20 23:23:53 2018 daemon.notice netifd: Interface 'vpn' is now down
Tue Mar 20 23:23:53 2018 daemon.notice netifd: Interface 'vpn' is setting up now

```

storchi, 2018-03-21:
was macht da ein ohybridproxy?
cs

mtippmann, 2018-03-21:
der ist mittlerweile wieder weg - ich wollte mal mdns einbauen - aber hab das erstmal gelassen.

storchi, 2018-03-21:
und ändert sich was beim vtund?
cs

mtippmann, 2018-03-21:
das ist unabhängig davon - meistens klappt es sofort mit der verbindung aber z.b. der router im mr braucht wirklich ne weile bis er verbindung zum netz hat..

storchi, 2018-03-21:
das ist irgendwie so eine vtund eigenart. kommt imho nur vor, wenn direkt vorher schon eine verbindung stand. weiss nich was ich da auf dem server einstellen soll
cs

mtippmann, 2018-03-21:
könnten wir mal beim Treffen schauen, ich hab keine Ahnung wo das Problem sein könnte - irgendwas geht auf Netzwerkebene schief - ich hab auch alles mögliche probiert u.a. sleep reinzubauen das aus Gebrannte Mandeln 1:1 nachzubauen - evtl. liegt es an der neueren vtun Version gegenüber GebrannteMandeln...

storchi, 2018-03-23:
Glaub ich nicht, da es schon immer so war

mtippmann, 2018-05-15:
downgrad to 3.0.3

closed 2018-06-29 by andibraeu