weimarnetz/weimarnetz #319: ip rules sind zu unspezifisch

mtippmann, 2018-05-29:
```
[ "$INTERFACE" = "vpn" ] && {                                                                                                                                                  
                                                                                                                                                                               
    gw=$(uci -q get network.vpn.gateway)                                                                                                                                       
    case "$ACTION" in                                                                                                                                                          
        ifup)                                                                                                                                                                  
            ip route add default via $gw table 50                                                                                                                              
            ip rule add to 10.0.0.0/8 prio 30001 table main                                                                                                                    
            ip rule add from 10.0.0.0/8 prio 30010 table 50                                                                                                                    
            ip rule add to 100.64.0.1/10 prio 30002 table main                                                                                                                 
            ip rule add from 100.64.0.1/10 prio 30003 table 50    
```

mtippmann, 2019-07-21:
Problem ist, dass wenn wan-Interface eine 10.0.0.0/8 Addresse hat und die ip rule Regeln müssten spezifischer sein und an das Interface gebunden

bittorf, 2019-07-21:
gute idee:
```
root@F36-stube:~ 8( ip rule add to 10.0.0.0/8 oif wanX prio 30000 table 50
root@F36-stube:~ :) ip rule show
0:	from all lookup local 
9999:	from all to 8.8.8.8 lookup main 
9999:	from 100.64.0.0/10 to 100.64.0.0/10 lookup main 
9999:	from 10.63.22.125 lookup main 
30000:	from all to 10.0.0.0/8 oif wanX [detached] lookup hideandseek 
32766:	from all lookup main 
32767:	from all lookup default 
```

mtippmann, 2019-08-02:
@bittorf ah cool, das sieht gut aus!

andibraeu, 2019-08-13:
ich bin irgendwie zu blöd, so eine Rule zu setzen:

```
root@ffWeimar-28:~ 8( ip rule add to 10.0.0.0/8 oif eth1  prio 30001 table main
ip: invalid argument 'oif' to 'ip'
```
Mit `iif`dagegen geht es, was mach ich falsch?

bittorf, 2019-08-22:
mist andi, das liegt an deiner iproute2-version. hier:
```
# ip -V
ip utility, iproute2-ss4.11.0-1-lede
```
es gibt noch das abgespeckte und das hast du wohl...vermutlich

andibraeu, 2019-08-24:
wir haben hier das aus busybox

@mtippmann sollten wir das umbauen? oder explodieren uns da die images?

mtippmann, 2019-08-24:
> ip-full_4.16.0-8_mipsel_24kc.ipk | 150.0 KB 
> ip-tiny_4.16.0-8_mipsel_24kc.ipk | 113.4 KB

ist beides ganz schön fett und laut doku sollte das busybox ip route auch iif und oif können

andibraeu, 2019-08-25:
iif macht er tatsächlich auch, aber oif nicht.

Meinst du diese Doku hier? https://busybox.net/downloads/BusyBox.html

mtippmann, 2019-08-25:
@andibraeu genau - evtl. gibt es noch eine andere möglichkeit das policy-routing zu konfigurieren, ich schau nochmal bei meshkit - dort ist das anders gelöst, ist evtl. für uns auch klüger (und dort wird auch nur iif benötigt): http://doc.meshkit.freifunk.net/daily/html/usage/policy-routing.html#infos-zu-routingtabellen-und-zum-debuggen

mtippmann, 2019-08-25:
also dort ist es etwas anders (dort packt olsr seine routen in eine eigene tabelle) aber das müsste sich analog auch für uns so anpassen lassen, evtl. geht es auch wenn wir keinen expliziten default-gw setzen sondern olsr den setzen lassen, wenn es vpn gibt...

mtippmann, 2019-08-26:
wäre eine idee, die wir sowieso machen müssten, spätestens wenn wir mal auf wireguard umstellen, weil wir dann n-paralle vpn verbindungen haben und dann sollte olsr/babel die default-route wählen, wenn wir olsr routen in eine eigene tabelle schieben wäre das auch etwas übersichtlicher und in dieser tabelle würde olsr dann die default-route setzen - ich setz das mal auf meine todo liste. das würde das problem recht elegant erschlagen.

andibraeu, 2019-08-26:
in unserem Setup sagt ja der Router per HNA (0.0.0.0), dass er Internet hat. Das sollten wir auch beibehalten und nicht Richtung Server verschieben. Dann bleibts dezentraler.

Sprich: Wenn der Server bezüglich Internet nicht zum OLSR-Nachbarn wird, dann ist alles gut

Ich frage mich, wie groß das Problem hier tatsächlich ist und wie dringend wir das lösen müssen. Es wird ja nur problematisch, wenn der Router an einem 10er-Netz hängt. Oder ein Router per WAN an einen anderen Freifunkrouter angeschlossen wird. 

Vielleicht sollten wir es erstmal belassen und nicht unnötig verkomplizieren, zumal Umbauarbeiten in Zukunft tatsächlich anstehen

andibraeu, 2019-10-08:
ah, iif und oif gibt es nur bei ip route, nicht bei ip rule

andibraeu, 2019-10-08:
@mtippmann ich hab das mal aus der alten firmware übernommen. Jetzt gibt es "nur" noch Probleme, wenn das WAN-Interface auf eine Adresse im 10.63er-Netz gestellt ist. Zumindest ein Fortschritt.

Was denkst du?

mtippmann, 2019-10-08:
yup, habe 10.0.0.0/8 wegen intercity-vpn genommen glaub ich, aber ich kann micht daran erinnern, dass jemals genutzt zu haben

andibraeu, 2019-10-09:
die regel besagt jetzt, dass alles, was aus dem 10.63er Netz kommt und nicht nach 10.0.0.0/8 oder ins Roamingnetz will über den VPN geht.
intercityvpn müsste weiter gehen, wenn die Netze über OLSR angekündigt werden (momentan nur das Leipziger Netz)
Ein Sonderfall wäre, wenn jetzt ein Leipziger Gerät mit unserem Netz meshen würde, dann würden dessen Pakete ins Internet nicht über VPN ausgeleitet werden.

andibraeu, 2019-10-09:
Noch ne Idee: es müsste schon sehr mit dem Zufall zugehen, wenn WAN eine 10.63er Adresse bekommt, und diese Adresse nicht von einem Weimarnetzrouter kommt, oder?
Wenn das der Fall ist könnte man das internet über VPN gar nicht erst zulassen, und nur ein innercity-VPN starten (das funktioniert sogar, hatte ich aus Versehen schon getestet)