weimarnetz/weimarnetz #320: Mesh on WAN Funktion

mtippmann, 2018-08-23:
WAN soll auch Meschen für 1-Dosen Geräte wie die CPEs und co. Entweder Knopf oder irgendwas automatisch machen mit Liste...

mtippmann, 2019-09-03:
sollten wir überlegen, Klaus hat berge an 1-Port Geräten, wo man eventuell zwischen LAN / WAN umschalten können sollte - gleiches Problem auch für die Nano's die mal am WAN-Port hängen oder über LAN meshen sollen. Müssten wir mal Recherche betreiben, wie das wo anders gelöst wurde.

andibraeu, 2019-09-04:
können wir feststellen, ob ein Gerät 1 oder mehr LAN-Ports hat?
Oder wir bauen es so: WAN-Port für WAN nutzen, WAN-Port für Mesh nutzen

würde ich aber erst im nächsten Release machen

mtippmann, 2019-09-04:
wir haben es gestern mit einem 703n getestest, da lief wan und lan parallel auf eth0 nach ffwizard - das hat wohl funktioniert - zumindest bekam das gerät eine ip via dhcp über wan, zusätzlich zur fest gesetzten lan ip - evtl. kommt olsr nicht klar, das müssten wir nochmal testen evtl. hatten wir gestern nur pech.

andibraeu, 2019-09-04:
Aber ist WAN und Mesh sinnvoll auf einem Interface? Wann sollte das funktionieren?

mtippmann, 2019-09-04:
gute frage, eine idee wäre umschalten zu können - z.b. wäre lan-modus für die nanos im dnt sinnvoll, dass man die via lan koppeln könnte - der häufigste use-case wäre aber wan-port - eventuell geht beides irgendwie - ich teste die tage nochmal mit dem 703n von klaus.

andibraeu, 2019-09-04:
aber mesh macht doch nur sinn, wenn die gegenseite auch meshen kann. Oder man macht VLANs und braucht dann aber trotzdem einen Switch dazwischen

mtippmann, 2019-09-30:
ist erstmal ein Thema für nach dem Release...