weimarnetz/weimarnetz #326: Weimarnetz Konfigurationsseite erneuern

andibraeu, 2019-07-21:
- [ ] Knotennummer eintragen, wenn keine vorhanden ist (frisch geflasht)
- [ ] Knotennummer ist über Weboberfläche nicht änderbar
- [ ] SSID für eigenen AP festlegen (Default ist Weimarnetz <Knotennummer>)
- [ ] VPN-Schalter für "VPN aus", "VPN Cloud Connect", "VPN The full monty"
- [ ] Kontaktdaten auf Karte veröffentlichen? (Default=off)

andibraeu, 2019-08-12:
die Knotennummer in der luci-web-app anzugeben macht derzeit wenig Sinn. Das ginge nur, wenn der Router noch keine Nummer registriert hat. Das macht er aktuell aber sobald er Internet hat.

Brauchen wir das Feature also?

kaese77, 2019-08-13:
Fänd ich schon gut, falss er kein Internet hatte bei der ersten Einrichtung.

VG, René

andibraeu, 2019-08-13:
Wenn wir weiter übers Meshkit Images ausliefern ist die Knotennummer aber auch schon da.

Klingt wie ein Edge-Case der jetzt kein Showstopper für Production ist :D

closed 2019-10-05 by andibraeu