weimarnetz/weimarnetz #330: vpn mss problem

mtippmann, 2019-09-24:
login.yahoo.com geht nicht über vpn2 

iptables -A FORWARD -p tcp --tcp-flags SYN,RST SYN -j TCPMSS  --set-mss 768

mtippmann, 2019-09-24:
768 klappt, 1280 interessanterweise nicht, 1180 klappt wieder. Mal rechen was da wie passen muss, dann wird das einfach fest gesetzt - warum das aber notwendig ist und vor allem warum es bei vpn3 klappt und bei vpn2 nicht wäre noch interressant - eventuell double-nat und das port-forwarding evtl. kommt icmp nicht durch für pmtu?

storchi, 2019-09-25:
cs@2 ~ % sudo tracepath -n yahoo.de            
 1?: [LOCALHOST]                      pmtu 1500
 1:  192.168.122.1                                         0.338ms 
 1:  192.168.122.1                                         0.256ms 
 2:  176.9.45.193                                          0.451ms 
 3:  213.239.229.93                                        0.599ms 
 4:  213.239.245.177                                       6.197ms 
 5:  91.213.211.120                                       20.418ms 
 6:  188.125.89.125                                       21.917ms 
 7:  188.125.89.53                                        26.042ms asymm  6 
 8:  66.196.65.210                                        25.228ms asymm  6 
 9:  66.196.65.217                                        41.756ms asymm  7 
10:  66.196.65.21                                         41.806ms asymm  9 
11:  no reply
12:  no reply
13:  no reply
14:  no reply
15:  no reply
16:  no reply
17:  no reply

storchi, 2019-09-25:
cs@vpn3 ~ % sudo tracepath -n yahoo.de 
 1?: [LOCALHOST]                                         pmtu 1500
 1:  77.87.48.33                                           0.395ms 
 1:  77.87.48.33                                           0.474ms 
 2:  185.1.74.26                                           0.908ms 
 3:  87.245.233.114                                       85.696ms asymm  7 
 4:  no reply
 5:  216.115.96.27                                        96.479ms asymm  6 
 6:  216.115.96.0                                        113.047ms 
 7:  216.115.96.121                                      134.511ms 
 8:  216.115.96.62                                       155.029ms 
 9:  66.196.67.109                                       159.328ms 
10:  67.195.37.69                                        159.688ms 
11:  no reply
12:  no reply
13:  no reply
14:  no reply

mtippmann, 2019-09-30:
mtu wurde nicht gesetzt, war nur im master-branch so o_O

commit ee22e230 (2019-09-30): set mtu when bringing up interface - fixes #330
closed 2019-09-30 by mtippmann