weimarnetz/weimarnetz #331: bittorfscher router

storchi, 2019-10-08:


storchi, 2019-10-08:
sysupgrade -n
router hat kein netz
freie knotennummer gesucht - 98
webif auf weimarnetz registrator knotennummer manuell klappt nicht -> keinerlei konfig
log: `Tue Oct  8 11:11:02 2019 authpriv.notice dropbear[2304]: Password auth succeeded for 'root' from 192.168.1.127:55122
Tue Oct  8 11:13:29 2019 user.debug registrator: registrator started
Tue Oct  8 11:13:34 2019 user.warn registrator: Failed to update/register node with mac a0:f3:c1:a8:75:9e with code -3
Tue Oct  8 11:13:34 2019 user.debug registrator: registrator finished
Tue Oct  8 11:13:35 2019 user.notice ffwizard_init: Reload Service
Tue Oct  8 11:13:35 2019 user.notice ffwizard: error: wrong or missing node number`

mtippmann, 2019-10-08:
`/etc/config/ffwizard` ist leer. Vermutlich muss `ffwizard_autoconf` zurück. https://github.com/weimarnetz/weimarnetz/commit/68863597683365eee58334d394bbfb232718a11d

andibraeu, 2019-10-08:
warum? /usr/sbin/ffwizard_autoconf hat nichts gemacht, wirklich nichts.

evtl ist das hier noch nicht vollständig? https://github.com/weimarnetz/weimarnetz/blob/brauhaus/files/etc/uci-defaults/10-ffwizard-defaults.sh

da müsste ffwizard ja angelegt werden

andibraeu, 2019-10-08:
hm, ja, kann sein, dass wenn ffwizard nicht existiert, diese hiermit nicht angelegt wird

andibraeu, 2019-10-08:
hoffe, es geht jetzt besser

mtippmann, 2019-10-08:
> warum? /usr/sbin/ffwizard_autoconf hat nichts gemacht, wirklich nichts.

ich dachte schon, siehe https://github.com/weimarnetz/weimarnetz/blob/b508c52b83b0ea5c378065d1131a7953ffee195f/files/etc/init.d/ffwizard_autoconf 

`/etc/uci-defaults/20-autoconf` setzt `ffwizard_autoconf=1` damit läuft das beim Start und fügt die enstprechenden Abschnitte in der `/etc/config/ffwizard` hinzu. Das `/etc/uci-defaults/10-ffwizard` setzt nur ein paar Defaults für den System-Abschnitt der Wizard selbst nutzt diese Einstellungen um z.b. zu erkennen ob man ein Radio/Wifi Device aktiv hat oder nicht - ist da nix in der Datei dann klappt das nicht, weil die uci hooks da sections erwarten oder wurde das auch schon umgebaut?

mtippmann, 2019-10-08:
Also quasi: erster boot oder nach `sysupgrade -n`: ffwizard_autoconf legt einmalig die `/etc/config/ffwizard` datei an, wo für jedes radio eine section erstellt wird zusätzlich für vpn / ethernet etc.pp - unabhängig davon basteln wir mit der `/etc/uci-defaults/10-ffwizard` ein paar default-werte in die `settings`-section rein. Diese sections werden von den ffwizard-scripten genutzt um die jeweiligen interfaces zu konfigurieren. Dadurch, dass das autoconf nur beim ersten boot und sonst nicht mehr läuft kann man die `/etc/config/ffwizard` dazu nutzen z.b. interfaces zu aktivieren bzw. zu deaktivieren. Soweit dachte ich dass es funktioniert. Damit kann man korrekt ohne viel Aufwand dual-band und single-band geräte abbilden z.b. ohne größartige Verrenken machen zu müssen.

mtippmann, 2019-10-08:
@andibraeu ich bin ein idiot, du hast recht - die init.d datei ist ja noch da - wtf!? aber ich glaube die uci-faults änderungen sind nicht korrekt, das ist eigentlich nur dazu da statisch ein paar werte reinzuhauen, eventuell liegt es daran? aber selbst das kann nicht sein.

andibraeu, 2019-10-08:
das script /etc/init.d/ffwizard_autoconf ist auch na wie vor da... daran habe ich auch nichts verändert. nur /usr/sbin/ffwizard_autoconf hat nichts gemacht, sondern gleich returned

andibraeu, 2019-10-08:
ja, siehe commit, evtl. ging nichts, weil die Datei nicht angelegt wurde, wenn sie nicht da war. das sollte dann behoben sein

mtippmann, 2019-10-08:
okay, das klingt plausibel, ich teste mal!

andibraeu, 2019-10-08:
das ist aber noch nicht gebaut, hab es nur committed

mtippmann, 2019-10-22:
läuft. @storchi  hat getestet.

closed 2019-10-22 by mtippmann