weimarnetz/weimarnetz #332: Sprechende Versionsinfo

andibraeu, 2019-10-09:
Aktuell sieht die Version z.B. im Monitoring so aus: `/ r6907+970-7e15e21766`

Das sollte besser lesbar sein und unseren Releasenamen enthalten

mtippmann, 2019-10-09:
ja, das ist die OpenWrt-Revision - evtl. hat sich da was geändert.

andibraeu, 2019-10-09:
hoffentlich fixed in https://github.com/weimarnetz/firmware/commit/cb91c6f41ddc999e74320b4b113bf1089c1fc519

andibraeu, 2019-10-09:
hat leider doch noch nicht geklappt...

andibraeu, 2019-10-09:
```
DISTRIB_ID='ffweimar brauhaus powered by OpenWrt'
DISTRIB_RELEASE='18.06-SNAPSHOT'
DISTRIB_REVISION='r6907+970-7e15e21766'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='ffweimar brauhaus powered by OpenWrt 18.06-SNAPSHOT r6907+970-7e15e21766'
DISTRIB_TAINTS='no-all busybox'
```
das steht drin, ggf. sollten wir einfach ein anderes feld ans monitoring schicken

mtippmann, 2019-10-10:
habe überlegt eventuell commit vom weimarnetz repo + packages repo + firmware repo nochmal extra irgendwo hinzupacken und das via imagebuilder $FILES variable einfach nach /etc/ zu packen. 

was für ein wtf - also deine änderung war schon korrekt, das problem war aber das sed nicht gegriffen hat und die echo zeile irgendwie auch nicht... mit noch abschauen bei berlin und dem hinzufügen von der config in common.config damit sed auch greift ist es jetzt zumindest in der `.config` nach `make prepare` mal sehen ob das klappt... 0eb9ab552aac73cd2d0e0065b768e48a8acc9270

mtippmann, 2019-10-10:
die änderungen bzgl. tmp sind nicht so wichtig, aber die stellen an deren das in berlin gelöscht wird sind besser als das in make prepare zu löschen.

closed 2019-10-11 by andibraeu