weimarnetz/weimarnetz #337: brauhaus-100 not compiling with 19.0.7.1 in ar71xx

kaese77, 2020-02-11:
cd /home/xtek/Dokumente/weimarnetz/firmware/build/brauhaus/openwrt; quilt push -a || ( [ $? -eq 2 ] && true )
Applying patch patches/000-ar71xx-tiny_reduce_kernelsize.patch
patching file target/linux/ar71xx/tiny/config-default
Hunk #1 succeeded at 74 (offset 4 lines).

Applying patch patches/0007-mac80211-ath10k-allow-simultaneous-AP-IBSS.patch
patching file package/kernel/mac80211/patches/950-ath10k_AP_IBSS.patch

Applying patch patches/201-luci-remove-freifunk-firewall.patch
patching file feeds/luci/modules/luci-mod-freifunk/Makefile

Applying patch patches/802-fragment-cache-size-4.9.patch
can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|Index: openwrt/target/linux/generic/config-4.9
|===================================================================
|--- openwrt.orig/target/linux/generic/config-4.9
|+++ openwrt/target/linux/generic/config-4.9
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch patches/802-fragment-cache-size-4.9.patch does not apply (enforce with -f)
Makefile:98: recipe for target '.stamp-patched' failed
make: *** [.stamp-patched] Error 1

kaese77, 2020-02-11:
changed title

andibraeu, 2020-04-19:
loift jetzt

closed 2020-04-19 by andibraeu